raspberrypi/firmware #1518: 2020.12.08 start4.elf does not properly clear xHCI Controller Not Ready (CNR) flag after reset

pbatard, 2020-12-24:
### Bug description

While testing the EDK2 UEFI firmware with recent `start4.elf`/`fixup.dat` releases on RPi4, we found that an ASSERT was triggered in the EDK2 code during xHCI initialization, after the xHCI controller is reset.

That ASSERT, which is located in xHCI initialization code (that is common to many platforms and not being specific to RPi) and is located [here](https://github.com/tianocore/edk2/blob/master/MdeModulePkg/Bus/Pci/XhciDxe/Xhci.c#L2018), is basically triggered if, after a chip hardware reset, the Controller Not Ready (CNR) flag remains active.

We have identified that this issue started to manifest itself with 919aee0ed75f7db48a38b8b96c13228a7584cfd7 (2020.12.08) and is not present in 08ed7a0c9ad4d9db559aaec462520ab435c7ce1c (2020.12.01). We have also confirmed that the issue is present in all subsequent releases, including current.

Strangely, e15ef4e4fe8be99cd816cec901d977224a1eb07e (2020.11.25), which pertains to xHCI reset (_"Don't reset BCM2711 XHCI from filesys in start.elf"_) does not seem to be the commit that introduces the problem, as we tested the `start4.elf` at this revision and it was fine.

### Steps to replicate

1. Download the latest UEFI release from https://github.com/pftf/RPi4/releases/tag/v1.21 and extract it to SD or USB media
2. Replace `RPI_EFI.fd` with the `Build\RPi4\DEBUG_GCC5\FV\RPI_EFI.fd` version from the [AppVeyor build artefact](https://ci.appveyor.com/project/pbatard/RPi4/build/artifacts) from the same release. The reason one needs to use the DEBUG version rather than the RELEASE version is because ASSERTs are ignored altogether in RELEASE, and the UEFI firmware ultimately boots, though with an extended delay (~20 seconds) during xHCI initialization). One also gets extended serial debug output with the DEBUG UEFI firmware.
3. Replace `start4.elf`/`fixup4.dat` on SD/USB with a 2020.12.08 or later version and boot.

### Result

After powerup, the UEFI boot process eventually fails with:

```
(...)
XhcCreateUsb3Hc: Capability length 0x20
XhcCreateUsb3Hc: HcSParams1 0x5000420
XhcCreateUsb3Hc: HcSParams2 0xFC000031
XhcCreateUsb3Hc: HcCParams 0x2841EB
XhcCreateUsb3Hc: DBOff 0x100
XhcCreateUsb3Hc: RTSOff 0x200
XhcCreateUsb3Hc: UsbLegSupOffset 0x0
XhcCreateUsb3Hc: DebugCapSupOffset 0x260
XhcSetBiosOwnership: called to set BIOS ownership
XhcResetHC!
ASSERT [XhciDxe] /home/appveyor/projects/rpi4/edk2/MdeModulePkg/Bus/Pci/XhciDxe/Xhci.c(2018): !((((BOOLEAN)(((XhcReadOpReg ((Xhc), (0x0004))) & ((0x00000800))) == ((0x00000800))))))
```

which is a direct consequence of the Controller Not Ready (CNR) flag not being cleared after Chip Hardware Reset, as per https://github.com/tianocore/edk2/blob/master/MdeModulePkg/Bus/Pci/XhciDxe/Xhci.c#L2014-L2018.

Note that, in case you wonder if the issue could be tied to the UEFI code not waiting for the CNR flag to change (which it should probably do, per the comments, but currently doesn't), we also tested versions were we added a 30 seconds wait for the CNR flag to clear, but the outcome was the same:

```
XhcCreateUsb3Hc: Capability length 0x20
XhcCreateUsb3Hc: HcSParams1 0x5000420
XhcCreateUsb3Hc: HcSParams2 0xFC000031
XhcCreateUsb3Hc: HcCParams 0x2841EB
XhcCreateUsb3Hc: DBOff 0x100
XhcCreateUsb3Hc: RTSOff 0x200
XhcCreateUsb3Hc: UsbLegSupOffset 0x0
XhcCreateUsb3Hc: DebugCapSupOffset 0x260
XhcSetBiosOwnership: called to set BIOS ownership
XhcResetHC!
XhcDriverBindingStart: waiting on Controller Ready...
XhcDriverBindingStart: waiting on Controller Ready...
XhcDriverBindingStart: waiting on Controller Ready...
XhcDriverBindingStart: waiting on Controller Ready...
XhcDriverBindingStart: waiting on Controller Ready...
XhcDriverBindingStart: waiting on Controller Ready...
XhcDriverBindingStart: waiting on Controller Ready...
XhcDriverBindingStart: waiting on Controller Ready...
XhcDriverBindingStart: waiting on Controller Ready...
XhcDriverBindingStart: waiting on Controller Ready...
XhcDriverBindingStart: waiting on Controller Ready...
XhcDriverBindingStart: waiting on Controller Ready...
XhcDriverBindingStart: waiting on Controller Ready...
XhcDriverBindingStart: waiting on Controller Ready...
XhcDriverBindingStart: waiting on Controller Ready...
XhcDriverBindingStart: waiting on Controller Ready...
XhcDriverBindingStart: waiting on Controller Ready...
XhcDriverBindingStart: waiting on Controller Ready...
XhcDriverBindingStart: waiting on Controller Ready...
XhcDriverBindingStart: waiting on Controller Ready...
[... 87 lines cut ...]
MESS:00:00:07.073800:0: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
MESS:00:00:07.077402:0: Loading 'bcm2711-rpi-4-b.dtb' to 0x1f0000 size 0xbb64
MESS:00:00:07.096477:0: brfs: File read: 47972 bytes
MESS:00:00:07.153627:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:07.156197:0: brfs: File read: 206 bytes
MESS:00:00:07.160916:0: brfs: File read: /mfs/sd/overlays/miniuart-bt.dtbo
MESS:00:00:07.186620:0: Loaded overlay 'miniuart-bt'
MESS:00:00:07.239581:0: brfs: File read: 1819 bytes
MESS:00:00:07.241477:0: Failed to open command line file 'cmdline.txt'
MESS:00:00:08.376971:0: brfs: File read: /mfs/sd/RPI_EFI.fd
MESS:00:00:08.379437:0: Loading 'RPI_EFI.fd' to 0x0 size 0x1f0000
MESS:00:00:08.385262:0: No compatible kernel found
MESS:00:00:08.389764:0: Device tree loaded to 0x1f0000 (size 0xbfd8)
MESS:00:00:08.397597:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:08.404917:0: uart: Baud rate change done...
MESS:00:00:08.406939:0: uart: Baud rate change done...
MESS:00:00:08.412025:0: bfs_xhci_stop
MESS:00:00:08.415186:0: XHCI-STOP
MESS:00:00:08.418302:0: xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
MESS:00:00:08.427562:0: PCI reset
NOTICE:  BL31: v2.3():v2.3
NOTICE:  BL31: Built : 10:40:51, Apr 21 2020
UEFI firmware (version UEFI Firmware v1.22-pre built at 16:26:35 on Dec 23 2020)
Board Rev: 0xD03114
```

vianpl, 2020-12-24:
I've been bitten by this change myself. From what I can tell starting at 919aee0  the _dma-ranges_ property inserted by firmware into PCI's DT node is different from what it used to be: DMA addresses now have an 8GB offset regardless of the SoC revision (on B0 revisions of the SoC the _CPU<->BUS_ address space translation used to be transparent). See how we're working to fix this in [u-boot](http://u-boot.10912.n7.nabble.com/PATCH-v5-00-13-Raspberry-Pi-400-Compute-Module-4-support-td435983.html).

IIUC you are also triggering an xHC firmware load through the mailbox interface. AFAIK this mailbox interface expects a configured PCIe bus honoring whatever is in DT.

pbatard, 2021-01-06:
@vianpl, thanks for the pointers. I believe our issue is related to #1495, where changes that were initially limited to CM4 and Pi 400 have now been applied to Pi 4 with newer SoC revisions. I was actually able to confirm this by testing that the newer `start4.elf` works with an older Pi 4 4 GB I have, purchased in September 2019, whereas it fails with a newer Pi 4 8 GB purchased in August 2020.

And the problem is that, rather than the xHCI not clearing the CNR flag after reset, the VL805 firmware is flushed out and not loaded again...

This really makes the VL805 Pi implementation a pain to work with, as it suddenly requires tons of people to invest time implementing something (taking care of loading firmware into an xHCI controller) that no other platform requires...

timg236, 2021-01-06:
This is the expected behaviour since it can't make assumptions about how the next stage will use PCIe. 

N.B. Not sure if it's in the current rpi-update but inbound SDRAM will be at 16GB rather than making this dynamic
" // Hard-code memory to sit at 0x4_00000000"

pelwell, 2021-01-06:
Yes, it first appeared in the 5.4.81 release.

jlinton, 2021-01-06:
I'm wondering if its possible to flip the dma window, make the mailbox call, then change the firmware back to more sane values.

jlinton, 2021-01-22:
It's actually trivial to fix this in edk2. I've done that, and it works. The problem is that we are trying to run ACPI based OSs. The correct fix is then requires a tweak to the _DMA() attribute to set the PCI root port translation.

We then hit the real problem! _DMA() is very poorly supported. Most !embedded arm platforms don't tend to do a lot of PCI DMA translations. Meaning, Linux and various other OS's don't have complete implementations. Given a quick test, OS's including Linux, that previously worked, are now failing.

As is mentioned in defect #1495 (XHCI_RESET mailbox command appears to assume a particular RC_BAR2 state) it would be beneficial if the VC firmware either honored the DT provided or read the current RC_BAR2 (and MEM_WIN0) state before trying to load the vl805 images. Alternatively, maybe a config.txt option could be provided that applies the previous 3G 1:1 translation across SOC/board revisions? My personal opinion is that the second choice, reading the existing RC_BAR2/etc state is the best one.

Without one of those three options, I fear we are left with the far less desirable choice of handling the vl805 firmware loads ourselves in UEFI. That is undesirable for a long list of reasons, but it gives UEFI the option to maintain a 1:1 DMA translation, and possibly also move the MMIO 0x600000000->0xf8000000 to a 1:1 as well.

timg236, 2021-01-22:
We've had a discussion and believe that it would be ok for the XHCI FW loading mailbox to get the inbound window address from RC_BAR2 since the firmware doesn't have any dependencies on this after the kernel is loaded.

The firmware will always configure this to be 16GB but presumably, the next stage can change this and patchup DT as appropriate.

samerhaj, 2021-02-12:
> We've had a discussion and believe that it would be ok for the XHCI FW loading mailbox to get the inbound window address from RC_BAR2 since the firmware doesn't have any dependencies on this after [...]

@timg236 is this firmware change still planned? We have a growing number of issues reporting XHCI no longer works with UEFI on the RPi4 (and a similar number of issues wanting UEFI on the RPi 400 and CM4). The change you proposed here will definitely help address this. Thanks!

andreiw, 2021-02-12:
I wholeheartedly support the change to honor RC_BAR2 and am equally eager to know when the change will land :-). Do you have any early preview builds to share?

timg236, 2021-02-13:
I’ll attach a patch next week. I’ve been to busy recently to test this properly and things aren’t changing. However it’s a small change so will do the basics and share

timg236, 2021-02-15:
The attached FW uses BAR2 registers to configure the VL805 FW address instead of device-tree. Tested with USBBOOT on Pi 4B 8GB Raspberry Pi OS.

[bar2-fw.zip](https://github.com/raspberrypi/firmware/files/5983076/bar2-fw.zip)

jlinton, 2021-02-16:
So, at a first pass this appears to be working on an older rpi4-4G (with the dedicated xhci flash), an older 8G without the flash, and a 8G with the newer SoC.

Woot, thanks!

Of course more testing is in order, but assuming that goes well, when might we expect this in one of the official releases?

andreiw, 2021-02-16:
I tried a Pi 400 and my "special" Pi4-8GiB with the new SoC. Works  great with the latest PFTF 1.22 release.

Same question as Jeremy :-). When?

timg236, 2021-02-16:
It should be in rpi update in about a week if @pelwell is happy with it. Apt takes a bit longer but I think there will be an apt release early March

samerhaj, 2021-02-16:
Thank you @timg236!

Matioupi, 2021-02-17:
> So, at a first pass this appears to be working on an older rpi4-4G (with the dedicated xhci flash), 

what is this dedicated 4G xHCI flash ? I have an old (rev 1.2 pi4 4G) and don't find info about this ?
regards

pelwell, 2021-02-17:
The earlier revisions of the 4B were fitted with a dedicated SPI EEPROM holding the VL805 firmware, whereas newer revisions have the VL805 firmware in the main boot EEPROM and the VL805 loads its firmware from system RAM.

Matioupi, 2021-02-17:
thanks, and is there some special actions to take to run with on fw or the other on the 4G rev 1.2 ?

pelwell, 2021-02-17:
No - the combination of the current common rpi-eeprom-update script and boot EEPROM image handles boards with and without EEPROM. However, OS images that reset the PCIe controller need to send the firmware a special mailbox call in order that it can reconfigure the VL805 on boards without the VL805 EEPROM.

evadot, 2021-02-17:
This also fixes using xhci in FreeBSD.
Will you commit and tag a release before next week ? We obviously don't care about rpi update to build our rpi-firmware port in FreeBSD land :)
Thanks.

timg236, 2021-02-17:
It's merged into our internal Git repo, not sure if it will be released to the rpi firmware repo before then but I'll update this bug if it is

timg236, 2021-02-24:
Updated in https://github.com/Hexxeh/rpi-firmware/commit/ca28024e2a78d6b1a05db5bbb77f6543fe569957

jeroenbrons, 2021-02-25:
keyboard and mouse work for me on a PI400 with the firmware from the link. currently booting win10 ARM64 to test if it works there as well

jeroenbrons, 2021-02-25:
Windows 10 for ARM64 just booted and seems to work.

pbatard, 2021-02-25:
@timg236, many thanks for the firmware update. I have confirmed that the latest `start4.elf` from the official repo addresses the original issue reported here. I will therefore close this ticket.

closed 2021-02-25 by pbatard

evadot, 2021-02-25:
Will there be any tags for today's update ?