raspberrypi/firmware #1519: We need 64bit firmware libs for Raspbian 64bit

zloop1982, 2020-12-30:
We can not build OpenGLES program for Raspbian 64bit system.
Because the "libbrcmGLESv2.so" in  "opt/vc/lib/" is not aarch64.
I don't know if the libGLESv2  in "/usr/lib/aarch64-linux-gnu/" is ok.
so we need an 64bit version firmware for Raspbian 64bit

Or , release the libGLESv2 libEGL source code , let people build it by self.

6by9, 2021-01-02:
All the source for libbrcmGLESv2.so is in https://github.com/raspberrypi/userland. Clone it and run ```./buildme --aarch64``` to build and install 64bit libraries for yourself.
I've never checked whether those GLES libraries work on 64bit, and it's not going to be a priority.

I presume you're on a Pi3, otherwise libbrcmGLESv2.so is irrelevant anyway as there is no firmware based GLES driver.
Pi4 uses Mesa and ARM side control for the 3D hardware. The preference would be to use the ARM side drivers on the Pi3 as well by loading ```dtoverlay=vc4-fkms-v3d``` or ```dtoverlay=vc4-kms-v3d```.

zloop1982, 2021-01-06:
> Pi4 uses Mesa and ARM side control for the 3D hardware. The preference would be to use the ARM side drivers on the Pi3 as well by loading `dtoverlay=vc4-fkms-v3d` or `dtoverlay=vc4-kms-v3d`.

Thank you, yes I use pi4
So,Raspbian 64bit os On Pi4, Don't need  libbrcmGLESv2.so, right?
So, On Pi4 with Raspbian 64bit,I just need install libgles2-mesa libgles2-mesa-dev xorg-dev, right?

JamesH65, 2021-01-06:
Just to clarify some terms, the name Raspbian has been replaced by Raspberry Pi OS for the official distro. And note, there has never been a 64bit Raspbian. So use something like RPiOS32 and RPiOS64 as abbreviations.

popcornmix, 2021-01-06:
Correct. On Pi4 libbrcmGLESv2.so is of no use. You need the standard debian mesa libraries installed through apt.

zloop1982, 2021-01-06:
> Correct. On Pi4 libbrcmGLESv2.so is of no use. You need the standard debian mesa libraries installed through apt.

Thank you. Got it.

asavah, 2021-03-16:
Please provide in this repo, as opposed to some google drive links I've seen around, pure aarch64 or static builds of binaries that can't be built from userland repo due to source files not being present, such as vcdbg and edidparser, the latter isn't strictly necessary as there are opensource alternatives, but the former is needed for diagnostics.
I'm aware that https://git.venev.name/hristo/rpi-vc-log/ exists but it would be nice to have an official tool available from official (binary) sources.

Please be aware that not every pi customer is using raspios mess and that building a multiarch os to support just one tool is an overkill.