raspberrypi/firmware #1521: Mailbox set clock property behaviour is different from Pi 3

worproject, 2021-01-04:
On a Pi 3 I've been able to overclock the ARM cores regardless of the `arm_freq` value in config.txt.
On the Pi 4 this has been changed, and the clock can be only set up to `arm_freq`. 

The issue is that if I attempt to set the clock higher than the value of `arm_freq`, the mailbox always returns my requested value instead of the actual clock rate.

pelwell, 2021-01-04:
Moved to the firmware repo - this has nothing to do with the EEPROM.

popcornmix, 2021-01-04:
The code does an explicit
```
   arm_freq = _min(_max(arm_freq, arm_min), arm_max);
```
where arm_min and arm_max come from config.txt.
So I don't believe you can do this on a Pi3 either.
This is intended behaviour.

worproject, 2021-01-04:
With the latest firmware probably not, but in the past I know for sure that it was possible. (i can't test it now)
The issue is that the mailbox doesn't check if the frequency has been changed successfully. It just returns the requested value, which may or may not be the actual clock.

Let's say `arm_freq` is set to `1600` in config.txt. If I set the ARM clock to `3000000000` (3 GHz) through mailbox (SkipTurbo=1), subsequent requests with the "Get clock rate" tag return `3000000000` instead of the actual clock rate, which is `1600000000`.

popcornmix, 2021-01-04:
That is deliberate. The kernel typically does a:
```
if (get_clock(arm) != desired)
  set_clock(desired)
```
which caused lost requests when the clock isn't at value the arm had last requested.
e.g. The arm requests 1200. Firmware side throttling forces the arm clock to 600. When idle the arm skips its request for 600 and so when throttling has completed we are at an unwanted 1200.

So GET_CLOCK_RATE will return the last value SET_CLOCK_RATE set.
If you need to know the true value which respects clamping, throttling and clock divider limitations then you can use GET_CLOCK_MEASURED.

worproject, 2021-01-04:
What's the tag of GET_CLOCK_MEASURED? I can't find it in https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface

pelwell, 2021-01-04:
GET_CLOCK_MEASURED is 0x30047.

closed 2021-01-04 by worproject