raspberrypi/firmware #1523: Broken tc358743 on the latest firmware

mdevaev, 2021-01-08:
**Describe the bug**
On the latest raspberry pi os update received via rpi-update, tc358743 detection on the i2c bus broke:
```
pi@raspberrypi:~ $ dmesg | grep tc3
[    6.347993] tc358743 10-000f: i2c_rd: reading register 0x0 from 0xf failed
[    6.348013] tc358743 10-000f: not a TC358743 on address 0x1e
```

**To reproduce**
1. Connect any HDMI-CSI bridge based on TC358743 to the Pi
2. Download and flash official Raspberry Pi OS Lite.
3. Add line `dtoverlay=tc358743` to `/boot/config.txt`
4. Reboot and make sure that th device was detected using `dmesg | grep tc3`.
5. Perform `rpi-update`. It will install linux 5.10.5.
6. Reboot and run `dmesg | grep tc3` again. Device will not be detected.

**Expected behaviour**
TC358743 must be detected

**Actual behaviour**
TC358743 is not detected

**System**
* Raspberry Pi 4 2Gb model
* ```
  pi@raspberrypi:~ $ cat /etc/rpi-issue
  Raspberry Pi reference 2020-12-02
  Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage
  ```
* ```
  pi@raspberrypi:~ $ vcgencmd version
  Jan  7 2021 18:27:29
  Copyright (c) 2012 Broadcom
  version fb345a0c2d5544957f4ba1a2b9e968970e3312c4 (clean) (release) (start)
  ```
* Linux raspberrypi 5.10.5-v7l+ #1391 SMP Thu Jan 7 17:51:37 GMT 2021 armv7l GNU/Linux

**Additional context**
The problem was detected on Arch Linux ARM when using kernel 5.4.

This version is still working:
https://github.com/raspberrypi/firmware/commit/8a5549c137dbce0e082ecf21607dc5e3feac1160

And this not:
https://github.com/raspberrypi/firmware/commit/9a0d11557e13e121ddac3c7ff4fac9d8bc629a76

mdevaev, 2021-01-08:
@6by9 Sorry to bother you, but you may be interested in this problem

6by9, 2021-01-08:
Almost certainly kernel and not firmware. I was about to say that the firmware drivers for TC358743 are totally unsupported based on the title!

That said there is a change to the way the firmware restores I2C pinmuxing in https://github.com/raspberrypi/firmware/commit/0ee1d9f2fc852b20e247bb832a3b8d0a13b6ada6
https://github.com/raspberrypi/firmware/commit/9a0d11557e13e121ddac3c7ff4fac9d8bc629a76 has a revert of that as it's being taken for a stable release. Could you just check those two please?

mdevaev, 2021-01-08:
Sure.
```
pi@raspberrypi:~ $ sudo rpi-update 0ee1d9f2fc852b20e247bb832a3b8d0a13b6ada6
 *** Invalid hash given
 ```
 Umm... Am I missing something?

popcornmix, 2021-01-08:
421c3cab89636953781dcfc624bd7641ef962220 is the right hash for the revert.
5467199574e4a6621ded42359dfd9aadfc4793ee is the hash that may have introduced the issue.
See: https://github.com/Hexxeh/rpi-firmware/commits/

6by9, 2021-01-08:
You'd quoted hashes in raspberrypi/firmware (this repo), so I'd done likewise.

rpi-update hash https://github.com/Hexxeh/rpi-firmware/commit/421c3cab89636953781dcfc624bd7641ef962220

6by9, 2021-01-08:
And the ouptut from ```raspi-gpio get``` after each has booted would be useful.

mdevaev, 2021-01-08:
421c3cab89636953781dcfc624bd7641ef962220: OK, device detected, capture is working. https://pastebin.com/raw/WAtPEP5a

5467199574e4a6621ded42359dfd9aadfc4793ee: FAIL, not detected

6by9, 2021-01-08:
Have you got the ouput from raspi-gpio with the failing firmware?

And presumably you're running with the camera enabled (```vcgencmd version``` reporting ```start_x```). I'm not expecting anything to have changed for the other variants as they don't do any I2C other than the HAT.

mdevaev, 2021-01-08:
Yes, sorry: https://pastebin.com/raw/uBiHAKeq
```
Jan  7 2021 18:27:29
Copyright (c) 2012 Broadcom
version fb345a0c2d5544957f4ba1a2b9e968970e3312c4 (clean) (release) (start)
```

6by9, 2021-01-08:
That double mapping of SDA0/SCL0 on 0/1 and 44/45 would do it.

Hang on, ```start```, not ```start_x```. Hmm, strange as I wouldn't expect any issues there. Checking....

mdevaev, 2021-01-08:
Let me ask a stupid question. What is the difference between these parameters? I've always used start_x to enable omx/mmal encoder

6by9, 2021-01-08:
@popcornmix Was the restating commit:
- built fresh from firmware source with the commits reinstated, or
- a revert to the previous built release?

Strings on start4.elf appears to say source hash fb345a0c2d5544957f4ba1a2b9e968970e3312c4, which is the old source commit.

Testing on 13fb7549d5f77de3aa534a91f512c4580da0a1f6 appears to be OK, but that was with a debug ELF.
Now building a release ELF....

6by9, 2021-01-08:
> Let me ask a stupid question. What is the difference between these parameters? I've always used start_x to enable omx/mmal encoder

```vcgencmd version``` should report which variant of the firmware is running be it:
- ```start_cd``` for cutdown - active when gpu_mem is <=16MB
- ```start``` for normal builds (no camera)
- ```start_x``` for camera support. ```start_x=1``` in config.txt.
- ```start_db``` for the debug build which includes all assert logging. ```start_debug=1``` in config.txt

If running on a Pi4 then it switches to the variants of the firmware in /boot with the 4 in the name.

6by9, 2021-01-08:
Release build issue by the looks of it :-(
13fb7549d5f77de3aa534a91f512c4580da0a1f6 debug build works
13fb7549d5f77de3aa534a91f512c4580da0a1f6 release build doesn't.

Now to work out what the compiler has optimised away!

mdevaev, 2021-01-08:
Many thanks. That is, should I use start instead of start_x for tc358743?

6by9, 2021-01-08:
TC358743 kernel drivers shouldn't need ```start_x=1```.

The x adds:
- the firmware camera stack (not needed)
- the MMAL rawcam component (not needed)
- mjpeg encoding (not sure why that's a difference)
- some audio capture code (I suspect used for the demo of the tc358743 firmware drivers, but I never got it to work. Use the I2S driver).

6by9, 2021-01-08:
Something weird going on and with release builds we don't get an I2C close from some device.
No close means we don't clean up.

6by9, 2021-01-08:
Oh my, the code to close the I2C handle after POE probe failure was within an assert! That would do it.

Then again there appears to be no close at all should the POE probe succeed. Another one to fix.

XECDesign, 2021-01-08:
And that's why I shouldn't be allowed near the firmware

6by9, 2021-01-08:
> And that's why I shouldn't be allowed near the firmware

I wasn't going to bother finding ```git blame```.

6by9, 2021-01-08:
Patch created.

As a temporary fix you can add ```disable_poe_fan=1``` to config.txt and it should avoid the issue.

mdevaev, 2021-01-08:
Lol nice :) I can confirm the fix is working on the latest build obtained by rpi-update

6by9, 2021-01-11:
Firmware patch merged internally. Should be in the next rpi-update release.

mdevaev, 2021-01-11:
Thanks! Will wait.

6by9, 2021-01-19:
The fix has been in the last two rpi-update releases. Could you retest and close if happy?

mdevaev, 2021-01-22:
```
pi@raspberrypi:~/ustreamer $ uname -a
Linux raspberrypi 5.10.9-v7l+ #1396 SMP Thu Jan 21 15:57:42 GMT 2021 armv7l GNU/Linux
pi@raspberrypi:~/ustreamer $ vcgencmd version
Jan 21 2021 16:02:55
Copyright (c) 2012 Broadcom
version a5bf70c44ba93e99f955a66530916240a1661132 (clean) (release) (start)
```
Everything is working. Many thanks!

closed 2021-01-22 by mdevaev