raspberrypi/firmware #1525: pi0 arm core voltage isnt being set right with freq scaling

cleverca22, 2021-01-12:
**Describe the bug**
the core voltage the firmware picks for a given frequency depends on what range of frequencies you can select

with the default config, manually cycling thru the 4 available frequencies gives the following:
```
arm_clock pllb    divisor   core_v
700mhz    2.8ghz  /4        1.2 
800mhz    3.2ghz  /4        1.35
900mhz    1.8ghz  /2        1.35
1ghz      2ghz    /2        1.35
```
as you might expect, the lowest clock has a lower voltage, however!
```
arm_freq_min=100
arm_freq=100
```
this changes which frequencies are available, and things now start to get weird:
```
arm_clock pllb    divisor   core_v
100mhz    2.4ghz  /24       1.2 
200mhz    2.4ghz  /12       1.35
300mhz    2.4ghz  /8        1.35
400mhz    3.2ghz  /8        1.35
500mhz    2ghz    /4        1.35
600mhz    2.4ghz  /4        1.35
700mhz    2.8ghz  /4        1.35
```
previously, 700mhz ran at 1.2v, but now the whole 200mhz-700mhz range runs at 1.35v instead, and the new slowest speed is the only one at 1.2v
it looks like the voltage selection code isnt aware of the actual frequency, and is just blindly giving the slowest speed 1.2v, and all others 1.35v

**To reproduce**
```
root@pi400:~# echo userspace > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor
root@pi400:~# echo 700000 > /sys/devices/system/cpu/cpufreq/policy0/scaling_setspeed 
root@pi400:~# vcgencmd measure_clock arm ; vcgencmd measure_clock pllb ; vcgencmd measure_volts core
```

**Expected behaviour**
700mhz should have the same voltage, reguardless of what other clocks i have available

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry PiZeroW: `Revision        : 9000c1`
* Which OS and version (`cat /etc/rpi-issue`)? `Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage4`
* Which firmware version (`vcgencmd version`)? `Nov 30 2020 22:13:46 version ab1181cc0cb6df52bfae3b1d3fef0ce7c325166c (clean) (release) (start)`
* Which kernel version (`uname -a`)? `Linux pi400 5.4.79+ #1373 Mon Nov 23 13:18:15 GMT 2020 armv6l GNU/Linux`

MichaIng, 2021-03-11:
This behaviour seems to be somehow expected, though not exactly documented linke that. `over_voltage_min` (1.2A by default) is only applied when the ARM runs at its lowest frequency, regardless which one this is based on the `arm_freq_min`/`arm_freq` and `over_voltage` (1.35A by default) is applied in any other case.

I have a long waiting pull request open to have this behaviour documented, including the aim to bring the question to the developers attention whether this is expected and reasonable that way of if it might deserve a change: https://github.com/raspberrypi/documentation/pull/1698

cleverca22, 2021-03-22:
@MichaIng the original goal of my underclocking, was to run the arm at ~25mhz, which i did sucessfully do, but this bug is something i found along the way, and i can see it harming power usage

given that 700mhz runs at 1.2v with all defaults
then the entire 100mhz-700mhz range should also run at 1.2v just fine, and that would then reduce the waste heat

MichaIng, 2021-03-23:
Yes as said, it behaves as expected, which does not mean that it is somehow smart. You need to lower `arm_freq_min` together with `over_voltage_min` and `arm_freq` together with `over_voltage`. Voltage is not automatically lowered when you lower the frequencies, but it detects and sticks with the hardcoded defaults only based on the fact whether the ARM runs at it's lowest scheduling frequency currently or not.

cleverca22, 2023-12-23:
https://forums.raspberrypi.com/viewtopic.php?p=2172164#p2172164

and i have observed the same bug on the pi5 as well
from this measurement, it looks like there is just a min/max voltage/clock
it then generates 100mhz steps in the clock, and just cuts the voltage range up into even chunks and slaps it on the clocks

so when the min clock changes, but min-voltage doesnt, it winds up using different voltages for everything