raspberrypi/firmware #1526: Possible regression in start.elf causing problems booting over PXE with an initramfs

alekow, 2021-01-13:
Hello,

There is a possible regression in start.elf since commit 11a76e0. Versions of start.elf and fixup.dat newer than 15.04.2020 are causing problems booting over PXE with an initramfs.
Swapping start.elf and fixup.dat with any version older than 16/04/2020 makes the installer boot without errors.

**Raspberry Pi configuration:**

Version 3b+, connected to PXE host with Ethernet cable. No microSD card inserted, no USB drives connected.

**PXE Host system configuration:**

Slackware64-current (13/02/2020)
Static IP: 192.168.1.1

dnsmasq.conf:

```
interface=eth1
domain=example.org
dhcp-range=192.168.1.2,192.168.1.254,24h
pxe-service=0,"Raspberry Pi Boot   "
enable-tftp
tftp-root=/tftpboot/sarpi-current
log-dhcp

```
Sarpi installer downloaded from:
[https://slackware.uk/sarpi/rpi3/current/img/sarpi3-installer-boot_slackcurrent_11Jan21_sp1.zip](https://slackware.uk/sarpi/rpi3/current/img/sarpi3-installer-boot_slackcurrent_11Jan21_sp1.zip)
Unpacked and put under /tftpboot/sarpi-current

/tftpboot/sarpi-current mounted under /nfs/sarpi-current
`mount -o bind /tftpboot/sarpi-current /nfs/sarpi-current`

cmdline.txt file changed to enable PXE boot:
`dwc_otg.lpm_enable=0 console=tty1 nofont root=/dev/nfs nfsroot=192.168.1.1:/nfs/sarpi-current,proto=tcp rw ip=dhcp rootwait elevator=deadline`

dnsmasq, nfs and rpc daemons enabled and nfs filesystems exported.

/etc/exports:

```
/nfs   192.168.1.1/24(rw,sync,no_subtree_check,no_root_squash) 
/tftpboot 192.168.1.1/24(rw,sync,no_subtree_check,no_root_squash)
```

**Symptoms:**

Installer starts the booting process but fails to load the initramfs with error:

```
RAMDISK: gzip image found at block 0
RAMDISK: incomplete write (11927 != 32768)
write error
```
As already mentioned swapping start.elf and fixup.dat with any version older than 16/04/2020 makes the installer boot without errors. Older version of the installer (stable – SlackwareARM-14.2, firmware version from 29/02/2020) boots without any problem in similar setup.

Regards,
Alex

emsi, 2021-01-22:
Same here. Just slightly different numbers:
[image]

You can see the whole boot video:
https://youtu.be/wd_5ak4-XFs?t=84

timg236, 2022-11-28:
There's been some improvements to TFTP and initramfs over the last few months. Please can you retest with the latest version of the firmware in APT.