raspberrypi/firmware #1527: Pi4 hang on boot when a active USB hub is connected

maza70, 2021-01-19:
**Describe the bug**
Rpi4 hangs on reboot or shutdown when restarting with a active Usb Hub is connected.
It hangs also when RPI power cable is re plug and a active Usb Hub is connected.
If the power unplug from the hub all runs fine. 

**To reproduce**
Plugin a active Usb Hub with external power and after pulgin the RPI-power cable the RPI hangs and will not boot.. 

**Expected behaviour**
Boot when a powered active usb hub is connected

**Actual behaviour**
not booting when a powered active usb hub is connected

**System**
https://pastebin.com/uXg3jw58
*Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi4  4GB Ram

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage2

* Which firmware version (`vcgencmd version`)?
Jan 18 2021 17:42:39
Copyright (c) 2012 Broadcom
version e5e540537282a28aa8ec29d9662d65e06d0d2e24 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux RPI4 5.10.7-v7l+ #1395 SMP Mon Jan 18 16:17:57 GMT 2021 armv7l GNU/Linux

**Logs**
Not booting

**Additional context**

pelwell, 2021-01-19:
If you unplug the Pi 4 power supply with the hub connected, does the red LED go out completely?

pelwell, 2021-01-19:
And at some point you might want to say _which_ USB hub it is.

maza70, 2021-01-19:
> nplug the Pi 4 power supply with the hub conne

maza70, 2021-01-19:
sorry accidentally closed (wrong button under cursor :-)  )

maza70, 2021-01-19:
seems a similar bug to ticket "Pi4 wont boot with USB powered hub #1352" but with my current firmware i have this problem.

timg236, 2021-01-20:
If the red light goes off then the processor will not be running and is likely to be due to the USB hub back-powering the Pi i.e. an electrical problem

maza70, 2021-01-20:
> 
> 
> If the red light goes off then the processor will not be running and is likely to be due to the USB hub back-powering the Pi i.e. an electrical problem

This can be the problem,
but every active Hub is back powering or you know one that work? 
In My case i want access a 3,5" HD over a Usb-Hub with Sata interface.
On my Rpi3 all works fine it can be handle this with no Problem!

maza70, 2021-02-11:
I think there is no one that can be help with the software-bug or hardware-bug for a powered Usb-Hub. I had all Rpi's but that the Rpi4 has trouble with backpowered Usb-hubs makes me pensive. No i make a hardware solution with a low level switch (relay) that  power off the USB hub when shutdown and power on when boot, this works well.

the solution:
1.) v5 Relais like this
https://www.az-delivery.de/products/kf-301-relais-modul-mit-low-level-trigger?_pos=5&_sid=ea832a8ab&_ss=r

2.) a systemd service that switch the relais on when start and off wenn stop

I close now this bug while i have a solution by my own and don't think that anybody want fix this bug on Rpi4.
If anybody need support for my solution give me a message.

closed 2021-02-11 by maza70

h45h3ncryp73d, 2021-07-08:
> I think there is no one that can be help with the software-bug or hardware-bug for a powered Usb-Hub. I had all Rpi's but that the Rpi4 has trouble with backpowered Usb-hubs makes me pensive. No i [...]

Hi,

could you please share a bit more information - connecting diagram and systemd code/script - about how you solved this. I have a 16 port USB 3.0 hub I need to fix. Thank you in advance.

maza70, 2021-07-08:
> Hi,
> 
> could you please share a bit more information - connecting diagram and systemd code/script - about how you solved this. I have a 16 port USB 3.0 hub I need to fix. Thank you in advance.

See here https://github.com/maza70/RpiUsbSwitch

h45h3ncryp73d, 2021-07-08:
>  [data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAD8AAAA/CAYAAABXXxDfAAAABHNCSVQICAgIfAhkiAAAACZJREFUaIHtwQENAAAAwqD3T20PBxQAAAAAAAAAAAAAAAAAAAAnBj5DAAGK8nHUAAAAAElFTkSuQmCC###24x24:true###][Tracking image][https://github.com/notifications/beacon/AM36SQIR63RK2UXF5HO2SX3TWWOODA5CNFSM4WI7CDYKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOGQ6RMUQ.gif]

Thank you!

Napomena: Ova poruka može sadržavati nepovezane riječi kao posljedicu korištenja prediktivne tipkovnice.

Note: This message could contain words out of context caused by predictive keyboard.

PennyWafel, 2025-01-24:
I am sorry for resurrecting an old issue, but I did not know where else to report my findings. I too ran into backpowering issues on my Raspberry pi 4, when trying to set it up as a NAS with an usb hub and some externally powered 3.5" HDDs. 

Because it can be very difficult to discover which usb hubs do or do not backpower, I want to report here that there is NO backpowering on the official Raspberry Pi branded usb 3.0 hub released in 2024. The official site does not (currently) specify, but I have inquired about it and tested it to confirm there is indeed no backpowering going on.

Hopefully this will be of help to anyone in the future running into backfeeding of power issues.

maza70, 2025-01-26:
Hi @all,
This is not meant to be an advertisement, but I would like to pass on the information that the TP-Link UH720 active USB hub works with the Raspberry 4. The Hub is in my case backpowered and a HDD 2,5" with USB Serial-ATA is connected. 

Here are the versions of my Raspberry 4 Model B (2018)

**vcgencmd version**        

> Nov 26 2024 12:52:28
> Copyright (c) 2012 Broadcom
> version 2ae30f53898ae2f1ba77ff570a92991bedfb0398 (clean) (release) (start)
 

 **cat /etc/os-release** 

> "Debian GNU/Linux 12 (bookworm)

pelwell, 2025-01-26:
You need to be careful with the word "backpowered". Back-powering is when the hub sends power back through the port to the host - this is against the spec and causes problems for some Pis. A hub that has its own power supply and sends power to the attached devices is what we want.

maza70, 2025-01-26:
@pelwell You right, sorry this Hub TP-Link UH720 seams too power only the devices

pelwell, 2025-01-26:
Excellent - that's a useful recommendation then. (I've edited your comment)