raspberrypi/firmware #1529: kernel/initramfs cannot be loaded from subdirectory with same name as volume label

Hello71, 2021-01-29:
**Describe the bug**
Raspbian/Raspberry Pi OS distribute disk images with kernel and initramfs in boot filesystem root directory. Alpine Linux distributes a tarball with those files in a subdirectory named "boot". Many users over several years have reported issues booting Alpine Linux and say that they go away after either moving those files to the root directory (and updating config.txt) or reformatting the boot filesystem. I found that this is caused by the firmware failing to load kernel and initramfs from a subdirectory that has the same name as the volume label. This results in Alpine Linux failing to boot when unpacked into a Raspbian/Raspberry Pi OS boot filesystem, but booting successfully when a fresh filesystem is created with mkfs.fat using the default volume label "NO NAME".

**To reproduce**
1. Copy Raspberry Pi OS onto SD card using dd or similar.
2. Delete all files from boot filesystem and replace with Alpine Linux files.
3. Boot Raspberry Pi.
4. Set the volume label using fatlabel to "aaaa".
5. Boot Raspberry Pi.
6. Move "boot" to "aaaa" and update config.txt.
7. Boot Raspberry Pi.

**Expected behaviour**
Raspberry Pi boots Alpine Linux three times.

**Actual behaviour**
Raspberry Pi successfully boots only on the second try. On the first and third try, Raspberry Pi fails to boot and blinks LED 7 times, then pauses, then 7 times again, etc. Other users report similar display over HDMI, but I have not tested that.

If BOOT_UART=1 is set, the messages "Failed to load 'boot/initramfs-rpi' - initramfs disabled" and "No compatible kernel found" are printed to serial.

**System**
I am using a Raspberry Pi Zero W, but other users have reported the same or highly similar issues with several models of Raspberry Pi. I have tested Alpine Linux 3.12, containing raspberrypi-bootloader 1.20200902-r0, as well as Alpine Linux 3.13, containing raspberrypi-bootloader 1.20210108-r0.

**Logs**
I didn't save all BOOT_UART logs, but I recall seeing dtbs successfully loaded from boot subdirectory. There were some other messages such as HDMI EDID not found which were obviously explainable by no monitor connection.

**Additional context**
More Alpine-specific information is available at https://gitlab.alpinelinux.org/alpine/aports/-/issues/12368, including links to earlier (undiagnosed) reports of this issue.

pelwell, 2021-01-29:
Please post the contents of your config.txt for one of the above scenarios - the one with the subdirectory and volume labelled "boot", for example.

You might also want to experiment with using `os_prefix=boot/` and removing any explicit `kernel_file=` to see if that helps.

pelwell, 2021-01-29:
By the way, if you have a serial cable of some description that you can connect to a host computer (or another Pi), the config.txt settings `enable_uart=1` and `uart_2ndstage=1` will give you output from the firmware during the boot process.

Hello71, 2021-01-29:
> Please post the contents of your config.txt for one of the above scenarios - the one with the subdirectory and volume labelled "boot", for example.
> 
> You might also want to experiment with using `os_prefix=boot/` and removing any explicit `kernel_file=` to see if that helps.

For the Alpine Linux tests, I used the tarballs from https://alpinelinux.org/downloads/, specifically the armhf tarball for Raspberry Pi Zero W. You can find the config.txt in those. It is generated at https://gitlab.alpinelinux.org/alpine/aports/-/blob/master/scripts/mkimg.arm.sh.

Hello71, 2021-01-29:
> By the way, if you have a serial cable of some description that you can connect to a host computer (or another Pi), the config.txt settings `enable_uart=1` and `uart_2ndstage=1` will give you output from the firmware during the boot process.

This doesn't seem to provide any additional information.

Full boot output with BOOT_UART=1, enable_uart=1, uart_2ndstage=1:

```

Raspberry Pi Bootcode

Found SD card, config.txt = 1, start.elf = 1, recovery.elf = 0, timeout = 0
Read File: config.txt, 565 (bytes)

Raspberry Pi Bootcode
Read File: config.txt, 565
Read File: start.elf, 2940128 (bytes)
Read File: fixup.dat, 7301 (bytes)
MESS:00:00:01.325012:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.329326:0: brfs: File read: 565 bytes
MESS:00:00:01.336779:0: brfs: File read: /mfs/sd/usercfg.txt
MESS:00:00:01.341082:0: brfs: File read: 30 bytes
MESS:00:00:01.398156:0: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:01.404320:0: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:01.410656:0: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:01.416993:0: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:01.423330:0: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:01.429667:0: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:01.436005:0: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:01.442340:0: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:01.448677:0: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:01.455014:0: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:01.461109:0: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:01.472474:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.476614:0: brfs: File read: 565 bytes
MESS:00:00:01.484240:0: brfs: File read: /mfs/sd/usercfg.txt
MESS:00:00:01.488472:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:01.515113:0: brfs: File read: 30 bytes
MESS:00:00:01.695567:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:01.701392:0: *** Restart logging
MESS:00:00:01.707147:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:01.713839:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:01.720696:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:01.727553:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:01.734411:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:01.741269:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:01.748126:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:01.754984:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:01.761842:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:01.768699:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:01.775315:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:01.781220:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:01.789011:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:01.795870:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:01.802727:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:01.809585:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:01.816443:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:01.823300:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:01.830158:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:01.837016:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:01.843873:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:01.850488:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:01.856103:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:01.864848:0: HDMI0: hdmi_pixel_encoding: 162000000
MESS:00:00:01.870543:0: vec: vec_middleware_power_on: vec_base: 0x7e806000 rev-id 0x00002708 @ vec: 0x7e806100 @ 0x00000420 enc: 0x7e806060 @ 0x00000220 cgmsae: 0x7e80605c @ 0x00000000
MESS:00:00:01.888120:0: Failed to load 'boot/initramfs-rpi' - initramfs disabled
MESS:00:00:01.898214:0: dtb_file 'bcm2708-rpi-zero-w.dtb'
MESS:00:00:01.908391:0: brfs: File read: /mfs/sd/bcm2708-rpi-zero-w.dtb
MESS:00:00:01.913309:0: Loading 'bcm2708-rpi-zero-w.dtb' to 0x100 size 0x66dd
MESS:00:00:01.932715:0: brfs: File read: 26333 bytes
MESS:00:00:01.955981:0: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
MESS:00:00:02.022427:0: brfs: File read: 1523 bytes
MESS:00:00:02.027210:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:02.031281:0: brfs: File read: 565 bytes
MESS:00:00:02.038994:0: brfs: File read: /mfs/sd/usercfg.txt
MESS:00:00:02.048791:0: brfs: File read: 30 bytes
MESS:00:00:02.052507:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:02.057186:0: Read command line from file 'cmdline.txt':
MESS:00:00:02.063076:0: 'modules=loop,squashfs,sd-mod,usb-storage quiet console=tty1'
MESS:00:00:02.084713:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:03.660720:0: brfs: File read: 60 bytes
MESS:00:00:03.673970:0: No compatible kernel found
MESS:00:00:03.677040:0: Device tree loaded to 0x1bfe9500 (size 0x6aa7)
MESS:00:00:03.683316:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:08.490545:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
```

pelwell, 2021-02-05:
I had a quick look at this and it appears that the lower levels of the firmware filesystem support are intercepting a directory name that matches a volume name.

kovaszab, 2021-03-25:
+1
Anyone found a solution to boot alpine on RPi? I'm trying to get it work on a RPi 4 8gb. Utlimately if I succed I want to install netboot alpine but with the current state I won't even bother.

nathanvercaemert, 2021-06-15:
Just for clarification (sorry, I'm a newb here,) can this bug be avoided by creating the partition for Alpine as the only thing on the SD card? In other words, by following the following instructions: https://wiki.alpinelinux.org/wiki/Raspberry_Pi

inklesspen, 2022-01-16:
@nathanvercaemert the bug can be avoided by using `fatlabel` to rename the boot partition to something like `RPI`

rahmanshaber, 2022-10-23:
Any update on this?

lexfrei, 2023-09-14:
Any update on this?

macmpi, 2024-09-29:
> if user install RPiOS before and use something like Alpine, it will have a name conflict?

AFAIK, firmware bug (i.e. name conflict between partition and 1st level directory) is supposedly still there.
However **RPiOS images no longer use `boot` name as FAT partition** since [this](https://github.com/RPi-Distro/pi-gen/issues/669): so collision is now much less likely (unless individual does voluntarily name partition `boot` while formatting SD...).