raspberrypi/firmware #1530: GL_EXT_texture_filter_anisotropic extension not available

oomek, 2021-01-30:
**Describe the bug**
`GL_EXT_texture_filter_anisotropic` is not shown as an accessible extension. When I call it from an OpenGL 2.1 context initialized with DRM.
`glGetFloatv( GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT, &aniso );` returns 0
Is this a known issue? I couldn't find any information regarding that.

**System**

* Pi4
* Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage2
* Sep  2 2020 21:13:13
Copyright (c) 2012 Broadcom
version 4439d2aaa6c376a2d1ef4402f142e1cf4de37c43 (clean) (release) (start)
* Linux retropie4 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux

popcornmix, 2021-01-31:
This isn't the right place for this. You aren't using the firmware gl driver (which is good) but the mesa driver.
As such I'd suggest reporting here: https://docs.mesa3d.org/bugs.html 

But it's not clear to me whether this is a bug as it looks like an optional extension (that perhaps doesn't map well to the hardware).
But mesa devs will know better than us.

oomek, 2021-01-31:
There is no other way of getting OGL 2.1 than through mesa. Thanks, I'm gonna chase mesa regarding that 20 year old "extension" that should have been core a long time ago ;)

oomek, 2021-02-01:
Mesa devs said after the first glance that it can be ported. Closing the issue.

closed 2021-02-01 by oomek