raspberrypi/firmware #1532: Official PoE HAT Fan never turns on after upgrade to 5.10.y

ndanyluk, 2021-02-05:
**Describe the bug**
After standard `apt update && apt upgrade -y`, which upgrades the kernel to 5.10.y and all firmware, the Pi PoE HAT Fan never turns on again despite various different configuration steps.  The systems in question are all running stock Raspberry Pi OS 32-bit Lite

**To reproduce**
From last 2020 lite image:
```
apt update && apt upgrade -y
shutdown -r now
```

I also added the following line to config.txt with no success (same result)
```
dtoverlay=rpi-poe,poe_fan_temp0=40000,poe_fan_temp0_hyst=2000,poe_fan_temp1=45000,poe_fan_temp1_hyst=2000,poe_fan_temp2=50000,poe_fan_temp2_hyst=2000,poe_fan_temp3=55000,poe_fan_temp3_hyst=5000
```

**Expected behaviour**
The Pi boots and the PoE fan starts

**Actual behaviour**
The Pi boots, but the PoE fan does not start at all.

**System**
```
System Information
------------------

Raspberry Pi 3 Model B Plus Rev 1.3
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage2

Linux pi-leader 5.10.11-v7+ #1399 SMP Thu Jan 28 12:06:05 GMT 2021 armv7l GNU/Linux
Revision        : a020d3
Serial          : 000000000f963a2a
Model           : Raspberry Pi 3 Model B Plus Rev 1.3
Throttled flag  : throttled=0x80000
Camera          : supported=0 detected=0

Videocore information
---------------------

Jan 27 2021 22:27:11
Copyright (c) 2012 Broadcom
version 99d9a48302e4553cff3688692bb7e9ac760a03fa (clean) (release) (start_cd)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem                                       1K-blocks       Used  Available Use% Mounted on
192.168.0.2:/data/fastnas/nfsroot/pi-leader       52132864    2806784   49326080   6% /
devtmpfs                                            465280          0     465280   0% /dev
tmpfs                                               498560          0     498560   0% /dev/shm
tmpfs                                               498560       6932     491628   2% /run
tmpfs                                                 5120          4       5116   1% /run/lock
tmpfs                                               498560          0     498560   0% /sys/fs/cgroup
tmpfs                                                  100          8         92   8% /var/metrics/prom
192.168.0.2:/var/metrics/environment                  1024          0       1024   0% /var/metrics/environment
192.168.0.2:/data/fastnas/tftpboot/192.168.0.3    91286528     205824   91080704   1% /boot
192.168.0.2:/data/nas                           4647582720   39214080 4608368640   1% /data/nas
192.168.0.2:/data/nas/docker-data/media        10890474496 6282105856 4608368640  58% /data/nas/docker-data/media
/dev/dm-1                                         10255636      48856    9666108   1% /var/lib/docker/devicemapper/mnt/a815b99d5127306aa8322bdc8f2bd7707ce8a24ea964276262868ea62e57f247
/dev/dm-2                                         10255636     109544    9605420   2% /var/lib/docker/devicemapper/mnt/1fc5909a0d99ae7adaa03683c788f8aacf046f153b65860f07ee8607e9a2a341
/dev/dm-3                                         10255636     167052    9547912   2% /var/lib/docker/devicemapper/mnt/4154e92650db1d7a450970210a7f321b34dfc6ef84e0f45f0dfa1bae0cea9076
[... 773 lines cut ...]
[   50.225635] docker_gwbridge: port 5(veth909ad26) entered disabled state
[   50.226401] device veth909ad26 entered promiscuous mode
[   50.227618] docker_gwbridge: port 5(veth909ad26) entered blocking state
[   50.227644] docker_gwbridge: port 5(veth909ad26) entered forwarding state
[   50.296195] docker_gwbridge: port 5(veth909ad26) entered disabled state
[   50.296409] IPv6: ADDRCONF(NETDEV_CHANGE): vethabec150: link becomes ready
[   50.296586] br0: port 5(veth3) entered blocking state
[   50.296595] br0: port 5(veth3) entered forwarding state
[   50.489730] IPv6: ADDRCONF(NETDEV_CHANGE): veth909ad26: link becomes ready
[   50.489806] docker_gwbridge: port 5(veth909ad26) entered blocking state
[   50.489815] docker_gwbridge: port 5(veth909ad26) entered forwarding state
[   52.671919] br0: port 5(veth3) entered disabled state
[   52.673362] eth0: renamed from vethabec150
[   52.744452] br0: port 5(veth3) entered blocking state
[   52.744468] br0: port 5(veth3) entered forwarding state
[   52.857102] docker_gwbridge: port 5(veth909ad26) entered disabled state
[   52.858109] eth1: renamed from veth6a162ae
[   52.913946] docker_gwbridge: port 5(veth909ad26) entered blocking state
[   52.913963] docker_gwbridge: port 5(veth909ad26) entered forwarding state
```

**Additional context**
This issue is reproducible on all of my Pi 3B+ units (3x bought from different retailers at different times) and a Pi 4B unit.  All 4 devices have the official PoE HAT and were working perfectly until earlier today when I upgraded the kernel.  I didn't realize the fans stopped until I upgraded and rebooted the last Pi.

Additionally, all 4 are running stock Raspberry Pi OS 32-bit lite and no modifications were made to the config.txt directly (only through `raspi-config`).

ndanyluk, 2021-02-05:
Also appears related to https://github.com/raspberrypi/firmware/issues/1531, but opposite (their fans don't turn off).

Is it possible that the kernel upgrade or bootloader upgrade broke the fan dtoverlay and they retain their last known settings?

pelwell, 2021-02-05:
Your dtoverlay line is too long. Use `sudo vcdbg log msg` to see the warnings from the firmware.

Parameters to an overlay can be split over multiple lines by starting the continuations with `dtparam=`. The current overlay remains "in scope" until the next `dtoverlay` line.

ndanyluk, 2021-02-05:
> Your dtoverlay line is too long. Use `sudo vcdbg log msg` to see the warnings from the firmware.
> 
> Parameters to an overlay can be split over multiple lines by starting the continuations with `dtparam=`. The current overlay remains "in scope" until the next `dtoverlay` line.

Fair enough, I'll reformat that and see what happens.  Just to be clear though, the problem originated before adding anything to config.txt - that was only a troubleshooting step.  Previously the fan "just worked" without any manual configuration required

pelwell, 2021-02-05:
Since writing that I have reproduced the problem, and am now confused.

ndanyluk, 2021-02-05:
> 
> 
> Since writing that I have reproduced the problem, and am now confused.

I tried formatting my config.txt like you suggested but same vcdbg log msg output.  There's some [speculation](https://github.com/raspberrypi/firmware/issues/1531#issuecomment-774090858) that a particular commit might have caused the issue

pelwell, 2021-02-05:
I'm closing this issue as a duplicate of #1531.

pelwell, 2021-02-08:
This isn't a duplicate after all, but it has been diagnosed and fixed in the firmware development repo. Expect a fix to be released in the next few days.

pelwell, 2021-02-08:
rpi-update contains the potential fix. Please test and report.

ndanyluk, 2021-02-08:
> 
> 
> rpi-update contains the potential fix. Please test and report.

That seems to have worked!  Will report if any strange things happen

ricktendo, 2021-02-08:
Working again!

bondskin, 2021-02-10:
Working here, too, thanks.

closed 2021-02-11 by pelwell