raspberrypi/firmware #1533: Reading from usb_pd/ain1 not updating (CM4)

markus-k, 2021-02-05:
**Is this the right place for my bug report?**
I think this should be a firmware issue, not userland issue. Please correct me if I'm wrong.

**Describe the bug**
I'm using the AIN0 and AIN1 pins on the CM4 for reading battery and supply voltage. This generally works, but I have noticed that the reading won't update as the applied voltage changes. The voltage only seems to be sampled once at boot time.

Commands used to read AIN0 and AIN1:
```
vcgencmd measure_volts usb_pd
vcgencmd measure_volts ain1
```

**To reproduce**
Apply a voltage (say, 1.5V) to AIN1, boot, read it using `vcgencmd measure_volts ain1`. Change voltage, e.g. to 2V, read again. `vcgencmd measure_volts ain1` now reads the old voltage again.

**Expected behaviour**
The voltage should be updated with every read, or at least in certain intervals.

**Actual behaviour**
The voltage is only sampled once at boot.

**System**
Model: Raspberry Pi CM4
OS: Raspbian Lite
Firmware version: version 194a85abd768c7334bbadc3f1911c10a7d18ed14 (clean) (release) (start_x)
Kernel version: 5.4.83-v7l+ 1379

pelwell, 2021-02-05:
Those commands were added as an experiment and are not supported.

closed 2021-02-05 by pelwell

markus-k, 2021-02-05:
Is there any official way of reading the AIN0 and AIN1 pins then?

pelwell, 2021-02-05:
Not that I know of - they are firmware-controlled resources, not for public consumption, unless you know otherwise?

markus-k, 2021-02-05:
The CM4 datasheet suggests they can be used as analog inputs, although it's not explicitly stated: "The MXL7704 datasheet should be consulted if these pins are to be used. Onboard filtering is provided by a 100nF capacitor to ground for each signal."

Since the PMIC is not accessible from the ARM, at least when using two cameras to my knowledge, manually reading from I2C is not an option either (and is controlled by firmware anyway as you say).

pelwell, 2021-02-05:
Actually there is a way - equally unsupported, with no warranty, but one that might work:
```
vcgencmd pmicrd 1c                 # AIN
vcgencmd pmicrd 1d                 # USB PD
```