raspberrypi/firmware #1534: Moto M4 Soundcard USB Disconnect Issue

DrCWO, 2021-02-07:
Hi guys,
I like to give you a pointer to the issue with same name I placed in the Linux category.

https://github.com/raspberrypi/linux/issues/4116

Sorry for duplicate posting. At the beginning I thought it is a Linux issue but after some tests described in the thread above I now believe it is an issue in the USB controller firmware of the Pi4.

On my Pi4 I installed the latest boot rom firmware from here https://github.com/raspberrypi/rpi-eeprom/releases. All tests i took were done with this firmware. 

If needed I could provide the Motu M4 audio interface...
I hope for a successful patch :-)

Best
DrCWO

popcornmix, 2021-02-08:
This repo is for reporting bugs with the gpu firmware. The GPU is not involved in USB when linux is runnng.
It's not used for reporting issues with firmware used as a part of linux.
I don't believe this is the right location for this issue (the linux issue is more appropriate).

DrCWO, 2021-02-11:
I found a solution to the issue.
Motu M4 is equipped with a USB-C connector and is shipped with a standard USB-C to USB-A cable. Not a USB-3 type!
I got a real USB3 cable with an extendes USB-A connector and a USB-C connector at the other side.
Using this cable I got no more disconnects :-)

closed 2021-02-11 by DrCWO