raspberrypi/firmware #1536: Newer firmware loading incorrect dtb on custom RPi?

acostach, 2021-02-09:
I'm testing a custom board based on the Pi3 Model B, with kernel 5.4.83 and the latest rpi-firmware, and bluetooth doesn't work, hciattach times out. This doesn't happen with the standard Pi3.

However, if I switch only the firmware to the old version 20180817 and nothing else, bluetooth works fine.

The output of wc -c /proc/device-tree/soc/gpio@7e200000/uart0_pins/brcm\,pins | cut -f 1 -d ' ' is 8 with the old firmware, also 8 on a standard Pi3 (regardless of firmware version), but 16 with the latest firmware.

I guess the old firmware detects that this is not a standard Pi 3 and loads the the device-tree named bcm2710-rpi-custom.dtb, which is just a copy of the standard bcm2710-rpi-3-b.dtb. Is it possible that newer firmware loads a different dtb?

Asking because with the new firmware I don't see the hci_uart kernel module being loaded.

If this is the case, should the bcm2710-rpi-3-b.dtb be copied to a file that has a different naming convention?

Thank you

Dark-Sky, 2021-02-09:
I am having the same issue of bluetooth not working with the latest firmware and had to revert back to firmware dated 20210128 I had previously installed for bluetooth to work

pelwell, 2021-02-09:
Have you wired up the flow controls signals to the Bluetooth modem? It sounds like you might not have.

Because of the varied history of flow control support to the Bluetooth modem, and to avoid splitting the DTB according to whether or not the current board supports it, the DTBs assume no flow control and the firmware sets it where it is supported.
Custom boards are an unknown quantity to the firmware, other than being based on a 3B, so as the result of a coin toss (OK, arbitrary choice) they are treated as if they had flow control (because who wouldn't want it).

If you want to opt out of this scheme, edit your .dts file to make a copy of the uart0_pins node with a different name and point the uart0 node at it.

pelwell, 2021-02-09:
@Dark-Sky Unless you are using a custom board then your comment doesn't belong here.

Dark-Sky, 2021-02-09:
Was thinking the 2 issues could be related. My bluetooth worked well before the latest firmware upgrade.

pelwell, 2021-02-09:
The only change vaguely UART-related - at least on the firmware side - in the last week or so is the removal of the code that automatically converts an old cmdline.txt reference to ttyAMA0 to `ttyS0` in the event that `ttyS0` is the primary (console) UART. All new images for years have used "serial0" as an alias, and now all `cmdline.txt`s have to either do the same or get it write and use `ttyS0`.

acostach, 2021-02-10:
Thank you very much @pelwell , it's indeed as you described and opting out solved the problem.

closed 2021-02-10 by acostach