raspberrypi/firmware #1537: No overclocking above 2500 MHz possible

cr-chsn1, 2021-02-18:
**Describe the bug**
I try to overclock the RPi4B above 2500 MHz and it seems to have a lock. Anything higher than 2500 MHz results in a fixed clock at 600 MHz on reboot. The performance of the RPi seems to be on the 600 MHz level.
The frequency is read with `vcgencmd measure_clock arm`.
Output: `frequency(48)=600169920`

**To reproduce**
Set `arm_freq` in `/boot/config.txt` on 2501 (or above).

**Expected behaviour**
As for the range from 1500 MHz to 2500 MHz any clock higher than 2500 MHz shouldn't be a problem.

**Actual behaviour**
Every clock higher than 2500 MHz (2501 or above) will result in a fixed clock at 600 MHz on the next boot.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi?
Raspberry Pi 4 Model B Rev. 1.1

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2021-01-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 21090519d85bdaa1615d5d5057d37b09368ea5d2, stage2

* Which firmware version (`vcgencmd version`)?
Feb 16 2021 14:17:48
Copyright (c) 2012 Broadcom
version d6d82cf99bcb3e9a166a34cfde53130957a36bd3 (clean) (release) (start_cd)

* Which kernel version (`uname -a`)?
Linux raspberrypi 5.10.16-v7l+ #1402 SMP Tue Feb 16 14:12:18 GMT 2021 armv7l GNU/Linux

popcornmix, 2021-02-19:
Yes, we define PLLB (which the arm typically uses with a divide by 2) as maxing a max frequency of 5GHz.
But do you really have a Pi4  hat is stable at 2500MHz? When running under load?
That is far higher than any I have heard of.

cr-chsn1, 2021-02-19:
I have a RPi4B stable at 2.5 GHz under load.
I can't upload a screenshot, but here you can see the setup: https://hwbot.org/submission/4674564_cr_chsn1_hwbot_prime_bcm2711_4806.5_pps

On air-cooling 2.3-2.4 GHz is do-able. With the BCM2711**C0** clocks above 2.5 GHz should be relatively easy, especially with slightly better cooling.

Do you plan to unlock this barrier of "5 GHz (divided by 2)"-barrier? 
Does this barrier fulfill any purpose which is needed so the RPi works the way it works or can you unlock it without much hassle?
No lock at all, would be very usefull for future projects.

btw: Does the RPi3B(+) have a max. frequency lock so it cannot reach clocks above 1600 MHz? 1601 MHz or higher result in no boot at all. On every of my RPi3B(+). Looks like a similar "problem".

popcornmix, 2021-02-19:
> btw: Does the RPi3B(+) have a max. frequency lock so it cannot reach clocks above 1600 MHz?

Yes it does.

tester962, 2021-02-24:
@popcornmix It would be great if the PLLB clock could be changed by the user (via config.txt perhaps?)
I'm interested in overclocking above 2500 MHz too.

arrowgent, 2021-05-08:
im curious what you consider stable

stable for 2 minutes while running all cores
or running for 180days at idle

my raspberry pi 4b is not stable at 2000mhz
after 3,4 days of idling it becomes unresponsive
set speed to 1900mhz and all is well long term
temperature is never an issue 30*C idle 45*C full make -j$4

im curious what you think you need more than 2000mhz for on a raspberry?
bragging rights?  benchmark scores? a specific software application that you want to run faster?

cr-chsn1, 2021-05-08:
I consider stable as stable under full load for many hours.
It's OK not every RPi is capable of going 2.5 GHz and above, it's silicon lottery like everywhere else in the computer industry.

Why do I need more than 2.0 GHz? I like to test hardware under extreme cicumstances. You learn things, when you do something new.
For my daily usage my stock-clocked RPi4B is more than enough to host my services.

But do you decide what kind options are available, only because you don't need them and don't understand others maybe have an interest in something else?
And why going ad homiem? Bragging rights?

If there is a special need for this software limitation of the PLLB, I'm fully OK with that. I'll wait until this limit is raised, like in the past.
But if this is an arbitrary limit, it would be great to release the full potential of the SoC.

The warranty bit isn't set any more since the RPi3B+, maybe enabling it again, when the user disables the PLLB divider or goes above a certain clock speed, is a way to deal with possible warranty problems.