raspberrypi/firmware #1539: raspberrypi-bootloader 20210216-1 breaks Bluetooth

alecov, 2021-02-22:
As per the title. After an upgrade from 20210205-1 to 20210216-1, Bluetooth stopped working.

The following appears in the kernel log:

```
Bluetooth: hci0: hardware error 0x00
```

Rolling back to 20210216-1 fixes the issue.

Running ArchLinuxARM kernel 5.10.16-3-ARCH (armv7l, 32-bit) on a Raspberry Pi 4B (4GB RAM).

pelwell, 2021-02-23:
I don't speak Arch version numbers - please provide `uname -a` and `vcgencmd version` for both versions.

alecov, 2021-02-23:
@pelwell Apologies for the confusion with the version numbers — I assumed they related to release tags but apparently they don't (these tags don't exist).

Anyway, the last working version is:
```
Jan 27 2021 22:19:57
Copyright (c) 2012 Broadcom
version 99d9a48302e4553cff3688692bb7e9ac760a03fa (clean) (release) (start)
```

And the first non-working version is:
```
Feb 16 2021 14:17:31
Copyright (c) 2012 Broadcom
version d6d82cf99bcb3e9a166a34cfde53130957a36bd3 (clean) (release) (start)
```

The following messages occasionally appear in the logs as well (with the non-working version):
```
Bluetooth: hci0: command 0x0c03 tx timeout
Bluetooth: hci0: BCM: Reset failed (-110)
```

The uname is:
```
Linux rasp 5.10.17-1-ARCH #1 SMP Mon Feb 22 13:33:42 UTC 2021 armv7l GNU/Linux
```

I've tested with some previous kernel versions, it seems the problem is indeed related to raspberrypi-firmware.

pelwell, 2021-02-23:
What is in your cmdline.txt?

pelwell, 2021-02-23:
You might as well post config.txt as well, in case there are any surprises.

alecov, 2021-02-23:
cmdline.txt
```
root=/dev/mmcblk0p2 rw rootwait console=ttyAMA0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=ttyAMA0,115200 elevator=noop quiet audit=0
```

config.txt
```
dtparam=audio=on
dtoverlay=vc4-kms-v3d-pi4
initramfs initramfs-linux.img followkernel
```

FWIW: there's also this btattach service for enabling Bluetooth:

`/usr/bin/btattach -B/dev/ttyAMA0 -Pbcm -S115200`

pelwell, 2021-02-23:
You shouldn't be putting `console=ttyAMA0,115200` in your cmdline.txt - ttyAMA0 is used to communicate with the Bluetooth modem. Change it to `console=serial0,115200` instead, which the firmware will translate into whichever is the non-Bluetooth UART. The same goes for `kgdboc=ttyAMA0,115200`, further along.

Between the two firmware versions you tried the translation in the firmware put there to correct pre-Pi3 command lines had been removed since _surely_ everybody had got the message by now.

alecov, 2021-02-23:
It worked, thanks.

> Between the two firmware versions you tried the translation in the firmware put there to correct pre-Pi3 command lines had been removed since surely everybody had got the message by now.

Not the ArchLinuxARM folks — the above is pretty much the installation's default.

Closing.

closed 2021-02-23 by alecov