raspberrypi/firmware #1541: Cirrus Logic Audio Card: Recording stutters

choff, 2021-02-23:
I have a Cirrus Logic Audio Card connected to a Raspberry Pi 4 and am facing a really strange issue since my last dist-upgrade: sometimes, recording audio from the sound card stutters and there are "gaps" in the recording. Also "arecord" reports overflow errors:
```
 $ arecord 1kHz-broken.wav
Recording WAVE '1kHz-broken.wav' : Unsigned 8 bit, Rate 8000 Hz, Mono
overrun!!! (at least 0.086 ms long)
overrun!!! (at least 0.111 ms long)
overrun!!! (at least 0.102 ms long)
overrun!!! (at least 0.108 ms long)
overrun!!! (at least 0.101 ms long)
overrun!!! (at least 0.115 ms long)
```
I have connected the 1 kHz tone generator of the Cirrus Logic Audio Card to the audio input path. The resulting (broken) recording is [provided here](https://github.com/Hexxeh/rpi-firmware/files/6025035/recordings.zip). You cannot hear the "gaps" during the recording as silence as "arecord" just seems to "jump" over the gaps. But what you can hear is that the 1 kHz audio is distorted. Th "gaps" in the recording (which unfortunately you cannot hear) are so bad that I cannot be understood by anyone when I am in a phone call. A good recording that doesn't show these issues is also in the same ZIP archive as the broken recording.

I have nailed this issue down to [this commit](https://github.com/Hexxeh/rpi-firmware/commit/43022f5adbbf5d6e05ea0e022a3090c2c9feff7c): "firmware: dmalib: Allow sdcard to borrow channel 6". Every firmware version before this commit works well. Starting from this commit, every firmware version (even the most recent one) shows this issue. I did a lot of testing of older firmware revisions with "sudo rpi-update [commit ID]", so this is why I am so sure. I also tested the same thing with another Cirrus Logic Audio Card and this one behaves in the same way. So this cannot be a software issue.

I also know that it cannot be a kernel issue because recording works well with the commit [453e49bdd87325369b462b40e809d5f3187df21d](https://github.com/Hexxeh/rpi-firmware/commit/453e49bdd87325369b462b40e809d5f3187df21d), which uses the exact same kernel as is used by the commit ["firmware: dmalib: Allow sdcard to borrow channel 6"](https://github.com/Hexxeh/rpi-firmware/commit/43022f5adbbf5d6e05ea0e022a3090c2c9feff7c). Only firmware files were changed between these two commits. The kernel version (as reported by "uname -a") in both of these commits is "5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020".

My test vehicle is a Raspberry Pi 4 without SD card (it has an external SSD connected via USB). I don't know if the Raspberry Pi 3 and other models are also affected, but I do have a Raspberry Pi 3, so I could give it a try.

If the issue happens and my colleagues complain about the stuttering, it often disappears after a reboot. Likewise, a reboot can also cause the stuttering to appear if it wasn't there before the reboot. So, the stuttering is a sporadic issue. It may happen or it may not, but if it happens then it keeps on happening in every recording until you reboot the system. And if it works, all recordings are fine until you reboot the system.

Do you have any clue how this change "firmware: dmalib: Allow sdcard to borrow channel 6" could cause this kind of failure? Do you have any ideas what I could still try out?

Thank you.

pelwell, 2021-02-23:
How many monitors are connected? And what does `aplay -l` report?

HiassofT, 2021-02-23:
OK, I could now reproduce this. In the working case the I2S got dma0chan6 for rx which should be DMA channel 8 if I got the brcm,dma-channel-mask calculation right
```
LibreELEC:/sys/kernel/debug/dmaengine # cat summary 
dma0 (fe007000.dma): number of channels: 9
 dma0chan0    | fef00700.hdmi:audio-rx
 dma0chan1    | fef05700.hdmi:audio-rx
 dma0chan2    | fe300000.mmcnr:rx-tx
 dma0chan3    | fe204000.spi:tx
 dma0chan4    | fe204000.spi:rx
 dma0chan5    | fe203000.i2s:tx
 dma0chan6    | fe203000.i2s:rx

dma1 (fe007b00.dma): number of channels: 2
```

In the non-working case I2S got dma0chan4 - which should be the DMA channel 6
```
# cat /sys/kernel/debug/dmaengine/summary 
dma0 (fe007000.dma): number of channels: 9
 dma0chan0    | fef00700.hdmi:audio-rx
 dma0chan1    | fef05700.hdmi:audio-rx
 dma0chan2    | fe300000.mmcnr:rx-tx
 dma0chan3    | fe203000.i2s:tx
 dma0chan4    | fe203000.i2s:rx
 dma0chan5    | fe204000.spi:tx
 dma0chan6    | fe204000.spi:rx

dma1 (fe007b00.dma): number of channels: 2
```

I saw overruns too and recorded file was only a few k
```
LibreELEC:~/audiotest # ./rec-cirrus.sh -f cd /tmp/rec.wav
Recording WAVE '/tmp/rec.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
overrun!!! (at least 0.073 ms long)
overrun!!! (at least 0.058 ms long)
^CAborted by signal Interrupt...

LibreELEC:~/audiotest # ls -la /tmp/rec.wav 
-rw-r--r--    1 root     root         22096 Feb 23 14:15 /tmp/rec.wav
```
So, yes, there seems to be some correlation with the firmware change you mentioned.

BTW: I think this might also be reproducible by using a "dummy" I2S card using spdif receiver/transmitter

choff, 2021-02-23:
Thanks a lot @HiassofT! You have been an invaluable help and it is good to know that not only I am seeing this issue.

@pelwell , just one monitor is connected (via HDMI). "aplay -l" reports the following:
```
**** List of PLAYBACK Hardware Devices ****
card 0: RPiCirrus [RPi-Cirrus], device 0: WM5102 AiFi wm5102-aif1-0 [WM5102 AiFi wm5102-aif1-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 2: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
```

HiassofT, 2021-02-23:
Seems I can easily reproduce this with the attached dual-generic overlay. Just added it to config.txt and also enabled spi so it has a chance to grab dma0chan3/4.

When I2S got dma0chan5/6 everything's fine and I also see IRQ 49 counting up:
```
LibreELEC:~/audiotest # cat /sys/kernel/debug/dmaengine/summary
dma0 (fe007000.dma): number of channels: 9
 dma0chan0    | fef00700.hdmi:audio-rx
 dma0chan1    | fef05700.hdmi:audio-rx
 dma0chan2    | fe300000.mmcnr:rx-tx
 dma0chan3    | fe204000.spi:tx
 dma0chan4    | fe204000.spi:rx
 dma0chan5    | fe203000.i2s:tx
 dma0chan6    | fe203000.i2s:rx

dma1 (fe007b00.dma): number of channels: 2
LibreELEC:~/audiotest # grep DMA /proc/interrupts
 42:       1369          0          0          0     GICv2 112 Level     DMA IRQ
 44:          0          0          0          0     GICv2 114 Level     DMA IRQ
 46:        345          0          0          0     GICv2 116 Level     DMA IRQ
 47:          0          0          0          0     GICv2 117 Level     DMA IRQ
 48:          0          0          0          0     GICv2 118 Level     DMA IRQ
 49:        511          0          0          0     GICv2 119 Level     DMA IRQ, DMA IRQ
LibreELEC:~/audiotest # ./arecord -D hw:Dual,DEV=1 -f cd -d 20 /tmp/rec.wav
Recording WAVE '/tmp/rec.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
LibreELEC:~/audiotest # ls -l /tmp/rec.wav
-rw-r--r--    1 root     root       3528044 Feb 23 15:19 /tmp/rec.wav
LibreELEC:~/audiotest # grep DMA /proc/interrupts
 42:       1369          0          0          0     GICv2 112 Level     DMA IRQ
 44:          0          0          0          0     GICv2 114 Level     DMA IRQ
 46:        345          0          0          0     GICv2 116 Level     DMA IRQ
 47:          0          0          0          0     GICv2 117 Level     DMA IRQ
 48:          0          0          0          0     GICv2 118 Level     DMA IRQ
 49:        671          0          0          0     GICv2 119 Level     DMA IRQ, DMA IRQ
```
But when I2S got dma0chan3/4 record fails - looks like we are not getting any rx DMA IRQs
```
LibreELEC:~/audiotest # cat /sys/kernel/debug/dmaengine/summary
dma0 (fe007000.dma): number of channels: 9
 dma0chan0    | fef00700.hdmi:audio-rx
 dma0chan1    | fef05700.hdmi:audio-rx
 dma0chan2    | fe300000.mmcnr:rx-tx
 dma0chan3    | fe203000.i2s:tx
 dma0chan4    | fe203000.i2s:rx
 dma0chan5    | fe204000.spi:tx
 dma0chan6    | fe204000.spi:rx

dma1 (fe007b00.dma): number of channels: 2
LibreELEC:~/audiotest # grep DMA /proc/interrupts
 42:          0          0          0          0     GICv2 112 Level     DMA IRQ
 44:          0          0          0          0     GICv2 114 Level     DMA IRQ
 46:        345          0          0          0     GICv2 116 Level     DMA IRQ
 47:          0          0          0          0     GICv2 117 Level     DMA IRQ
 48:          0          0          0          0     GICv2 118 Level     DMA IRQ
 49:          0          0          0          0     GICv2 119 Level     DMA IRQ, DMA IRQ
LibreELEC:~/audiotest # ./arecord -D hw:Dual,DEV=1 -f cd -d 20 /tmp/rec.wav
Recording WAVE '/tmp/rec.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
arecord: pcm_read:2152: read error: Input/output error
LibreELEC:~/audiotest # ls -l /tmp/rec.wav
-rw-r--r--    1 root     root            44 Feb 23 15:27 /tmp/rec.wav
LibreELEC:~/audiotest # grep DMA /proc/interrupts
 42:          0          0          0          0     GICv2 112 Level     DMA IRQ
 44:          0          0          0          0     GICv2 114 Level     DMA IRQ
 46:        345          0          0          0     GICv2 116 Level     DMA IRQ
 47:          0          0          0          0     GICv2 117 Level     DMA IRQ
 48:          0          0          0          0     GICv2 118 Level     DMA IRQ
 49:          0          0          0          0     GICv2 119 Level     DMA IRQ, DMA IRQ
```
[dual-generic-overlay.dts.txt](https://github.com/raspberrypi/firmware/files/6029904/dual-generic-overlay.dts.txt)

HiassofT, 2021-02-23:
I can also confirm that Dec 14 firmware is fine and that Dec 15 breaks it - the DMA IRQ 47 went AWOL - all running on kernel 5.10.16, only replaced start4x/fixup4x firmware files.

```
LibreELEC:~/audiotest # vcgencmd version
Dec 14 2020 13:17:13
Copyright (c) 2012 Broadcom
version 54a311a9f425370230dbc493745c974762e344d0 (clean) (release) (start_x)
LibreELEC:~/audiotest # cat /sys/kernel/debug/dmaengine/summary
dma0 (fe007000.dma): number of channels: 9
 dma0chan0    | fef00700.hdmi:audio-rx
 dma0chan1    | fef05700.hdmi:audio-rx
 dma0chan2    | fe300000.mmcnr:rx-tx
 dma0chan3    | fe203000.i2s:tx
 dma0chan4    | fe203000.i2s:rx

dma1 (fe007b00.dma): number of channels: 2
LibreELEC:~/audiotest # grep DMA /proc/interrupts
 41:          0          0          0          0     GICv2 112 Level     DMA IRQ
 43:          0          0          0          0     GICv2 114 Level     DMA IRQ
 45:        345          0          0          0     GICv2 116 Level     DMA IRQ
 46:          0          0          0          0     GICv2 117 Level     DMA IRQ
 47:          0          0          0          0     GICv2 118 Level     DMA IRQ
LibreELEC:~/audiotest # ./arecord -D hw:Dual,DEV=1 -f cd -d 20 /tmp/rec.wav
Recording WAVE '/tmp/rec.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
LibreELEC:~/audiotest # ls -l /tmp/rec.wav
-rw-r--r--    1 root     root       3528044 Feb 23 17:24 /tmp/rec.wav
LibreELEC:~/audiotest # grep DMA /proc/interrupts
 41:          0          0          0          0     GICv2 112 Level     DMA IRQ
 43:          0          0          0          0     GICv2 114 Level     DMA IRQ
 45:        345          0          0          0     GICv2 116 Level     DMA IRQ
 46:          0          0          0          0     GICv2 117 Level     DMA IRQ
 47:        160          0          0          0     GICv2 118 Level     DMA IRQ
```

```
LibreELEC:~/audiotest # vcgencmd version
Dec 15 2020 14:49:28
Copyright (c) 2012 Broadcom
version 4b4aff21f72c5b9ba39d83c7b0f8fa910a6ef99b (clean) (release) (start_x)
LibreELEC:~/audiotest # cat /sys/kernel/debug/dmaengine/summary
dma0 (fe007000.dma): number of channels: 9
 dma0chan0    | fef00700.hdmi:audio-rx
 dma0chan1    | fef05700.hdmi:audio-rx
 dma0chan2    | fe300000.mmcnr:rx-tx
 dma0chan3    | fe203000.i2s:tx
 dma0chan4    | fe203000.i2s:rx

dma1 (fe007b00.dma): number of channels: 2
LibreELEC:~/audiotest # grep DMA /proc/interrupts
 41:          0          0          0          0     GICv2 112 Level     DMA IRQ
 43:          0          0          0          0     GICv2 114 Level     DMA IRQ
 45:        345          0          0          0     GICv2 116 Level     DMA IRQ
 46:          0          0          0          0     GICv2 117 Level     DMA IRQ
 47:          0          0          0          0     GICv2 118 Level     DMA IRQ
LibreELEC:~/audiotest # ./arecord -D hw:Dual,DEV=1 -f cd -d 20 /tmp/rec.wav
Recording WAVE '/tmp/rec.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
arecord: pcm_read:2152: read error: Input/output error
LibreELEC:~/audiotest # ls -l /tmp/rec.wav
-rw-r--r--    1 root     root         22096 Feb 23 17:26 /tmp/rec.wav
LibreELEC:~/audiotest # grep DMA /proc/interrupts
 41:          0          0          0          0     GICv2 112 Level     DMA IRQ
 43:          0          0          0          0     GICv2 114 Level     DMA IRQ
 45:        345          0          0          0     GICv2 116 Level     DMA IRQ
 46:          0          0          0          0     GICv2 117 Level     DMA IRQ
 47:          1          0          0          0     GICv2 118 Level     DMA IRQ
```

choff, 2021-02-23:
Thanks a lot @HiassofT for your investigation!

@popcornmix, I think you wrote the firmware change ["firmware: dmalib: Allow sdcard to borrow channel 6"](https://github.com/Hexxeh/rpi-firmware/commit/43022f5adbbf5d6e05ea0e022a3090c2c9feff7c) that - as both Hias and myself have verified - breaks recording over I2S. Do you have any idea why recording no longer works properly with this firmware version?

popcornmix, 2021-02-24:
I just built that firmware. The code change was @pelwell's.

pelwell, 2021-02-24:
Thanks for that overlay, @HiassofT - I can reproduce the problem with `vc4-kms-v3d` (I was previously using `vc4-fkms-v3d`, which requires the VPU to have more DMA channels). Rolling the firmware back to `453e49b` fixes this particular case, but breaks others.

My guess is that channel 6 is not being completely released by the firmware.

pelwell, 2021-02-24:
> My guess is that channel 6 is not being completely released by the firmware.

... which turned out to be correct. A patch to fix this is up for review internally.

choff, 2021-02-24:
Great, thanks a lot @pelwell !

popcornmix, 2021-02-25:
rpi-update firmware contains the potential fix.

HiassofT, 2021-02-25:
thanks a lot, quick tests with the "dual-generic" overlay and Cirrus Logic Audio Card look good with the updated firmware!

choff, 2021-02-25:
Indeed, I am also no longer able to reproduce the issue. @HiassofT, my special thanks goes to you for your excellent support in diagnosing this issue! And thanks also to you, Phil, for the final fix.

Best regards,
   Christian

closed 2021-02-25 by choff