raspberrypi/firmware #1543: h264_freq and isp_freq default behaviour

Consti10, 2021-03-01:
**Is this the right place for my bug report?**
I think yes. raspivid should probably set the isp and core frequency according to the level requested by the user

**Describe the bug**
The documentation https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md states that
h264_freq_min is 500mhz on the Rpi4.
However, no matter what resolution/framerate I use, the core frequency never exceeds ~200mhz when measured using
`vcgencmd measure_clock h264`

This can result in frame drops even at level 4.0. 

**To reproduce**
run raspivid and then measure h264 freq using `vcgencmd measure_clock h264`

**Expected behaviour**
Clock at 500mhz or at least 360 on CM4 io board. For older pi models, It looks as if the same issue exists, but obviously the max frequency changes per model.

**Workaround**
Manually set h264_freq_min=500 (or equivalent for pi other than CM4) in /boot/config.txt. When doing so, the clock speed is 0 when core is not in use, and 500 when core is in use.

**System**
Rpi CM4 IO board, also Rpi 3B+, latest raspbian.

**Additional context**
https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=303030&p=1829016#p1829016

pelwell, 2021-03-01:
This is a property of the firmware, not the userland, hence the transfer.

You'll need to provide the output of "vcgencmd version" and "uname -a" so we can establish how current your image is.

Consti10, 2021-03-01:
```
vcgencmd version
Jan 27 2021 22:20:57 
Copyright (c) 2012 Broadcom
version 99d9a48302e4553cff3688692bb7e9ac760a03fa (clean) (release) (start_x)
```

```
uname -a
Linux raspberrypi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
```

Consti10, 2021-03-01:
On stock raspbian and on CM4, when I run
`raspivid -w 1280 -h 720 -fps 60 -t 0 -n`
The h264 core clock is:
```
vcgencmd measure_clock h264
frequency(28)=250000496
```

Consti10, 2021-03-01:
When I run
`raspivid -w 1920 -h 1080 -fps 30 -t 0 -n`
The core clock is
```
vcgencmd measure_clock h264
frequency(28)=277646496
```

Consti10, 2021-03-01:
Since any rpi can have frame drops at these (resolution@framerate)<->core clock tuples
And since here https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md
The lowest clock listed for CM4 is 360mhz
I am not sure if this is the best behaviour.

popcornmix, 2021-03-02:
The 264 clock is scaled with the arm clock, which varies with arm load. I suspect the arm is relatively idle in your use case.
The h264 will get boosted to 500MHz by firmware when decoding 1080p60 video (but not when encoding).
The isp does get boosted to 300MHz when in use.

As you say h264_freq_min=500 is a workaround. As is switching to performance governor before running raspivid:
```
echo performance | sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
```

Consti10, 2021-03-02:
yes, the cpu is almost idle when doing h264 encoding only.

popcornmix, 2021-03-02:
What value of h264_freq_min seems to be sufficient for you?

Consti10, 2021-03-02:
It is hard to say, because as already pointed out, it differs by pi model and other factors.
Is there a spec from the manufacturer ?
E.g. this many macroblocks/s need that much frequency ?

popcornmix, 2021-03-02:
I believe 4.1 should work with 250MHz, but 4.2 requires an increase. @6by9 may have more accurate answers.

Consti10, 2021-03-02:
@6by9 already stated:
Only 4.0 is in specs, 4.1 is already out of spec (but works most of the time), and I tested that 4.2 is completely impossible in real time even on pi4.

popcornmix, 2021-03-02:
Well that will be the answer. If 4.1 is out of spec then it won't be in the spec. You can only confirm by experimentation. I'm not sure if we have those numbers (and it will likely be unpredictable as contents of image affects the operations done).

Consti10, 2021-03-02:
Still, I'd expect the core to clock to the max value per model (aka what's listed [here](https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md) ) if it is struggling.

Consti10, 2021-03-27:
Is it possible to add a "maxperf-mode=enable" option to raspivid ?
That'd be a good solution in my opinion.
E.g. raspivid, option maxperf-mode:
If enabled, clock the encoder core to the maximum speed possible on this model.

popcornmix, 2021-03-29:
You could just launch through a wrapper script. Something like (untested)
```
#!/bin/bash
echo performance | sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
raspivid "$@"
echo powersave | sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
```

Consti10, 2021-03-29:
But you don't necesarily want the cpu to run on max clock, only the encoder ... ?