raspberrypi/firmware #1544: current debian image (bullseye 2021-02-26 09:58:-0600) not starting on USB  SSD (but ok on SD card)

andrewufrank, 2021-03-03:
**Is this the right place for my bug report?**
I use the images distributed  from https://raspi.debian.net/daily-images/ for the raspi 4 (64 bits) and use the newest distribution. The images I download work when written on an SD card, but not when written on an SD (the SD and the bridge works with other raspi 4 images). 
The problems I encounter seem to be related to https://github.com/raspberrypi/firmware/issues/1515 and I have upgraded the eeprom and applied the fix descirbe there (added USB_MSD_PWR_OFF_TIME=0  to the rpi-eeprom).

I am not exactly clear about the relation between the firmware team and the group bringing together the images, but I sense there is a connection; I assume that the daily images are including the fixes discussed a month ago.

**Describe the bug**
The image should start but it hangs in the boot sequence after 5.8 seconds stating
```
dw2 fe98000.usb:  EPs: 8, dedicated fifos 4080 entities in SPRAM, 
loading essential drivers, 
mounting root file system
running scripts premount
running scripts local-block - this line is repeated till given up with:
vcc-sd: disabling.
...
ALERT: LABEL=RASPIROOT is not existing  -- I checked, it is the label of the second partition 
```

**To reproduce**
Writing the downloaded image to an USB, starting the raspi

**Expected behaviour**
continue the boot process with reading from the USB

**Actual behaviour**
after giving up to wait, goes to initramfs
but there is no input possible (dead keyboard)

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? Raspi 4, 8gb
* 
* Which OS and version (`cat /etc/rpi-issue`)?
(starting the same raspi on a raspios images, not the failing one!):
* Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7252c154838ec5b4576f29c996ac8fe3750cae12, stage4

* Which firmware version (`vcgencmd version`)?
* vcgencmd version
Feb 25 2021 12:10:40 
Copyright (c) 2012 Broadcom
version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start)

* Which kernel version (`uname -a`)?
* Linux baden 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux

* the eeprom: 

```
BCM2711 detected

VL805 firmware in bootloader EEPROM
Checking for updates in /lib/firmware/raspberrypi/bootloader/stable
Use raspi-config to select either the default-production release or latest update.
BOOTLOADER: up-to-date
CURRENT: Tue 16 Feb 2021 01:23:36 PM UTC (1613481816)
 LATEST: Tue 16 Feb 2021 01:23:36 PM UTC (1613481816)
RELEASE: stable
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1
```

sudo rpi-eeprom-config
```
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
USB_MSD_PWR_OFF_TIME=0

```

**Logs**
no access to the system possible

Thank you for help and your effort to produce a regular, easy to use Debian image for the RPi4!

andrewufrank, 2021-03-05:
The new image from 2021-03-05 https://raspi.debian.net/daily/raspi_4_bullseye.img.xz cannot find a the root image and hangs with 
    running-scripts/local-block 
and goes eventually to initramfs (with ALERT). 

the rpi4 used in this test has the eeprom from summer 2020 and not an updated version (I do not have the tools here to do it, testing must wait till monday).

joaormatos, 2021-03-13:
This is specific to the Debian images, which are not affiliated with the Raspberry Pi foundation, so no, this is not the place.

If you want a stable Debian system you should be using **buster** (current stable release) and the tested images from [this page](https://raspi.debian.net/tested-images/), not the daily images of bullseye (current testing release).

That being said, you're experiencing the bug described here: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=977694

The cause is the lack of the `reset_raspberrypi` module in the initramfs, which was introduced in linux 5.10 and is required for USB to work, but wasn't being included in the initramfs image.

A fix will probably be included in a future update to Debian's `raspi-firmware` package.

Meanwhile, you can add `reset_raspberrypi` to `/etc/initramfs-tools/modules` and rebuild the initramfs image with `update-initramfs -uk all`.
You'll have to do it on an SD card and then manually copy the initrd from your SD card boot partition and into your USB boot partition.

And again, be prepared to keep dealing with this kind of breakage if you're going to be using bullseye before it's stable.

andrewufrank, 2021-03-26:
Thank you very much for your informative answer to a bug report which was not appropriately directed. It took a while for me to come back to this issue and I have followed your advice -- using buster -- till bullseye becomes stable on the raspi4. I will follow the progress on the raspi-firmware package to see when it becomes updated. Understanding the error and having a receipt to fix it in case I need it is very valuable. Again, thank you for the help and you and your team for bringing a clean Debian to the raspi!.

closed 2021-03-26 by popcornmix