raspberrypi/firmware #1545: Error!  The dkms.conf for this module includes a BUILD_EXCLUSIVE directive which update 1.20210303-1

pheiduck, 2021-03-03:
Get this from todays Kernel Update updated via apt / stable buster repo
```
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.17+
Error!  The dkms.conf for this module includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.17-v7+
Error!  The dkms.conf for this module includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.17-v7l+
Error!  The dkms.conf for this module includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.17-v8+
Error!  The dkms.conf for this module includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.
```

popcornmix, 2021-03-04:
This repo is for reporting bugs in the firmware - this problem is not related to firmware.
Maybe try here: https://github.com/RPi-Distro/repo/issues/ ?

closed 2021-03-04 by popcornmix

pheiduck, 2021-03-04:
sure, but it seems to be a Kernel-headers bug?

popcornmix, 2021-03-04:
> sure, but it seems to be a Kernel-headers bug?

which is nothing to do with the firmware repo.