raspberrypi/firmware #1546: USB Webcams make system unusable on latest firmware

sgtwilko, 2021-03-04:
**Describe the bug**
Since upgrading following the December release the USB webcams have been almost unusable.
I've been trying to track down the issue, I've tried multiple webcams, and tried configuring different things, but the only thing that has made it slightly better by using the setpci command listed here.
https://github.com/raspberrypi/firmware/issues/1282#issuecomment-560105610

Basically when the webcam is in use the entire system grinds to almost a halt, keyboard presses are not registered,mouse movement is very very laggy and jumpy.
Basically the system becomes unusable.

The reason why I think this is the correct place to log this error, is because of the previous similar issues with the firmware and USB devices like webcams.
Having upgraded several time since the issue first appeared it doesn't seem to be limited to any one kernel, but during all this time the firmware was the same version.

**To reproduce**
1. USB webcam
2. Start google hangout or jitsi video meeting in Chrome
3. System becomes unresponsive and video/audio freezes

**Expected behaviour**
System will still be usable and people will be able hear/see me even if frames are dropped.

**Actual behaviour**
As soon as the camera is started the RPi runs very very slow
System becomes unresponsive and video/audio freezes

**System**
```
System Information
------------------

Raspberry Pi 4 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage4

Linux ConniePi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
Revision	: b03111
Serial		: 10000000b4f9bf60
Model		: Raspberry Pi 4 Model B Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Jan 27 2021 22:19:57 
Copyright (c) 2012 Broadcom
version 99d9a48302e4553cff3688692bb7e9ac760a03fa (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       14989948 6990212   7339580  49% /
devtmpfs          732488       0    732488   0% /dev
tmpfs             864584  200200    664384  24% /dev/shm
tmpfs             864584    8812    855772   2% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             864584       0    864584   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   48676    209420  19% /boot
tmpfs             172916       4    172912   1% /run/user/1000

Filename				Type		Size		Used	Priority
/var/swap                               file		102396		0	-2

[... 589 lines cut ...]
[  228.617372] usb 1-1.3: reset high-speed USB device number 3 using xhci_hcd
```

* Which model of Raspberry Pi? Pi4B 2GB
* Which OS and version (`cat /etc/rpi-issue`)? Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage4

* Which firmware version (`vcgencmd version`)? Jan 27 2021 22:19:57 
Copyright (c) 2012 Broadcom
version 99d9a48302e4553cff3688692bb7e9ac760a03fa (clean) (release) (start)

* Which kernel version (`uname -a`)? Linux ConniePi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux

**Logs**
lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 093a:2510 Pixart Imaging, Inc. Optical Mouse
Bus 001 Device 005: ID 04d9:0006 Holtek Semiconductor, Inc. 
Bus 001 Device 004: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 003: ID 046d:0825 Logitech, Inc. Webcam C270
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

**Additional context**
Add any other relevant context for the problem.

P33M, 2021-03-18:
Can you use rpi-update to install this specific version and re-test?
`sudo rpi-update dea7234943c604462e476a8afc13c587418e8709`

And if that works, then does 453e49bdd87325369b462b40e809d5f3187df21d break?

sgtwilko, 2021-03-21:
Hi @P33M 
I've tested this, and I can confirm the dea723 works fine, but 453e49 suffers from this issue.

Let me know if there's anything else I can do to test this.

P33M, 2021-03-29:
We've managed to reproduce your issue locally. Please try again with the latest rpi-update firmware.

sgtwilko, 2021-04-01:
@P33M  I ran `rpi-update` and, from my limited & unscientific testing, that actually seems to be better than version `dea723`.

Thank you for working on this!

closed 2021-04-07 by P33M