raspberrypi/firmware #1547: 1711f63 breaks USB booting on RPI 4B

tomty89, 2021-03-08:
Upgraded bootloader images from 530cfdd to 1711f63, attempt to boot from a USB drive gets the RPI 4B stuck at the rainbow splash.

Distro is ArchLinux ARM, kernel 5.10.20-1 (https://github.com/raspberrypi/linux/commit/95b18fac5a849708d553759f2d7bf76449303bf8). The problem occurs with [raspberrypi-bootloader 20210305-2 but not 20210305-1](https://github.com/archlinuxarm/PKGBUILDs/commit/c744836ba2023b40632d3b5ab4c80db4dfdbc85c).

EEPROM is of version `Tue Feb 16 01:23:36 PM UTC 2021 (1613481816)`, if it matters.

mikey1963, 2021-03-08:
Yea broke me on the friday update.  Now rebuilt but not updating yet.

—-

Kind regards,  Mr Michael A Smith.

Please accept my apologies in advance for any inaccuracies or errors, as this email is from a mobile device.

Distro is ArchLinux ARM, kernel 5.10.20-1 (raspberrypi/linux@95b18fa<https://github.com/raspberrypi/linux/commit/95b18fac5a849708d553759f2d7bf76449303bf8>). The problem occurs with raspberrypi-bootloader 20210305-2 but not 20210305-1<https://github.com/archlinuxarm/PKGBUILDs/commit/c744836ba2023b40632d3b5ab4c80db4dfdbc85c>.

pelwell, 2021-03-08:
Duplicate of https://github.com/Hexxeh/rpi-firmware/issues/258. A fix will be in the next firmware release, but in the meantime there's a trial firmware that should solve the problem: https://github.com/Hexxeh/rpi-firmware/issues/258#issuecomment-792274118

closed 2022-11-28 by timg236