raspberrypi/firmware #1548: FKMS: No HDMI output

alecov, 2021-03-09:
A recent firmware update has broken HDMI output when using FKMS.

I suspect this is related to the monitor's EDID not being recognized. The following appears in dmesg:
```
[    7.591902] vc4-drm gpu: [drm] HDMI-A-1: EDID is invalid:
[    7.591911]  [00] BAD  00 ff ff ff ff ff ff 00 4c 2d f6 0d 00 0e 00 01
[    7.591919]  [00] BAD  01 1b 01 03 80 66 39 78 0a 23 ad a4 54 4d 99 26
[    7.591922]  [00] BAD  47 4a bd ef 80 71 4f 81 c0 81 00 81 80 95 00 a9
[    7.591925]  [00] BAD  c0 b3 00 01 01 04 74 00 30 f2 70 5a 80 b0 58 8a
[    7.591928]  [00] BAD  50 1d 74 00 00 1e 02 3a 80 18 71 38 2d 40 58 2c
[    7.591931]  [00] BAD  45 00 50 1d 74 00 00 1e 00 00 00 fd 00 18 4b 0f
[    7.591934]  [00] BAD  1e 00 0a 20 20 20 20 20 20 00 00 00 fc 00 53 41
[    7.591936]  [00] BAD  4d 53 55 4e 47 0a 20 20 20 20 20 01 50 02 03 4f
```

HDMI output does work in this case with the `vc4-kms-v3d-pi4` overlay, and with FKMS as well in previous firmware versions (such as `99d9a48302e4553cff3688692bb7e9ac760a03fa`).

----

Model: Raspberry Pi 4B (4GB)
Kernel: 5.10.20-1-ARCH armv7l
Firmware version: `8ed5b5711855476273c45bb0754d9e23903a42e7`

cmdline.txt
```
root=/dev/mmcblk0p2 rw rootwait console=serial0,115200 console=tty1 selinux=0 plymouth.enable=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=serial0,115200 quiet audit=0
```

config.txt
```
dtparam=audio=on
dtoverlay=vc4-fkms-v3d
initramfs initramfs-linux.img followkernel
```

pelwell, 2021-03-09:
See also the latter half of https://github.com/Hexxeh/rpi-firmware/issues/258, although this is a better home.

6by9, 2021-03-09:
Test FW attached that should hopefully resolve the issues with EDID reading. It'd be appreciated if you could backup the relevant files from /boot and drop these in and retest.

[fw_1548.zip](https://github.com/raspberrypi/firmware/files/6110569/fw_1548.zip)

CaptainMidnight, 2021-03-09:
> @CaptainMidnight Test firmware attached to this issue that will hopefully resolve your issues with reading EDIDs.

Just seen this I'll give it a go.  Initially I'll test with EDID override setting left in config.txt, then without. I'll take a vcdbg log for both situations......

CaptainMidnight, 2021-03-09:
Whoop whoop, yes I can confirm using the new test firmware supplied by @6by9 above, there is no difference in reboot/boot time between having the EDID ignore option (```hdmi_ignore_edid=0xa5000080```) in /boot/config.txt and having it removed. 

1st vcdbg output : Test conditions - latest firmware fix applied, EDID ignore option in config.txt

```
004197.112: brfs: File read: /mfs/sd/config.txt
004197.884: brfs: File read: 1091 bytes
004263.715: HDMI1:EDID error reading EDID block 0 attempt 0
004269.738: HDMI1:EDID error reading EDID block 0 attempt 1
004275.762: HDMI1:EDID error reading EDID block 0 attempt 2
004281.783: HDMI1:EDID error reading EDID block 0 attempt 3
004287.806: HDMI1:EDID error reading EDID block 0 attempt 4
004293.828: HDMI1:EDID error reading EDID block 0 attempt 5
004299.851: HDMI1:EDID error reading EDID block 0 attempt 6
004305.872: HDMI1:EDID error reading EDID block 0 attempt 7
004311.895: HDMI1:EDID error reading EDID block 0 attempt 8
004317.917: HDMI1:EDID error reading EDID block 0 attempt 9
004318.931: HDMI1:EDID giving up on reading EDID block 0
004319.972: brfs: File read: /mfs/sd/config.txt
004335.478: brfs: File read: 1091 bytes
004858.205: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
004860.516: *** Restart logging
004874.716: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
004880.742: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
004886.765: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
004897.128: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
004903.912: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
004912.775: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
004918.804: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
004924.835: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
004930.865: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
004936.897: hdmi: HDMI1:EDID error reading EDID block 0 attempt 9
004937.915: hdmi: HDMI1:EDID giving up on reading EDID block 0
004942.970: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
004949.002: hdmi: HDMI1:EDID error reading EDID block 0 attempt 1
004955.031: hdmi: HDMI1:EDID error reading EDID block 0 attempt 2
004961.063: hdmi: HDMI1:EDID error reading EDID block 0 attempt 3
004967.093: hdmi: HDMI1:EDID error reading EDID block 0 attempt 4
004973.124: hdmi: HDMI1:EDID error reading EDID block 0 attempt 5
004979.153: hdmi: HDMI1:EDID error reading EDID block 0 attempt 6
004985.184: hdmi: HDMI1:EDID error reading EDID block 0 attempt 7
004991.222: hdmi: HDMI1:EDID error reading EDID block 0 attempt 8
004997.254: hdmi: HDMI1:EDID error reading EDID block 0 attempt 9
004998.272: hdmi: HDMI1:EDID giving up on reading EDID block 0
004998.294: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
004998.312: HDMI0: hdmi_pixel_encoding: 600000000
004998.326: HDMI1: hdmi_pixel_encoding: 162000000
005003.403: dtb_file 'bcm2711-rpi-4-b.dtb'
005006.177: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
005006.196: Loading 'bcm2711-rpi-4-b.dtb' to 0x100 size 0xbfc2
005022.031: brfs: File read: 49090 bytes
005028.054: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
005106.643: brfs: File read: 1523 bytes
005108.245: brfs: File read: /mfs/sd/config.txt
005108.414: dtparam: i2c_arm=off
005118.831: dtparam: i2s=off
005128.523: dtparam: spi=off
005138.686: brfs: File read: 1091 bytes
005147.119: brfs: File read: /mfs/sd/overlays/vc4-kms-v3d-pi4.dtbo
005225.108: Loaded overlay 'vc4-kms-v3d'
005415.825: brfs: File read: 3831 bytes
005418.127: brfs: File read: /mfs/sd/overlays/disable-bt.dtbo
005435.876: Loaded overlay 'disable-bt'
005475.670: brfs: File read: 1073 bytes
005476.339: brfs: File read: /mfs/sd/overlays/disable-wifi.dtbo
005487.843: Loaded overlay 'disable-wifi'
005488.021: dtparam: watchdog=on
005498.878: dtparam: act_led_trigger=timer
005537.588: brfs: File read: 387 bytes
005538.674: brfs: File read: /mfs/sd/cmdline.txt
[... 76 lines cut ...]
005566.184: brfs: File read: 1073 bytes
005566.847: brfs: File read: /mfs/sd/overlays/disable-wifi.dtbo
005578.352: Loaded overlay 'disable-wifi'
005578.540: dtparam: watchdog=on
005589.403: dtparam: act_led_trigger=timer
005628.131: brfs: File read: 387 bytes
005629.231: brfs: File read: /mfs/sd/cmdline.txt
005629.271: Read command line from file 'cmdline.txt':
005629.285: 'root=PARTUUID=66f19b0b-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait loglevel=3 quiet logo.nologo'
006542.118: brfs: File read: 114 bytes
006850.537: brfs: File read: /mfs/sd/kernel8.img
006850.559: Loading 'kernel8.img' to 0x80000 size 0x771c73
008226.497: Kernel relocated to 0x200000
008226.512: Device tree loaded to 0x2eff3a00 (size 0xc59c)
008230.755: bfs_xhci_stop
008230.765: XHCI-STOP
008230.993: xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
008231.041: USBSTS 18
009263.939: vchiq_core: vchiq_init_state: slot_zero = 0xdf000000, is_master = 1
009267.695: TV service:host side not connected, dropping notification 0x00000002, 0x00000001, 0x00000010
```

Previously as mentioned, without the EDID ignore configuration I would have witnessed at reboot/boot time an increase of the standard delta (6-7sec) + 90sec approx until a successful boot to command prompt. From my personal perspective, the cumulative additions of @timg236 and @6by9 firmware fixes solve all of my issues with the latest currently available rpi-update build.

Thanks for the quick turn around of these fixes.

6by9, 2021-03-10:
Thanks for the confirmation - I'll get that patch pushed into the normal firmware images.

popcornmix, 2021-03-10:
rpi-update firmware now contains the fix

alecov, 2021-03-11:
I confirm the fix works for me, too, no more EDID problems. Thanks for the effort of everyone involved.

Closing.

closed 2021-03-11 by alecov