raspberrypi/firmware #1551: Explicitly use empty cmdline.txt

tianyuanhao, 2021-03-15:
**Describe the bug**
When using an empty `cmdline.txt`, `start.elf` uses the built-in default cmdline instead.

**To reproduce**
Just boot the machine. It can always be reproduced.

**Expected behaviour**
`start.elf` does not append anything to the necessary cmdline.

**Actual behaviour**
`start.elf` appends the built-in cmdline to the necessary cmdline, even if `cmdline.txt` only contains a space or a blank line.

**System**

* Which model of Raspberry Pi? *Raspberry Pi 4 Model B Rev 1.2*

tianyuanhao, 2021-03-15:
I know this could be by design. But I think we can append the built-in cmdline when `cmdline.txt` does not exist, and use only the necessary cmdline when using an empty `cmdline.txt`.

asavah, 2021-03-16:
You might be wrong in your conclusions.
Default cmdline is embedded into kernel itself and AFAIK has nothing to do with start.elf.
See https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm/configs/bcm2711_defconfig#L43

pelwell, 2021-03-16:
There are also contributions from the DTB:
```
		bootargs = "coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1";
```

pelwell, 2021-03-16:
I'm inclined to just close this issue, but I'll give you a chance to explain why providing an empty command line might be useful.

tianyuanhao, 2021-03-17:
@asavah @pelwell I am pretty sure it comes from `start.elf` and not from defconfig or dts, because:

- I am using the upstream vanilla Linux kernel, not something from <https://github.com/raspberrypi>.
- When I forcibly modified the built-in cmdline in `start.elf`, I immediately got the result I wanted.

You can check the built-in cmdline of `start.elf` with the following command:

```sh
$ strings start.elf | grep console=
console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
```

> I'm inclined to just close this issue, but I'll give you a chance to explain why providing an empty command line might be useful.

It is not useful, but necessary. In some cases, the Raspberry Pi cannot boot when using the built-in cmdline. However, I couldn't clear this cmdline, except reluctantly adding a stupid `:)` to `cmdline.txt`.

pelwell, 2021-03-17:
`--` is the official way to terminate command line parsing - I suggest you use that.

tianyuanhao, 2021-03-17:
Well, there is **no difference** between `--` and `:)`. @pelwell You can close this issue. :slightly_frowning_face:

pelwell, 2021-03-17:
We may reconsider the empty cmdline handling at some point, but that could take months to roll out - this way is better for you.

closed 2021-03-17 by pelwell