raspberrypi/firmware #1558: libmmal.so not available for arm64

dwery, 2021-04-01:


pelwell, 2021-04-02:
MMAL is not (yet) supported in 64-bit environments.

closed 2021-04-02 by pelwell

ricardo-reis-1970, 2021-12-01:
Hello!

I'm developing a medical device. I'm using `picamera` with Python 3.9, on a PI 4, 64bits. Or rather, I'm _trying to use_ it, but it complains of this lib missing.

Is there any news on the subject? Can't this lib just be compiled for 64 bits, or are there dependencies and ramifications?

JamesH65, 2021-12-01:
It will not work when compiled for 64 bits, and is unlikely to do so - our emphasis now is on libCamera (that does work in both 32 and 64 bits) and a new version of picamera based on it. However, that's not ready, and won't be for some time. 

Note, that our 64bi distro has not been released which means it is still very much a prototype and should not be used for commercial systems. Unless you specifically need it, I would suggest remaining on Buster and 32bit where everything will work fine. There are only minor performance benefits to 64bit anyway.

ricardo-reis-1970, 2021-12-01:
Thanks James!

We'll be eagerly waiting for the new `picamera` (and possibly the official 64 bits OS distro) and until then we'll follow your suggestion and put 64bits back in the drawer. Our issue is that we have legacy code on `picamera`. My CV engineer actually thinks he could use `ffmpeg` directly.

Before I forget, you guys are doing a terrific job with the "Pie". Thank you!

6by9, 2021-12-01:
https://github.com/raspberrypi/userland/issues/688

ardinusawan, 2022-06-10:
Raspberry Pi OS (64-bit) is officially released, is any update on the picamera?

JamesH65, 2022-06-10:
There is also this updated picamera2 release. https://forums.raspberrypi.com/viewtopic.php?t=335832