raspberrypi/firmware #1559: Pi4 reboot not reliable, hangs forever while rebooting.

ganzgustav22, 2021-04-06:
**Is this the right place for my bug report?**
Not exactly sure, the Raspberry Pi hangs when rebooting, often at the rainbow screen, so I suspect it's the firmware. But maybe it's the eeprom/vl805 or maybe even the Linux kernel.

**Describe the bug**
Rrebooting is not reliable on the Pi4. Sometimes, after issueing the reboot command and after Linux has shutdown, it doesn't come back up and just sits there. After a while (not sure something around 20 seconds), the screen turns off and that's it. There seem to be different failure modes, sometimes the screen stays black (and/or turns off, probably due to missing HDMI signal), sometimes the Rainbow screen stays forever, sometimes there are red/black lines visible on the screen. Only way to get it out of this is to power-cycle.

**To reproduce**
1. Write a fresh sdcard with Raspberry Pi OS Lite 2021-01-11
2. Run "rpi-eeprom-update -a" (not sure if it's eeprom or vl805 firmware related, just wanted to make sure that it's using the "official" versions shipped with the distribution)
3. Write "reboot" into /etc/rc.local
4. Reboot
5. Let it reboot over and over again, wait until it it is stuck. Sometimes it's stuck after 20 Minutes or so, sometimes it takes a day or two, but eventually, it'll be stuck.

**Expected behaviour**
Rebooting works all the time. We have Raspberry Pi4's deployed in hard to access remote locations, powercycling those Raspberries is not easy.

**Actual behaviour**
Pi4 sometimes hangs during reboot.

**System**

```System Information
------------------

Raspberry Pi 4 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2021-01-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 21090519d85bdaa1615d5d5057d37b09368ea5d2, stage4

Linux raspberrypi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux
Revision        : c03111
Serial          : 10000000ff334b1f
Model           : Raspberry Pi 4 Model B Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Jan  8 2021 14:31:16
Copyright (c) 2012 Broadcom
version 194a85abd768c7334bbadc3f1911c10a7d18ed14 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        7196148 4465868   2385404  66% /
devtmpfs         1827800       0   1827800   0% /dev
tmpfs            1959896       0   1959896   0% /dev/shm
tmpfs            1959896   16964   1942932   1% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs            1959896       0   1959896   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   46273    211823  18% /boot
tmpfs             391976       4    391972   1% /run/user/1000

Filename                                Type            Size    Used    Priority
/var/swap                               file            102396  0       -2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20201210+nmu1
[... 627 lines cut ...]
MESS:00:00:06.092298:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined

MESS:00:00:06.099552:0: *** Restart logging

MESS:00:00:06.126987:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead

MESS:00:00:06.132929:0: HDMI0: hdmi_pixel_encoding: 300000000

MESS:00:00:06.138404:0: HDMI1: hdmi_pixel_encoding: 300000000

MESS:00:00:06.148824:0: dtb_file 'bcm2711-rpi-4-b.dtb'

M:0:0:8.58160:gpomn:gpomn_e_i_nm:pi SCAD_ONRO_PWE nt efne
ES:0:0:1.1336:: pima: poange_innu: inSDAR_OTOLPOE nt efne
MESS:00:00:18.309308:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:18.334594:0: Loading 'bcm2711-rpi-4-b.dtb' to 0x100 size 0x0
MESS:00:00:18.535275:0: Failed to read command line file 'cmdline.txt'
MESS:00:00:19.646034:0: Loading 'kernel7l.img' to 0x8000 size 0x0
MESS:00:00:19.649026:0: Device tree loaded to 0x2eff4100 (size 0xbe4e)
MESS:00:00:19.657010:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:19.664348:0: uart: Baud rate change done...

```
**Additional context**
Have tested this with two different Pi4 models (one 2GB and one 4GB model) and two different microSD card models: Kingston Industrial 8GB and Sandisk Ultra 32GB). Power supply used was the official Raspberry Pi4 Power Supply. The issue also occurs with no peripherals connected (only power and serialport TX and GND connected).

hastou, 2021-04-09:
@ganzgustav22 I have found similar problem fix in recent eeprom changelog. [https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/release-notes.md](https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/release-notes.md). You have firmware from Jan  8 2021, you should try to update eeprom and firmware.

ganzgustav22, 2021-04-12:
Thanks for the info, totally missed that.

I'm not sure if upgrading the firmware and kernel is a good idea for me (as it's not a "stable" firmware and kernel). I did however just update to Pi EEPROM 2021-03-18. 

Now the hang occurs even faster, have been able to reproduce it twice now within about two hours.

ganzgustav22, 2021-04-12:
The issue occured on a different machine today (unfortunately this one didn't have serial logging connected). Firmware etc. on that one are still like in the first post (i.e. Raspberry Pi OS 2021-01-11 and the firmware/eeprom/vl805fw that comes with it)

This time it drew some weird pixels on the screen, mostly blue ones, but also some red ones in a seemingly random pattern:
[image]

ganzgustav22, 2021-04-12:
Okay, I've upgraded the other one now:

sudo apt update
sudo apt full-upgrade

Now I will test again with everything updated:
`root@raspberrypi:/home/pi# uname -a
Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux

root@raspberrypi:/home/pi# rpi-eeprom-update
BOOTLOADER: up-to-date
   CURRENT: Thu 18 Mar 08:54:11 UTC 2021 (1616057651)
    LATEST: Thu 18 Mar 08:54:11 UTC 2021 (1616057651)
   RELEASE: stable (/lib/firmware/raspberrypi/bootloader/stable)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up-to-date
   CURRENT: 000138a1
    LATEST: 000138a1
`

ganzgustav22, 2021-04-13:
It got stuck again with the above version numbers (it's basically Raspberry Pi OS 2021-03-04.

Could a Pi developer please comment on this or look into it? Running the latest Raspberry Pi OS version, everything is in default state, but still, reboots are not working properly.

pelwell, 2021-04-13:
Can you try adding a small over-voltage (`over_voltage=2` in config.txt)?

ganzgustav22, 2021-04-13:
Thanks for your help. Yes, I'll try that as soon as I get to the Pi (probably tomorrow).

ganzgustav22, 2021-04-14:
I've added over_voltage=2 about two hours ago, now it's stuck again with a black screen. I'm not sure if it's just coincidence, but I think without the serial debugging options enabled it seems to happen faster (i.e. doesn't take as many reboots until it's stuck again).

Asmardos, 2021-04-20:
How is it going?
Is it fixed somehow?
I have the same problem as well and try to figure it out.
it hapens not very often but i expirienced it from time to rime in the last jear.
i just updated my system and now it hapend in round about 3 out of 20 reboots.

ganzgustav22, 2021-04-21:
It got stuck again twice (without serial logging enabled). Now I have serial logging enabled again and am again testing ...

ganzgustav22, 2021-04-26:
It's rebooting now for 5 days (with serial logging enabled) and it's still not stuck. It seems, that it only acts up when nobody is watching :-)

JamesH65, 2021-04-26:
https://en.wikipedia.org/wiki/Heisenbug

ganzgustav22, 2021-07-28:
Just tried it again with Raspberry Pi OS 2021-05-27 (and also ran rpi-eeprom-update -a to make sure it's using the firmware that comes with the release).

Issue still persists, it took about an hour of reboot loop until it got stuck again.

timg236, 2021-07-28:
I think we'd need to see the serial logs including boot logs at the point of failure.  For an unrelated hardware test I've just run 30K reboots on 10 different Pi 4Bs using the latest OS release / bootloader without error. The reboot mechanism was 'sudo reboot' in .bashrc with Raspberry Pi OS lite.

ganzgustav22, 2021-07-28:
See above, when I enable serial logging, the issue seems to go away (or atleast takes way longer to reproduce, I stopped after 5 days of continous reboots.

Can you reproduce it when you do it exactly like I described in the first post?

ganzgustav22, 2021-07-28:
It has been stuck three times now since I posted about an hour ago. It seems to occur more often than with Raspberry Pi OS 2021-01-11 (and the corresponding rpi-eeprom-update).

timg236, 2021-07-28:
Serial logging from the bootloader shouldn't make any difference unless there is a problem with the USB UART hardware. For start.elf use the following configuration to avoid any impact on the core clock 
enable_uart=1
dtoverlay=disable-bt

The picture of the HDMI screen suggests that this might be a hardware failure. I don't think this is a software issue in Raspberry Pi OS.

ganzgustav22, 2021-07-29:
I did not make hundreds of tests, so I cannot say for sure that it doesn't happen with serial logging enabled, but it looked very much like it. I'll try again though with serial logging enabled to make sure.

Regarding hardware failure: Definitely no. I have more than 10 Pi4s (bought from different vendors over a timeframe of about 6 months about a year ago) and I have tried with several different ones, they all show these weird graphic patterns from time to time. Also tried different sdcards and power supplies (including the official Pi4 power supply).

pelwell, 2021-07-29:
> For start.elf use the following configuration to avoid any impact on the core clock

The effect of the `disable-bt` overlay is more profound than affecting the core clock speeds - on a Bluetooth-equipped Pi it changes the UART used for the kernel console to UART0/ttyAMA0/PL011. Disagreements between the EEPROM, firmware and kernel about the console UART can lead to the kernel hanging early in the boot process.

ganzgustav22, 2021-07-29:
Found a better/faster/more-convenient way for reproducing the issue, have updated the first post.

(by adding "reboot -f" to /etc/init.d/raspi-config, this will speed up the whole loop for faster reproduceability and also allows to interrupt the bootloop to get back into the system by holding down the shift key).

sude22, 2021-11-03:
I'm facing similar issues (with latest OS/FW), is there a solution for this problem?

mikemanso, 2021-12-19:
I all,

I'm facing the same problem... anyone reached any conclusion on this issue?

Thanks.

ganzgustav22, 2021-12-23:
Just had problem occur on a CM4.

ganzgustav22, 2022-01-05:
Same issue occurs with Raspberry Pi 4 4GB Revision 1.4.

raphael303, 2022-01-23:
My raspberry pi 4, 8GB of Ram, all uptodate, eeprom everything, bullseye lite, booting from USB SSD hangs on EVERY reboot. Reboot ONLY works by unplugging it.

pelwell, 2022-01-23:
Without a serial cable attached this is tricky to debug, but a common cause for reboot failure is that the attached device is back-powering the Pi 4. If you have a USB hub to hand it would be interesting to see if that affects the reboot behaviour.

acyanic, 2022-01-25:
RAM perhaps?  I'm by no means an expert, but I found this when experiencing similar issues with my router ...
[https://dev.archive.openwrt.org/ticket/17839]

janoskut, 2022-03-02:
I discovered this issue the recent images ([2022-01-28](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2022-01-28/), [2021-11-08](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-11-08/))(lite, 32 bit), with **every single** `sudo reboot`. Via console I see `[   37.357020] hwmon hwmon1: Failed to get throttled (-110)` as the main error and then repeated output of (~every minute):
```
[   66.037634] rcu: rcu_sched kthread starved for 997 jiffies! g1301 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x402 ->cpu=1
[   66.047880] rcu: 	Unless rcu_sched kthread gets sufficient CPU time, OOM is now expected behavior.
[   66.056979] rcu: RCU grace-period kthread stack dump:
[  119.065535] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[  119.071556] rcu: 	0-...0: (3 GPs behind) idle=212/1/0x40000000 softirq=2440/2440 fqs=3701 
```

After I disabled bluetooth (`dtoverlay=disable-bt`) I got clearer console output, and saw "under-voltage".

#### My solution
I had supplied my Raspberry Pi 4 with 5.00 V through pins 2/4&6. After increasing the supply voltage to **5.20 V**, the reboot trouble is gone. At least the deterministic one. Maybe that helps others too.

bencat86, 2022-03-19:
I've had this issue too. No need to loop, it just hangs on every reboot.

Running headless with an external HDD, so will try unplugging that and make sure everything's up to date.

LeducH, 2022-03-23:
> I've had this issue too. No need to loop, it just hangs on every reboot. Pi4 8GB.
> 
> Running headless with an external HDD, so will try unplugging that and make sure everything's up to date.
> 
> UPDATE Just updated bootloader to `Tue 25 Jan 2022` - lastest default release. Seems to reboot no problem now, HDD and all :)

I did this 
```
rpi-eeprom-update -a
BOOTLOADER: up to date
   CURRENT: Tue 25 Jan 2022 14:30:41 UTC (1643121041)
    LATEST: Tue 25 Jan 2022 14:30:41 UTC (1643121041)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
huy@piholebian ~ % sudo reboot
Connection to piholehome.lab closed by remote host.
Connection to piholehome.lab closed.
```

still the command `sudo reboot` hangs with an external ssd

bencat86, 2022-03-30:
Sorry, you're right - I've checked it again, and same issue.

MartinVerges, 2022-03-30:
Same issue, when I execute `reboot` the system freezes.

```
root@raspberrypi:~# rpi-eeprom-update -a
BOOTLOADER: up to date
   CURRENT: Thu 10 Mar 11:57:12 UTC 2022 (1646913432)
    LATEST: Thu 10 Mar 11:57:12 UTC 2022 (1646913432)
   RELEASE: stable (/lib/firmware/raspberrypi/bootloader/stable)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
``

matzrm, 2022-04-12:
Maybe is too early, but adding a 10 secs delay with boot_delay=10 in config.txt, seems to fix. Could someone test it?

wmaxfield, 2022-04-12:
Initial testing indicates that seems to work here also.

mikeperalta1, 2022-04-12:
> Maybe is too early, but adding a 10 secs delay with boot_delay=10 in config.txt, seems to fix. Could someone else test it?

Not working for me. I put it outside any sections. Did you place that in a particular section? Using 64 bit RaspiOS.

matzrm, 2022-04-13:
Yes, was too early. Yestarday stop to works after 12 hours, today after 20 mins. :(

foorschtbar, 2022-04-26:
Same problem here after switching today to Raspberry Pi OS (64 bit) and external USB SSD. Reboot command dont work, i need to pull the power cable every time :(

wmaxfield, 2022-04-27:
I found that command "shutdown -r" did not work, but "reboot" did.  In
addition I found that if there are any issues with HDMI, reboot would hang
(and appear not to reboot).

After I changed /boot/config.txt to force HDMI with settings I wanted (even
if no cable plugged in), the reboots started working almost every time.

32 bit was easier to get to consistently reboot than 64 bit.

btrepp, 2022-05-22:
I seem to be getting this on my 8GB PI4.
It's powered using POE, from a ubiquity switch.
It doesn't have anything else plugged in.

I have it auto security updating, it used to be fine, my ansible scripts reboot sometimes, so I've used reboot reliably in the past.

Unsure if maybe some security patches have caused this?. I seem to now have to powercycle the device from the switch when doing reboots (luckily there is a webui for that, but it's fairly annoying)

foorschtbar, 2022-06-04:
My problem has disappeared. This week suddenly i saw these errors (dmesg) and a lot of undervoltage messages. 
[image]

Then I have:
- Replaced the external NVMe enclosure and the NVMe with a new case and NVMe.
- Removed a USB-C 90°/Edge adapter in the power line (Original Raspberry USB-C Power Adapter)
- Reinstalled the operating system (Bullseye, 64-bit)
- Updated the EEPROM firmware from January to April version.

I don't remember what exactly fixed the problem, but now my Pi reboots fine.

RaptorBE, 2022-06-04:
Hello, 
Thank you for your answer ! Unfortunately for me, I don't have a similar configuration [no nvme, official power supply, fresh install and up-to-date eeprom] I continue my tests and will come back here if I end up finding something. :)

paul-ridgway, 2022-06-07:
I recently found that mine failed to reboot properly if I had a usb serial cp210x adapter connected but once removed it was fine - not sure if this helps but wanted to share.

RaptorBE, 2022-06-12:
Hello,
In my case, I finally solved my problem in two steps:

-First, I activated the hdmi_safe option in the config.txt ( or 
hdmi_force_hotplug=1
hdmi_ignore_edid=0xa5000080
config_hdmi_boost=4
hdmi_group=2
hdmi_mode=4
)
-Then I saw another problem that happened more rarely indicating a stuck task problem. I found a solution on the web that seems to have worked so far.simply add these two lines in the sysctl.conf file if you notice the same problem :

vm.dirty_background_ratio = 5
vm.dirty_ratio = 10

bye :)

ganzgustav22, 2022-06-21:
I think the above comments are about a different/unrelated issue.

The issue I originally reported more than a year ago happens without any extra hardware attached. Only power-supply connected. Nothing else. Also, it doesn't happen everytime, it only happens "occasionally", i.e. after 20 or so reboots, it will happen once.

Again, this is not some kind of weird thing that happens only with a particular hardware, or only on full moons, or only under certain hard-to-reproduce circumstances.

It's quite easy to reproduce. Simply take 10 minutes, do the three steps in the first post, then let it sit until it crashed. I can take a fresh brand-new Pi4 out of it's packaging, set it up like above and it'll happen, 100% guaranteed. We use lots of Pi4s for Kiosks and other tasks, they all show this behaviour.

mimiza, 2022-07-05:
Same problem here. I have 2 Pi 4 8GB running clean/fresh install of headless Pi OS 64-bit. Everytime I `sudo reboot` the Pi via SSH, the terminal hangs forever. The only way (for now) to reboot the Pi 4 is to unplug it. This problem didn't happen with the same Pis before when they were running Pi OS 32-bit. Hope to see this fixed soon :( I'm so desperate!!!

iAmSaugata, 2022-07-11:
I am also having problem of rebooting on my Pi 4 4GB with headless x64 bit.

'''pi@RaspberryPi:~ $ sudo rpi-eeprom-update -a
BOOTLOADER: up to date
   CURRENT: Tue 26 Apr 10:24:28 UTC 2022 (1650968668)
    LATEST: Tue 26 Apr 10:24:28 UTC 2022 (1650968668)
   RELEASE: stable (/lib/firmware/raspberrypi/bootloader/stable)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
'''

pelwell, 2022-07-11:
Does `sudo reboot -f` work?

pelwell, 2022-07-11:
It works for me.

pelwell, 2022-07-11:
1. Which OS image are you running?
2. What hardware is attached?
3. Over which network interface are you sshing?
4. `tail -4 /proc/cpuinfo`.

iAmSaugata, 2022-07-11:
This usually happened if I try to reboot RPI after a day or two, it will stuck on reboot. If I reboot it within few hours of last reboot, it will reboot without any issue. I have added 10 second delay as suggested here https://github.com/raspberrypi/firmware/issues/1559#issuecomment-1096825328.

I will reboot tomorrow.

MartinVerges, 2022-07-11:
It's not related to ssh, same happens on tty shells

Rjames13, 2022-07-18:
I am having the same put it happens every single time. Just have power and Ethernet plugged, ssh over openvpn tunnel.

iAmSaugata, 2022-07-19:
> Maybe is too early, but adding a 10 secs delay with boot_delay=10 in config.txt, seems to fix. Could someone else test it?

Strangely it did fix the reboot issue for me too.

cat-microwave, 2022-07-21:
I have this same problem, but it also happens when I shut down after I do ANY reboot
got any ideas? sd card says it's fine on everything

pelwell, 2022-07-21:
People need to start answering my questions above (https://github.com/raspberrypi/firmware/issues/1559#issuecomment-1180621794) if they want this problem investigated further.

xhemp, 2022-07-21:
> 1. Which OS image are you running?
> 
>     2. What hardware is attached?
> 
>     3. Over which network interface are you sshing?
> 
>     4. `tail -4 /proc/cpuinfo`.

Running Ubuntu 22.04 64 bit, have an SSD , 1 ZWave and 1 Zigbeee dongle attached (although on reboot, I can see all services going down and the disk unmounted), I am connecting over eth0.

tail -4 /proc/cpuinfo:

# tail -4 /proc/cpuinfo
Hardware        : BCM2835
Revision        : d03114
Serial          : 10000000ce8d36ec
Model           : Raspberry Pi 4 Model B Rev 1.4

pelwell, 2022-07-21:
v. Are you booting from the SSD, or is it addition storage?

vi. Are those peripherals attached via a powered hub?

vii. Does removing the dongles and/or any hub affect the reboot behaviour?

viii. What do you see when you do try to reboot - what happens on screen (or UART), and to the LEDs?

Rjames13, 2022-07-21:
> 1. Which OS image are you running?
> 
> 2. What hardware is attached?
> 
> 3. Over which network interface are you sshing?
> 
> 4. `tail -4 /proc/cpuinfo`.

1. raspbian lite 64 bit
2. Xbee and Ethernet 
3. Tun0/ OpenVPN and eth0
4. Will share in a bit

pelwell, 2022-07-21:
I think that what we are calling reboot failures can be divided into three categories:

a. Getting stuck in software - either systemd or (perhaps) the kernel - before the watchdog reset triggers the reboot.

b. Back-powering over USB (probably) that prevents the reset from occurring or working properly.

c. The hardware not responding properly after the reset - e.g. SD card IO voltage not being reset to 3.3V, preventing it from booting.

If you have access to a serial cable (e.g. an FTDI USB adaptor, or a second Pi cross-wired) then I recommend using `sudo rpi-eeprom-config -e` to set the value of `BOOT_UART` to 1 (you must then reboot to activate the change). After that, reboots look like this:
```
pi@raspberrypi:~$ sudo reboot
[   70.212165] reboot: Restarting system

RPi: BOOTLOADER release VERSION:6efe41bd DATE: 2022/01/25 TIME: 14:30:41 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1643121041 0x8becbb81 0x00c03114 0x0004c71a
PM_RSTS: 0x00000020
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Micron' 16Gb x2 total-size: 32 Gbit 3200

XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 1
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
Boot mode: SD (01) order f4
SD HOST: 250000000 CTL0: 0x00000000 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 200000 Hz actual: 200000 HZ div: 1250 (625) status: 0x1fff0000 delay: 540
OCR c0ff8000 [2]
CID: 001b534d30303030301081142a1800f9
CSD: 400e00325b590000773f7f800a404000
SD: bus-width: 4 spec: 2 SCR: 0x02b58002 0x00000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
MBR: 0x00002000,  524288 type: 0x0c
MBR: 0x00082000,30728192 type: 0x83
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 516190
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 516190
Read config.txt bytes     2123 hnd 0x00000000
Read start4.elf bytes  2241824 hnd 0x00000000
Read fixup4.dat bytes     5352 hnd 0x00000000
Firmware: e5a963efa66a1974127860b42e913d2374139ff5 Mar 24 2022 13:19:26
0x00c03114 0x00000000 0x00000fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0
+
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.15.32-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022
...
```
Also add `enable_uart=1` and `uart_2ndstage=1` to `/boot/config.txt`, and remove `quiet` from `/boot/cmdline.txt`.

If you never see `reboot: Restarting system` then the higher levels of software are blocking the reboot. It may be necessary to wait 5-10 minutes to be certain that it isn't just taking a very long time; I'm not saying that's acceptable, but it is necessary to understand the nature of the fault in order to be able to fix it.

If you don't see `RPi: BOOTLOADER release VERSION` then it's a type b failure.

If it stops or loops before `Starting start4.elf` then it's a type c failure.

If in doubt, upload the UART log.

pelwell, 2022-07-21:
To help eliminate type b failures, try removing or changing as much external hardware as possible.

[82 more replies]

closed 2023-06-13 by pelwell