raspberrypi/firmware #1560: DVFS Not working on Raspberry Pi 4 Model B

panda-saroj, 2021-04-08:
According to the documentation [https://www.raspberrypi.org/documentation/hardware/raspberrypi/frequency-management.md] On Raspberry Pi 4 Model B, firmware from late November 2019 onwards implements Dynamic Voltage and Frequency Scaling. I have the latest firmware on my Raspberry Pi 4 Model B

Given below is the output from vcgencmd.

pi@raspberrypi:~ $ vcgencmd version

Mar 26 2021 16:34:17 Copyright (c) 2012 Broadcom
version edf2e9c318863999c97c50cdb74eee235ede3af5 (clean) (release) (start_x)

I ran a test with the following python code to change the frequency to each of the available frequencies and checked the voltage and frequency through the command

watch -n1 "vcgencmd measure_clock arm && vcgencmd measure_volts core"

While the frequency changes as per the frequency set by the script, the voltage remains the same all the time.

Script to change frequency and governors is given below.

`
import time
import os
import subprocess

NUM_CORE = 4
SLEEP_TIME = 30

#Sets Frequency of the available cores to freq
def set_freq(freq, num_core):

	set_freq_part1 = 'sudo sh -c "echo '
	set_freq_part2 = ' > /sys/devices/system/cpu/cpu'
	set_freq_part3 = '/cpufreq/scaling_setspeed"'

	get_freq_part1 = 'sudo cat /sys/devices/system/cpu/cpu'
	get_freq_part2 = '/cpufreq/cpuinfo_cur_freq'

	for core in range(num_core):
		set_freq = set_freq_part1 + str(freq) + set_freq_part2 + str(core) + set_freq_part3
	
		# print(set_freq)
		os.system(set_freq)
		print("Core {} Frequency successfully set to ...".format(core))
		get_freq = get_freq_part1 + str(core) + get_freq_part2
		os.system(get_freq)

def set_gov_userspace(num_core):
	set_gov_part1 = 'sudo sh -c "echo userspace > /sys/devices/system/cpu/cpu'
	set_gov_part2 = '/cpufreq/scaling_governor"'
	
	for core in range(num_core):
		set_gov = set_gov_part1 + str(core) + set_gov_part2
		os.system(set_gov)

# Sets the governor to the specified governor gov
def set_gov(gov, num_core):
	set_gov_part1 = 'sudo sh -c "echo ' + gov 
	set_gov_part2 = ' > /sys/devices/system/cpu/cpu'
	set_gov_part3 = '/cpufreq/scaling_governor"'
	
	for core in range(num_core):
		set_gov = set_gov_part1 + set_gov_part2 + str(core) + set_gov_part3
		os.system(set_gov)
		
		
set_gov_userspace(NUM_CORE)

get_freq_str = "sudo cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_frequencies"

result = subprocess.check_output(get_freq_str, shell=True)

#convert byte stream to string
result = result.decode("utf-8") 
freq_list = result.split()
print(freq_list)

print("\n")

# Set the cores to each of the available frequencies one by one
for freq in freq_list:
	set_freq(freq, NUM_CORE)
	# print(f"Sleeping for {SLEEP_TIME} seconds")
	print(f"Sleeping for {SLEEP_TIME} seconds")
	time.sleep(SLEEP_TIME)
	print("\n")
	
for gov in ("conservative", "ondemand", "powersave", "performance", "schedutil"):
	print("Setting Gov to :", gov)
	set_gov(gov, NUM_CORE)
	print(f"Sleeping for {SLEEP_TIME} seconds")
	# print("Sleeping for 10 seconds")
	time.sleep(SLEEP_TIME)
	print("\n")
`

popcornmix, 2021-04-09:
Try adding `dvfs=1` to config.txt. See [here](https://www.raspberrypi.org/forums/viewtopic.php?p=1804133#p1804133).

panda-saroj, 2021-04-10:
Thank you @popcornmix  for your response.
I added `dvfs=1` to /boot/config.txt file.
I only see two voltages on a Pi 4 running **32 bit raspbian OS**
**uname -a
Linux raspberrypi 5.10.27-v7l+ #1409 SMP Tue Apr 6 18:26:02 BST 2021 armv7l GNU/Linux**

- volt=0.8350V for frequency set to 600000
- volt=0.8625V for the rest of the frequencies supported

The scaling_available_frequencies on the system are as follows.
['600000', '700000', '800000', '900000', '1000000', '1100000', '1200000', '1300000', '1400000', '1500000']

**Is this is the way DVFS works on Pi 4?**
My understanding of DVFS is, each of the supported frequencies would have a different voltage label.

I also added `dvfs=1` to the config file of a Pi 4 running **64 bit of the raspbian OS** having the same firmware version.
**uname -a
Linux raspberrypi 5.10.25-v8+ #1408 SMP PREEMPT Mon Mar 22 12:52:01 GMT 2021 aarch64 GNU/Linux**

I see there is no effect on the voltage. The voltage remains constant for all of the frequencies. Does this mean that DVFS is not supported on **64 bit of the raspbian OS?** 

It seems that Pi 4 is only supporting frequency scaling and not DVFS, which involves a change in voltage for frequency change or am I missing something?

popcornmix, 2021-04-10:
You need to look at all of arm, core, v3d, h264, isp, hevc to determine if a lower voltage is possible.
v3d may be getting set higher than default when using 3d (which includes desktop use).
core may be  getting set higher than default when using higher resolution hdmi modes.

panda-saroj, 2021-04-11:
Thank you @popcornmix for your quick response.
I am using the Raspberry Pi 4 Model B in the **Headless mode.** 
No HDMI, Keyboard, or mouse connected. 
I use ssh to login to my Pi 4 and run the same CPU Intensive application at all the core frequencies supported.

From different documentation, I understand that the **voltage drops to 0.8350V** only when all clocks are at their freq_min values and that is most likely the minimum core voltage required.

My expectation was as I set the CPU core Frequencies to higher values **e.g '800000', '900000', '1000000', and so on** the core voltage should have increased gradually. But, what I observe is that the **core voltage remains constant (volt=0.8625V)** for all the higher core frequencies. 

Is it that Pi 4 supports only those two core voltage labels or are there any config settings that I need to add to have the core voltage increased as the core frequency increases?