raspberrypi/firmware #1561: Firmware does not boot my bare metal software anymore

alepi, 2021-04-09:
**Describe the bug**
Hi, I am writing bare metal software for the raspberry pi 4. To boot it I used to copy the FAT partition from a raspberry pi os image and then add in my kernel and config.txt. This does not work anymore.

Firmware from 2019-09-26-raspbian-buster.img boots bare metal software perfectly.
Firmware from 2021-03-04-raspios-buster-armhf.img hangs at the coloured square.

My config.txt is as follows:

arm_64bit=1
kernel_old=1
disable_commandline_tags=1

**To reproduce**
get this code (it's not my actual code but I tried this one as a test and got the same problem) https://github.com/isometimes/rpi4-osdev/tree/master/part5-framebuffer and build it.
Copy the firmware from 2021-03-04-raspios-buster-armhf.img and it hangs.
Copy the firmware from 2019-09-26-raspbian-buster.img and it boots.

**Expected behaviour**
I would expect the software to boot normally as before.

**Actual behaviour**
The software hangs at the initial coloured square.

**System**
 
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Raspberry pi 4 4Gb
* Which OS and version (`cat /etc/rpi-issue`)?
*  None. It's a bare metal program
* Which firmware version (`vcgencmd version`)?
* The firmware from 2021-03-04-raspios-buster-armhf.img
* Which kernel version (`uname -a`)?
* None. It's a bare metal program

popcornmix, 2021-04-09:
Can you identify the exact update which caused this. See:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with this error.

alepi, 2021-04-09:
I'll do that and report it back here when I find out what the specific commit is.
Thanks :-)

alepi, 2021-04-10:
Ok, I did the tests. I have two consecutive commits where the earliest works and the latest doesn't

commit 634e380a4d041492f859712bd2c81112a535b515 DOES work
commit da3752a358a86014cdcce5fc3be5b18d7ec074c4 does NOT work

I am leaving that to you guys. Let me know :-)

pelwell, 2021-04-12:
`firmware: arm_ldconfig: Honour the kernel8 text offset` looks like a likely culprit.

pelwell, 2021-04-12:
That sounds like a regression. Does it work with `kernel_address=0`?

pelwell, 2021-04-12:
Never mind - it won't. The fix looks easy.

alepi, 2021-04-12:
Cool thanks :-)

pelwell, 2021-04-12:
The fix was, indeed, simple - it's in the internal software repo and a firmware build including the patch has successfully loaded the example bare-metal image and put some graphics on the monitor. Once the next firmware release is available we'll comment here - close the issue when you can confirm the problem is solved.

alepi, 2021-04-13:
cool thanks :-)

popcornmix, 2021-04-13:
Not sure if you noticed the github message above, but the fix is in current rpi-update firmware.

alepi, 2021-04-14:
Hi,

thanks :-)

Sorry for the slow response, it can take me a while to do the rpi-update as I am not using raspbian and every time I have to reinstall it.

I just reinstalled it, did an rpi-update and still it does not work.

I tried with the specific commit with hash 1df55790fb191704c0ce630d4d0713a8beb43a7d and it complained that the hash was invalid.

pelwell, 2021-04-14:
@popcornmix This update didn't make its way to Hexxeh's rpi-firmware repo, so it isn't available via rpi-update. Is this by design?

@alepi In the short term you will need to download the files manually from here: https://github.com/raspberrypi/firmware/tree/master/boot

popcornmix, 2021-04-14:
Sorry - didn't get pushed to Hexxeh - done now.

alepi, 2021-04-14:
rip-update works now :-)

closing the issue

closed 2021-04-14 by alepi