raspberrypi/firmware #1563: Missing firmware for USB WiFi Adapters based on the mt7610u Chipset

morrownr, 2021-04-16:
Problem diagnosed as follows:

$ dmesg | grep mt
[    9.419439] mt76x0u 1-1.3:1.0: ASIC revision: 76100002 MAC revision: 76502000
[    9.423426] mt76x0u 1-1.3:1.0: mediatek/mt7610e.bin not found, switching to mediatek/mt7610u.bin
[    9.423499] mt76x0u 1-1.3:1.0: Direct firmware load for mediatek/mt7610u.bin failed with error -2
[    9.432698] mt76x0u: probe of 1-1.3:1.0 failed with error -2

I looked. The file, mt7610u.bin does not exist in Raspberry Pi OS.

Local solution:

Download mt7610u.bin from the following location:

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek

$ sudo mkdir /lib/firmware/mediatek
$ sudo cp <mt7610u.bin> /lib/firmware/mediatek
$ sudo reboot

Request that this firmware be including in Raspberry Pi OS by default. Thank you.

MrEngman, 2021-04-16:
I think the real fix is to change file **src/linux/drivers/net/wireless/mediatek/mt76/mt76x0/mt76x0.h** to get the driver to load the firmware from directory **/lib/firmware** rather than from the directory that does not exist **/lib.firmware/mediatek**

e.g. using the following patch
```
--- old-mt76x0.h        2021-04-16 12:25:52.359294186 +0100
+++ mt76x0.h    2021-04-16 17:03:50.213710112 +0100
@@ -20,10 +20,10 @@
 #include "../mt76x02.h"
 #include "eeprom.h"

-#define MT7610E_FIRMWARE               "mediatek/mt7610e.bin"
-#define MT7650E_FIRMWARE               "mediatek/mt7650e.bin"
+#define MT7610E_FIRMWARE               "mt7610e.bin"
+#define MT7650E_FIRMWARE               "mt7650e.bin"

-#define MT7610U_FIRMWARE               "mediatek/mt7610u.bin"
+#define MT7610U_FIRMWARE               "mt7610u.bin"

 #define MT_USB_AGGR_SIZE_LIMIT         21 /* * 1024B */
 #define MT_USB_AGGR_TIMEOUT            0x80 /* * 33ns */

```
As a temporary fix I use commands 
```
wget http://ftp.uk.debian.org/debian/pool/non-free/f/firmware-nonfree/firmware-misc-nonfree_20190114-2_all.deb
sudo dpkg -i firmware-misc-nonfree_20190114-2_all.deb
sudo apt-mark hold firmware-misc-nonfree
```
This loads the necessary firmware in the directory **/lib/firmware/mediatek** where the current driver expects it to be. The **apt-mark hold** command stops the directory being removed if you run apt command which will load the mt7610 firmware in directory **/lib/firmware** and remove the **/lib/firmware/mediatek** where the current driver needs it to be.

MrEngman

morrownr, 2021-04-16:
MrEngman,

Your patch gets a thumbs up from me. Have you submitted it to the Linux Wireless folks?

Regardless of the location, the file is not in RasPiOS while the driver is. I have an adapter based on the mt7612u chipset and it works without issue with RasPiOS but when trying to use my adapter based on the mt7610u chipset, well, it goes without saying that it doesn't work without going to the trouble of installing the firmware. Why is one supported and the other not?

Cheers.

MrEngman, 2021-04-16:
For the mt7610 wifi in the RaspOS the firmware is installed in directory **/lib/firmware** using the firmware package **firmware-misc-nonfree** when using **apt upgrade** and the driver used this directory to get it's firmware from before being installed in the RaspOS image but for some reason or other with the driver currently installed in RaspOS it now tries to load the firmware from the **/lib/firmware/mediatek** directory.

I'm sure at some point the same issue occurred with the mt7612 driver when it was first included in the RaspOS images but that was fixed fairly quickly I believe. So don't know why the mt7610 drive has not been fixed yet.

With the fix I use it uses an older version of the **firmware-misc-nonfree** firmware package which loads the firmware in directory **/lib/firmware/mediatek** for whatever reason.

MrEngman

ganzgustav22, 2021-04-21:
This is an upstream bug in debian, the file "mt7610u.bin" is missing from the firmware-misc-nonfree package.
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=931561