raspberrypi/firmware #1564: Raspberry Pi4 hangs on boot from USB HDD

MisteryX, 2021-04-16:
**Describe the bug**
Add a clear and concise description of what you think the bug is.
Nearly month ago configured RPI4 as Ubuntu 20.04 headless server, booted from USB HDD and everything was fine, but today (16.04.2021) at 4:30am Ubuntu automatically downloaded an update and have installed it (files in boot partition had changed). I figured out, that RPI4 hangs in the morning, so I rebooted device, but it did not boot anymore. 

**To reproduce**
Just try to boot from USB HDD in Ubuntu 20.04 server

**Expected behaviour**
Normal boot

**Actual behaviour**
When I had connected to HDMI I noticed that boot process hangs on string 
"Running /scripts/local-premount" - hangs several minutes, then shows a lot of strings:
"mdadm: No arrays found in config file or automatically"
Then
"Gave up waiting for root file system device"
"LABEL=writable does not exists. Dropping to a shell"
and drops to BusyBox shell

**System**
* Raspberry Pi4 model B
* HDD Toshiba Canvio Advanced - 1Tb
* Ubuntu 20.04 server
* Latest
* Latest

**Additional context**
I have checked HDD with smartmon and found no problems. I have checked HDD FS with fsck and successfuly fixed several errors on ext4 (writable) and fat32 (system-boot) partitions. HDD was connected to RPI4 through active USB hub Ginzzu GR-388UAB, but direct connection to RPI4 did not fix the problem.

popcornmix, 2021-04-16:
It's probably best to ask in an Ubuntu forum as we are not involved in those images, and so don't know what the update may have changed.
Given you have got as far as a "Running /scripts/local-premount" message it won't be firmware related.

MisteryX, 2021-04-16:
Thanx, I will try Ubuntu forums

whitty, 2021-06-10:
@MisteryX - could you follow up with a link to the ubuntu forums if you did post?  Google has brought me to this page due to the same described problem and I'm left at a dead end.

MisteryX, 2021-06-11:
Yep, I have solved the problem. It was because of damage of boot partition. I downloaded boot firmware files from official Raspberry github repo and copied them to boot partition and it helped to boot. Damage of boot partition is typical problem because of unreliable FS - FAT32 in Raspberry. You should have backup for boot partition for your Raspberry. My source of damage was unreliable USB hub, wich produced a lot of mistakes during data transfer.

closed 2021-06-11 by MisteryX

IsmartPrimos, 2021-09-02:
> Yep, I have solved the problem. It was because of damage of boot partition. I downloaded boot firmware files from official Raspberry github repo and copied them to boot partition and it helped to [...]

How did you solve it? I have very similar problem. Is there any link?

JamesH65, 2021-09-02:
You cannot blame an unreliable FAT32 system on Raspberry when you have a broken hub! I've never encountered a corrupted boot partition through any failure of Pi hardware or software.

To solve problem associated with corrupted files on the boot partition, you need to copy good files from somewhere else over the top of the broken ones, either by creating a new Raspberry Pi OS SD card, or I think you can get hold of the files from github but I cannot remember the location off hand.