raspberrypi/firmware #1566: unable to change GPIO14&15 to alt5(uart1)

emanuel-bacigala, 2021-04-19:
I am unable to set GPIO14&15 to alt5(uart1) in config.txt using:
gpio=14=a5=pn
gpio=15=a5=pu

after reboot pins are set as alt0:
GPIO 14: level=1 fsel=4 alt=0 func=TXD0
GPIO 15: level=1 fsel=4 alt=0 func=RXD0

I would expect to see GPIO14&15 set to alt5.

If I commnet out:
#gpio=14=a5=pn
#gpio=15=a5=pu

pins are set as input (witch is ok - that's defaut state)
GPIO 14: level=0 fsel=0 func=INPUT
GPIO 15: level=0 fsel=0 func=INPUT

board type: CM3+

vcgencmd version:
Mar 26 2021 16:34:43
Copyright (c) 2012 Broadcom
version edf2e9c318863999c97c50cdb74eee235ede3af5 (clean) (release) (start)

uname -a:
Linux raspberrypi 4.19.127-v7 #1 SMP Tue Oct 20 16:21:29 CEST 2020 armv7l GNU/Linux

pelwell, 2021-04-19:
The UART-handling code in the firmware is quite complex, with a number of different factors to consider. It would help to know if you have set "enable_uart=1" in config.txt.

emanuel-bacigala, 2021-04-20:
enable_uart=1 is not present in my config.txt

emanuel-bacigala, 2021-04-20:
Here is my config.txt:

gpu_mem=32 
avoid_warnings=1
disable_splash=1
ignore_lcd=1
hdmi_ignore_edid=0xa5000080
hdmi_drive=1

gpio=14=a5,pn
gpio=15=a5,pu

dtparam=i2c2_iknowwhatimdoing=on
dtparam=i2c2_baudrate=100000

dtparam=audio=off
avoid_pwm_pll=1

pelwell, 2021-04-20:
Compute Modules prior to CM4 are unusual because they don't have onboard Bluetooth (making UART0 the obvious console UART for GPIOs 14 & 15) but they also allow a UART to be enabled on the higher GPIOs (32 & 33, 40 & 41). For other non-Bluetooth Pis you wouldn't want to enable UART1 on 14 & 15 because UART0 is superior in every respect, but this isn't necessarily true for CM1 & CM3 - as you have shown.

The firmware tries to configure its UARTs to match the kernel UART usage by examining the Device Tree, otherwise you can end up with deadlocks when the kernel start. It takes its cues from the DT aliases "serial0" and "serial1". On a CM3, these are set up such that "serial0" points to UART0 and "serial1" points to UART1, where "serial0" is expected to be the console UART on 14 & 15. Based on this, the firmware concludes that it should configure UART0 to be on 14 & 15, overriding your `gpio=` setting.

Unfortunately there is currently no convenient way to achieve your perfectly reasonable goal. The options are:
1. You customise your cm3 dtb file (bcm2710-rpi-cm3.dtb) to make "serial0" point to UART1. This is the way Compute Modules were expected to be used.
2. We create an overlay whose sole job it is to swap the serial aliases.
3. We modify the existing uart1 overlay so that can also set "serial0".
4. We get the firmware to recognise GPIO14==a5 as an explicit choice by the user - this isn't a default configuration.

I think option 4 is definitely required, but options 2 and/or 3 are probably also sensible.

emanuel-bacigala, 2021-04-20:
Thank you for your answer. I would also prefer option 4.

pelwell, 2021-04-20:
I'm testing a patch for this at the moment. It will be necessary to use something like "dtoverlay=uart1" (or a custom dtb) if you want to enable ttyS0 in the kernel.

pelwell, 2021-04-20:
The patch has been merged and will be in future releases. You can download a preview firmware build (just start.elf and fixup.dat) here: https://drive.google.com/file/d/1tquz4DGRZ25qCJMpbaEDhao-wSoPBsEp/view?usp=sharing

emanuel-bacigala, 2021-04-21:
From my point of view FW now works as expected. Thank you for your effort. Can I close this issue?

pelwell, 2021-04-21:
Please do!

closed 2021-04-21 by emanuel-bacigala

popcornmix, 2021-04-21:
This fix is now in latest rpi-update firmware.