raspberrypi/firmware #1567: system fails to boot after renaming kernel

emanuel-bacigala, 2021-04-23:
After renaming kernel7.img to kernel7-l318.img my CM3+ fails to boot.
Firmware hash is a48d332c35ee1c1c1ab433228e23317f62dcc5fb
Linux version 5.10.31-v7+

cmdline.txt:
dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 elevator=deadline root=/dev/mmcblk0p2 rootfstype=ext4 fsck.repair=yes fsck.mode=force rootwait

config.txt:
kernel=kernel7-l318.img

framebuffer_depth=32
gpu_mem=256
avoid_warnings=1
disable_splash=1
ignore_lcd=1
hdmi_ignore_edid=0xa5000080
hdmi_drive=1

uart_2ndstage=1
uart0=on
gpio=32=a3,pn
gpio=33=a3,pu

dtparam=i2c2_iknowwhatimdoing=on

serial console output:
Raspberry Pi Bootcode
Read File: config.txt, 234
Read File: start.elf, 2955968 (bytes)
Read File: fixup.dat, 7311 (bytes)
MESS:00:00:01.908563:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.913030:0: brfs: File read: 234 bytes
MESS:00:00:01.972213:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.976626:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:02.008121:0: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
MESS:00:00:02.014176:0: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
MESS:00:00:02.021923:0: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
MESS:00:00:02.029367:0: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
MESS:00:00:02.037110:0: gpioman: gpioman_get_pin_num: pin BT_ON not defined
MESS:00:00:02.043770:0: gpioman: gpioman_get_pin_num: pin WL_ON not defined
MESS:00:00:02.077090:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:02.082918:0: *** Restart logging
MESS:00:00:02.086792:0: brfs: File read: 234 bytes
MESS:00:00:02.102188:0: HDMI0: hdmi_pixel_encoding: 162000000
MESS:00:00:02.107149:0: gpioman: gpioman_get_pin_num: pin CAMERA_0_I2C_PORT not defined
MESS:00:00:02.113971:0: dtb_file 'bcm2710-rpi-cm3.dtb'
MESS:00:00:02.123012:0: brfs: File read: /mfs/sd/bcm2710-rpi-cm3.dtb
MESS:00:00:02.127673:0: Loading 'bcm2710-rpi-cm3.dtb' to 0x100 size 0x690a
MESS:00:00:02.146868:0: brfs: File read: 26890 bytes
MESS:00:00:02.212805:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:02.216856:0: dtparam: i2c2_iknowwhatimdoing=on
MESS:00:00:02.235846:0: brfs: File read: 234 bytes
MESS:00:00:02.239349:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:02.244361:0: Read command line from file 'cmdline.txt':
MESS:00:00:02.250227:0: 'dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 elevator=deadline root=/dev/mmcblk0p2 rootfstype=ext4 fsck.repair=yes fsck.mode=force rootwait'
MESS:00:00:03.929490:0: gpioman: gpioman_get_pin_num: pin WL_ON not defined
MESS:00:00:03.938715:0: brfs: File read: 174 bytes
MESS:00:00:04.508699:0: brfs: File read: /mfs/sd/kernel7-l318.img
MESS:00:00:04.513097:0: Loading 'kernel7-l318.img' to 0x80000 size 0x60c690
MESS:00:00:04.519791:0: Device tree loaded to 0x2eff9200 (size 0x6d9c)
MESS:00:00:04.527513:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:04.533803:0: uart: Baud rate change done...
MESS:00:00:04.537235:0: uart: Baud rate change done...
MESS:00:00:04.542985:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined

...system hags here...

if I renamed kernel7-l318.img to kernel7.img and commented out kernel=... in config.txt system boots ok:
Raspberry Pi Bootcode
Read File: config.txt, 234
[... 90 lines cut ...]
[    0.010556] smp: Brought up 1 node, 4 CPUs
[    0.010611] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.010643] CPU: All CPU(s) started in HYP mode.
[    0.010671] CPU: Virtualization extensions available.
[    0.011634] devtmpfs: initialized
[    0.029043] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.029320] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.029381] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.032642] pinctrl core: initialized pinctrl subsystem
[    0.033869] NET: Registered protocol family 16
[    0.038339] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.043996] audit: initializing netlink subsys (disabled)
[    0.044314] audit: type=2000 audit(0.040:1): state=initialized audit_enabled=0 res=1
[    0.044912] thermal_sys: Registered thermal governor 'step_wise'
[    0.046095] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.046159] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.046463] Serial: AMBA PL011 UART driver
[    0.063807] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[    0.080162] raspberrypi-firmware soc:firmware: Attached to firmware from 2021-04-21T15:49:36, variant start
[    0.090172] raspberrypi-firmware soc:firmware: Firmware hash is a48d332c35ee1c1c1ab433228e23317f62dcc5fb
[    0.134082] Kprobes globally optimized
[    0.139079] bcm2835-dma 3f007000.dma: DMA legacy API manager, dmachans=0x1
...

This bug? might have been introduced with 5.10.y because when I tried 5.4.72 (7a06d62ff1c0c6bd72d4b9a91b6e38eaf68aeb70) there was no problem with different kernel name.

pelwell, 2021-04-23:
And what happens if you use `kernel=kernel7.img`?

pelwell, 2021-04-23:
I understand now - you are being bitten by a heuristic that says that, in the absence of other information, kernel names ending "8.img" should be run in 64-bit mode. Add `arm_64bit=0` to disable this feature.

emanuel-bacigala, 2021-04-23:
After adding arm_64bit=0 system boots ok. Growing FW intelligence begins to cause problems. Thank you for helping me.

closed 2021-04-23 by emanuel-bacigala

pelwell, 2021-04-23:
I may reconsider that particular heuristic.

emanuel-bacigala, 2021-04-23:
Adding some uart stage2 debugging msg like "CPU running in XXbit mode" can also help to identify similar problems.