raspberrypi/firmware #1569: GPIO filters in config.txt are treated as mutually exclusive even when on different GPIO.

thagrol, 2021-04-25:
GPIO filters in config.txt are treated as mutually exclusive even when on different GPIO.

Given the following (somewhat artificial) config.txt, only the last GPIO filter in each group is acted on.
```
[all]
gpio=19=ip,pu
gpio=20=ip,pu

[gpio19=0]
[gpio20=0]
gpu_mem=16
[all]

[gpio19=0]
[gpio20=1]
gpu_mem=32
[all]

[gpio19=1]
[gpio20=0]
gpu_mem=48
[all]

[gpio19=1]
[gpio20=1]
gpu_mem=64
[all]
```
Across multiple boots pull GPIOs 19 and 20 into all combinations and check allocated GPU RAM each boot. It can be seen that only the state of GPIO 20 makes a difference. The state of GPIO 19 is ignored.

Because of this and the last one wins nature of config.txt parameters gpu_mem is only ever set to 48 or 64.

Same behaviour is observed with more than 2 GPIO in use and is not specific to GPIOs 19 and 20.

Pi Model: Raspberry Pi 4B Rev 1.1
OS: Linux devpi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux

I suspect all Pi models are affected.

The following excerpt from the documentation for conditional filters may be related:
> Combining conditional filters
Filters of the same type replace each other, so [pi2] overrides [pi1], because it is not possible for both to be true at once.
>
>Filters of different types can be combined simply by listing them one after the other

thagrol, 2021-04-25:
From `sudo vcgencmd bootloader_version`:

Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
update-time 0
capabilities 0x00000000

pelwell, 2021-04-25:
Two gpio filters are clearly of the same type, therefore cannot be combined. The feature is behaving as expected, and as documented,

closed 2021-04-25 by pelwell

thagrol, 2021-04-25:
Of the same type, yes. But filters on different GPIO are not mutually exclusive in the same way that [pi2] and [pi1] are. Which, from the documentation, appears to be the reason behind the current ban on stacking two filters of the same type.

That it matches the documentation doesn't make the behaviour optimal.

jharris1993, 2021-04-26:
> Two gpio filters are clearly of the same type, therefore cannot be combined. The feature is behaving as expected, and as documented,

I would beg to differ.

As documented, maybe.  As expected? Absolutely not.

In the documentation on combining conditionals in the config.txt file it says:
>Filters of the same type replace each other, so [pi2] overrides [pi1], **_because it is not possible for both to be true at once_**.

The important takeaway is that the cannonical example, and reasoning behind this "rule" is that the two choices are mutually exclusive - a boolean cannot be both TRUE and FALSE at the same time, (outside of quantum computing), so that doesn't work.

In the same way, a Pi cannot be a Pi-2 and a Pi-4 at the same time.

However, GPIO pins are individual, discrete, items that can, (and often do), have distinct and individual values.

Obviously, GPIO-16 cannot be both high and low, however it is very much possible for GPIO-16 to be high while GPIO-17 is low.

Saying that individual GPIO pins are "the same type" is like saying that setting the GPIO memory split, setting the HDMI mode and a specific monitor to some value are "the same type" because they're all video/GPIO parameters.  And that's true, all of these are things that affect the GPIO and are "video" parameters, but they are obviously discrete things that do different things and have different functions.

I would also beg to differ with the tone of your response as being dismissive.  The fact that it may be "as designed" doesn't mean the design isn't incredibly silly in this regard, and it does not mean that it shouldn't be reconsidered.

Saying it's "as documented", (it's **_not_** "as documented", because GPIO pins are **_not_** mutually exclusive), and then closing the issue is tantamount to saying "The Great Oz Has Spoken to impudent children, so go 'naff-off and don't bother your betters!"

This communicates the wrong image and leaves a horrible taste in the mouth of anyone else who may wish to raise what appears to be a legitimate issue in the future.

The correct response is to re-open the issue and assign it to someone for research and discussion.

jharris1993, 2021-04-27:
@thagrol 

This is so incredibly silly that I cannot believe it's not being considered.

IMHO, the only way to get this done is to fork the repo, fix the issue, and create a pull-request.  I could fork it to my repo and add you, or you could fork it to your repo and add me.

Would you like to take a peek at this with me and see what we can do?

pelwell, 2021-04-27:
While you are drafting your request for enhanced functionality, consider this:
```
[gpio4=0]
dtoverlay=overlay-a

[gpio5=0]
dtoverlay=overlay-b
```
If GPIO4 is high and GPIO5 is low, which overlays get loaded? Previously the answer would have been `overlay-b`.

jharris1993, 2021-04-27:
That's exactly the point.

With the GPIO's not considered "the same type", you can logically AND them:
```
# default overlay for any other states of gpio4 and gpio5
dtoverlay=defaultoverlay

[gpio4=0]
[gpio5=1]
dtoverlay=overlay-a

[gpio4=1]
[gpio5=0]
dtoverlay=overlay-b
```

Obviously, it is up to the person writing the config.txt file to account for possibilities and handle them.

pelwell, 2021-04-27:
You didn't answer the question.

jharris1993, 2021-04-27:
I thought I did.

>If GPIO4 is high and GPIO5 is low, which overlays get loaded?

By definition, overlay-b because the condition for overlay-a hasn't been satisfied.

Also, by definition, **_if both are low, it will still be overlay-b_** because of the rule that if you set the same thing more than once, (in this case, "dtoverlay"), the last setting wins.  This is why I can set a default dtoverlay before the conditions - if a condition is true it overrides the default setting.

What I am talking about is making the two mutually exclusive.  For example

[gpio4=0]
[gpio5=1]
dtoverlay=overlay-a
arm_64=1
[all]

[gpio4=1]
[gpio5=0]
dtoverlay=overlay-b
gpu_memory_split=128
[all]

In this case, it is necessary for them to be mutually exclusive because you need to guard against the condition of both being zero which will cause both gpu memory to be set ***AND*** arm-64 to be set too, which is not the desired behavior.

jharris1993, 2021-04-27:
Question:
*  If I decide to "have a go" at this and try to implement the modification on my own and submit a pull-request, exactly what do I have to do, (detailed steps please), to build a firmware file that I can use to update a Pi-4?
* Do I need to create a firmware file for the Pi-3?
* How would this be implemented on an earlier Pi, if possible?

pelwell, 2021-04-27:
> because of the rule that if you set the same thing more than once, (in this case, "dtoverlay"), the last setting wins

No - dtoverlay= is a directive with side effects.

> By definition, overlay-b because the condition for overlay-a hasn't been satisfied.

So are you saying that (with GPIO4 high and GPIO5 low):
```
[gpio4=0]
dtoverlay=overlay-a

[gpio5=0]
dtoverlay=overlay-b
```
should cause overlay-b to be loaded, but:
```
[gpio4=0]
[gpio5=0]
dtoverlay=overlay-b
```
should do nothing?

pelwell, 2021-04-27:
The firmware is closed source, which is why this is a repo full of binaries.

jharris1993, 2021-04-27:
> So are you saying that (with GPIO4 high and GPIO5 low):
> 
> ```
> [gpio4=0]
> dtoverlay=overlay-a
> 
> [gpio5=0]
> dtoverlay=overlay-b
> ```
> 
> should cause overlay-b to be loaded, but:
> 
> ```
> [gpio4=0]
> [gpio5=0]
> dtoverlay=overlay-b
> ```
> 
> should do nothing?

If gpio4=1 and gpio=0, (the actual condition of the pins, not the conditional state)

Then [gpio4=0] has to fail because gpio4 is actually = 1.

With the two conditions set low as you said:
[gpio4=0]
[gpio5=0]
dtoverlay=overlay-b

That is exactly correct - nothing happens - because the condition [gpio4=0] AND [gpio5=0] is not true.

Taking the point that this is "expected" behavior, I would beg to differ.

I have been both following and posting on several different threads within the Raspberry Pi forums, including the PINN sticky, and ***everyone I have discussed this with has assumed that GPIO conditions were not mutually exclusive***.

Other Raspberry Pi forums I participate in have universally assumed that GPIO pins could be AND'ed together based on the conditions as documented. Certain posters have **_actually quoted that sentence_** as proof that they could be AND'ed together.

No one, not a single one, after reading the documentation carefully, saw the GPIO pins being mutually exclusive because the example is something that **_is mutually exclusive by definition_**, (pi1 vs pi2) and the documentation explicitly states that ***mutual exclusivity is the basis for the rule***.  Ergo, it is correct to assume that things which are not mutually exclusive by definition are capable of being AND'ed together.

Every single person assumed that gpio conditions could be AND'ed together.

Though it may be "by design", IMHO that design misunderstands the basis for the rule - the objects are absolutely mutually exclusive - when applied to things that are not mutually exclusive.

IMHO, one of two things need to happen:
1.  (The best choice) Relax this rule with respect to GPIO pins.
2.  Change the documentation to explain that, even though individual GPIO pins are not mutually exclusive, they still cannot be AND'ed together.

jharris1993, 2021-04-27:
> So are you saying that (with GPIO4 high and GPIO5 low):
> 
> ```
> [gpio4=0]
> dtoverlay=overlay-a
> 
> [gpio5=0]
> dtoverlay=overlay-b
> ```
> 
> should cause overlay-b to be loaded, but:
> 
> ```
> [gpio4=0]
> [gpio5=0]
> dtoverlay=overlay-b
> ```
> 
> should do nothing?

Actually that's not exactly true.

Based on the published definition of ANDing conditions together, conditions that follow each other without an intervening [all] statement add additional restrictions.

```
[gpio4=0]
do this thing

[gpio5=0]
do some other thing
```

According to the current documentation, nothing should happen because:
1.  gpio4 != 0, therefore, "do this thing: is skipped
2. [gpio5=0] does not supersede the previous condition but rather ADDS a new condition to it, *i.e.* a logical AND.
3. Therefore because gpio4=0 AND gpio5=0 is FALSE, "do some other thing" is also skipped.

thagrol, 2021-04-27:
> 
> 
> While you are drafting your request for enhanced functionality, consider this:
> 
> ```
> [gpio4=0]
> dtoverlay=overlay-a
> 
> [gpio5=0]
> dtoverlay=overlay-b
> ```
> 
> If GPIO4 is high and GPIO5 is low, which overlays get loaded? Previously the answer would have been `overlay-b`.

In that specific case I'd expect neither to be loaded.

There is no [all] between the two filters so, based on the current behaviour of stacked conditionals, the condition for loading overlay-b  would appear to be gpio4=0 AND gpio5=0

That's the behaviour I'd expect from this when running on a 4B regardless of the state of gpio4:
```
[pi2]
dtoverlay=overlay-a
[gpio4=0]
dtoverlay=overlay-b
```
On a 2B with gpio 4 pulled low I'd expect both overlays to be loaded as dtoverlay is not a config.txt directive where last one wins.

thagrol, 2021-04-27:
To expand on the above,

With
```
[gpio4=0]
[gpio4=1]
## do something
```
The current behaviour is acceptable.

With
```
[gpio4=0]
[gpio5=0]
## do something
```
I'd expect somthing to be done only if both gpio4 AND gpio5 are 0.

pelwell, 2021-04-27:
Whether or not it is what you expect, case A (I've labelled them for clarity) with 4 high and 5 low currently causes overlay-b to be loaded. Changing the semantics of the gpio filter to cause gpio filters with different GPIO numbers to be combined would break systems that expect the existing semantics created by users who understand the difference between type and instance.

jharris1993, 2021-04-27:
> Changing the semantics of the gpio filter to cause gpio filters with different GPIO numbers to be combined would break systems that expect the existing semantics created by users who understand the difference between type and instance.

I thought this was exactly the point.  GPIO pins, because there are more than one on any individual Pi, it is reasonable to expect them to have different values and be testable that way.

jharris1993, 2021-04-27:
> In that specific case I'd expect neither to be loaded.
>
> There is no [all] between the two filters so, based on the current behaviour of stacked conditionals, the condition for loading overlay-b [...]

As I mentioned before, this is not a misunderstanding limited to just the two of us.  I have seen this assumption, based on how the documentation is written, that GPIO conditionals could be stacked to be the common perception by many people on many different venues.

thagrol, 2021-04-27:
I suspect few people actually use the GPIO filters else this would likely have come up before. Though that's another argument against making the change.

IIRC weren't gpio filters added after conditional filters were? (My git fu isn't good enough for me to search the revision history). If I'm correct it would seem little thought was given to this back at the design stage leaving us with a suboptimal filter.

jharris1993, 2021-04-27:
> I suspect few people actually use the GPIO filters else this would likely have come up before.

GPIO filtering is one of the "hidden gems" of the Raspberry Pi.

I have seen a large number of people - especially on the PINN sticky - who jump through hoops, burn brown rice, sacrifice small rodents, and sell their souls - to accomplish via software or SSH what is trivially done with a GPIO filter.

Especially with the proliferation of "headless" implementations for the Pi, (mine is a robot without an attached display), there is a need to be able to create different configuration options at boot-time which is where GPIO conditionals shine.

This is a topic that really deserves its own article on the Raspberry Pi site.  It would make so many people's lives so much easier.

jharris1993, 2021-04-27:
> This is a topic that really deserves its own article on the Raspberry Pi site. It would make so many people's lives so much easier.

. . . . and would allow the Raspberry Pi to be used in so many  new applications where it's not being used now, because of the, (perceived), inability to make fundamental configuration changes at boot-time without jumping through hoops.