raspberrypi/firmware #1570: '/boot/dtb-/bcm2710-rpi-3-b-plus.dtb' prevents FLASH-KERNEL from creating as symbolic link

AradoSKY, 2021-04-25:
**Is this the right place for my bug report?**
This repository contains the GPU firmware used on the Raspberry Pi. This software is the closed source part of the Raspberry Pi system, it includes booting (including network booting and USB booting), low-level power and clock control, FKMS and legacy HDMI control (not full KMS that is in the linux kernel), hardware legacy codecs (MPEG2, H264 and VC1), encode hardware including the ISP (image sensor pipeline) and camera control, audio output (analogue and HDMI audio).

FLASH-KERNEL package for RPI 3B+. 

If you believe that the issue you are seeing is within this area, this is the right place. If not, we have other repositories for the linux kernel at [github.com/raspberrypi/linux](https://github.com/raspberrypi/linux) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
FLASH-KERNEL for ubuntu 20.10 and now 20.20 fails to install. 

**To reproduce**
sudo apt-get install upgrade 
Setting up flash-kernel (3.103ubuntu1~20.10.1) ...
Using DTB: bcm2710-rpi-3-b-plus.dtb
Installing /etc/flash-kernel/dtbs/bcm2710-rpi-3-b-plus.dtb into /boot/dtbs//./bcm2710-rpi-3-b-plus.dtb
Taking backup of bcm2710-rpi-3-b-plus.dtb.
Installing new bcm2710-rpi-3-b-plus.dtb.
ln: failed to create symbolic link '/boot/dtb-/bcm2710-rpi-3-b-plus.dtb': Operation not permitted
dpkg: error processing package flash-kernel (--configure):
 installed flash-kernel package post-installation script subprocess returned error exit status 1

**Expected behaviour**
FLASH KERNEL installs flawlessly. 

**Actual behaviour**
Setting up flash-kernel (3.103ubuntu1~20.10.1) ...
Using DTB: bcm2710-rpi-3-b-plus.dtb
Installing /etc/flash-kernel/dtbs/bcm2710-rpi-3-b-plus.dtb into /boot/dtbs//./bcm2710-rpi-3-b-plus.dtb
Taking backup of bcm2710-rpi-3-b-plus.dtb.
Installing new bcm2710-rpi-3-b-plus.dtb.
ln: failed to create symbolic link '/boot/dtb-/bcm2710-rpi-3-b-plus.dtb': Operation not permitted
dpkg: error processing package flash-kernel (--configure):
 installed flash-kernel package post-installation script subprocess returned error exit status 1

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW = 
     Pi3B+
* Which OS and version (`cat /etc/rpi-issue`)? 
  astroberry@astroberry:~$ lsb_release -a
  No LSB modules are available.
  Distributor ID:	Ubuntu
  Description:	Ubuntu 20.10
  Release:	20.10
  Codename:	groovy

* Which firmware version (`vcgencmd version`)?
Oct 29 2020 16:18:54 
Copyright (c) 2012 Broadcom
version a816a344715bd44c8ca62090bf4fa7165dbf49aa (tainted) (release) (start)

* Which kernel version (`uname -a`)?i
astroberry@astroberry:~$ uname -a
Linux astroberry.local 5.4.73-v7+ #1360 SMP Thu Oct 29 15:53:18 GMT 2020 armv7l armv7l armv7l GNU/Linux

**Logs**
If applicable, add the relevant output from `dmesg` or similar.
N/A Console output.
**Additional context**
astroberry@astroberry:~$ ls -all /boot/dtbs//.
total 80
drwxr-xr-x  3 root root  4096 Apr 25 18:09 .
drwxr-xr-x 11 root root 16384 Dec 31  1969 ..
drwxr-xr-x  2 root root  4096 Oct 20  2019 1
-rwxr-xr-x  1 root root 27054 Apr 25 18:09 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x  1 root root 27054 Apr 25 17:42 bcm2710-rpi-3-b-plus.dtb.bak
astroberry@astroberry:~$ ls -all /boot/dtbs/  
total 80
drwxr-xr-x  3 root root  4096 Apr 25 18:09 .
drwxr-xr-x 11 root root 16384 Dec 31  1969 ..
drwxr-xr-x  2 root root  4096 Oct 20  2019 1
-rwxr-xr-x  1 root root 27054 Apr 25 18:09 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x  1 root root 27054 Apr 25 17:42 bcm2710-rpi-3-b-plus.dtb.bak
astroberry@astroberry:~$ ls -all /boot/dtbs//
total 80
drwxr-xr-x  3 root root  4096 Apr 25 18:09 .
drwxr-xr-x 11 root root 16384 Dec 31  1969 ..
drwxr-xr-x  2 root root  4096 Oct 20  2019 1
-rwxr-xr-x  1 root root 27054 Apr 25 18:09 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x  1 root root 27054 Apr 25 17:42 bcm2710-rpi-3-b-plus.dtb.bak
astroberry@astroberry:~$ ls -all /boot/dtbs//.
total 80
drwxr-xr-x  3 root root  4096 Apr 25 18:09 .
drwxr-xr-x 11 root root 16384 Dec 31  1969 ..
drwxr-xr-x  2 root root  4096 Oct 20  2019 1
-rwxr-xr-x  1 root root 27054 Apr 25 18:09 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x  1 root root 27054 Apr 25 17:42 bcm2710-rpi-3-b-plus.dtb.bak
astroberry@astroberry:~$ ls -all /boot/dtbs/  
total 80
drwxr-xr-x  3 root root  4096 Apr 25 18:09 .
drwxr-xr-x 11 root root 16384 Dec 31  1969 ..
drwxr-xr-x  2 root root  4096 Oct 20  2019 1
-rwxr-xr-x  1 root root 27054 Apr 25 18:09 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x  1 root root 27054 Apr 25 17:42 bcm2710-rpi-3-b-plus.dtb.bak
astroberry@astroberry:~$ ls -all /boot/dtb    
total 20
drwxr-xr-x  2 root root  4096 Sep  2  2020 .
drwxr-xr-x 11 root root 16384 Dec 31  1969 ..
astroberry@astroberry:~$ ls -all /boot/dtb-1
total 20
drwxr-xr-x  2 root root  4096 Oct 20  2019 .
drwxr-xr-x 11 root root 16384 Dec 31  1969 ..
astroberry@astroberry:~$ ls -all /boot/dtb- 
total 20
drwxr-xr-x  2 root root  4096 Apr 22 22:54 .
drwxr-xr-x 11 root root 16384 Dec 31  1969 ..
astroberry@astroberry:~$ ls -all /boot/dtbs/1
total 64
drwxr-xr-x 2 root root  4096 Oct 20  2019 .
drwxr-xr-x 3 root root  4096 Apr 25 18:09 ..
-rwxr-xr-x 1 root root 27054 Oct 20  2019 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x 1 root root 27054 Oct 20  2019 bcm2710-rpi-3-b-plus.dtb.bak

XECDesign, 2021-04-26:
Duplicate of https://github.com/raspberrypi/firmware/issues/1501

We have nothing to do with Ubuntu, flash-kernel or their patches to flash-kernel.

juergh, 2021-04-26:
If you want to report a flash-kernel problem, you need to use launchpad: https://launchpad.net/ubuntu/+source/flash-kernel

closed 2021-04-26 by popcornmix