raspberrypi/firmware #1571: arm_freq_min desn't set frequency lower than 600

emanuel-bacigala, 2021-04-26:
arm frequency can't be set lower than 600MHz. Older firmwares (distributed with kernels 5.4.y and older) enables setting arm_freq_min to much lower values.  

board type: CM3,CM3+

vcgencmd version:
Mar 26 2021 16:34:43
Copyright (c) 2012 Broadcom
version edf2e9c318863999c97c50cdb74eee235ede3af5 (clean) (release) (start)

popcornmix, 2021-04-26:
See:
https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md
https://github.com/raspberrypi/firmware/issues/1431
It's not currently supported.

emanuel-bacigala, 2021-04-27:
Ok. I will keep my eye on [https://github.com/raspberrypi/firmware/issues/1431](url).

> Note that reducing this value below the default does not result in any significant power savings and is not currently supported.
According to our measurments on CM3 power saving up to 10% can be achieved. Not much but it also counts.
power draw 34mA@12V with: arm_freq_min=100 sdram_freq_min=100 gpu_freq_min=90 over_voltage_min=-3
power draw 37mA@12V with: arm_freq_min=600 sdram_freq_min=100 gpu_freq_min=90 over_voltage_min=-3

I hope that in near future it will be supported again.

closed 2021-04-27 by emanuel-bacigala