raspberrypi/firmware #1574: after last update, raspistill hangs every 30 min

sferlix, 2021-04-28:
Hi,

I have a Pi4+ with 4GB RAM and Raspbian Lite. This morning I have updated the firmware to the last version. 

Now I have:
`uname -a
Linux AllSkyCamPi 5.10.32-v7l+ #1413 SMP Tue Apr 27 14:43:13 BST 2021 armv7l GNU/Linux`

all of a sudden, my (every-3-minutes job in crontab) raspistill hangs. It works for around 30 minutes and then.. it hangs.
In order to investigate, I have added start_debug=1 into config.txt.

checking with sudo vcdbg log assert, I get:

`04876.873: assert( timeout != 0 ) failed; ../../../../../vcfw/drivers/chip/vciv/2708/clock_2711.c::clock_wait_busy line 2014 rev a48d332
vcdbg_ctx_get_dump_stack: dump_stack failed`

checking with sudo vcdbg log msg, I get:

`425.381: camsubs: Camera found OK
089350.841: imx477_open: read id 1143 succes 0!
116519.256: imx477_open: read id 1143 succes 0!
226123.930: imx477_open: read id 1143 succes 0!
296520.729: imx477_open: read id 1143 succes 0!
476564.772: imx477_open: read id 1143 succes 0!
656604.678: imx477_open: read id 1143 succes 0!
836599.585: imx477_open: read id 1143 succes 0!
1016624.792: imx477_open: read id 1143 succes 0!
1196640.645: imx477_open: read id 1143 succes 0!
1376645.807: imx477_open: read id 1143 succes 0!
1382107.500: vcos_abort: Halting
1384096.620: mmal: fail_destroy: Timed out waiting to destroy ril.camera
`
after this error, if I launch raspistill from command line, it does not respond at all. Ctrl-C is not working. 
I only have to close the SSH session, open a new SSH session and reboot.
After a while, it hangs again.
of course, in my ps -ef I get a lot of raspistill (as they are in crontab every 3 minutes)

After reboot, when it hangs, I get additional messages in log:

`004876.873: assert( timeout != 0 ) failed; ../../../../../vcfw/drivers/chip/vciv/2708/clock_2711.c::clock_wait_busy line 2014 rev a48d332
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1382107.453: assert( *p==0xa55a5aa5 ) failed; ../../../../../vcfw/rtos/common/rtos_common_malloc.c::rtos_pool_aligned_free line 205 rev a48d332
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1384096.690: assert( Timed out waiting to destroy ril.camera ) failed; ../../../../../interface/mmal/vc/ril/mmal_ril.c::fail_destroy line 1471 rev a48d332
vcdbg_ctx_get_dump_stack: dump_stack failed
`

last reboot I had this one:

`debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

004977.204: assert( timeout != 0 ) failed; ../../../../../vcfw/drivers/chip/vciv/2708/clock_2711.c::clock_wait_busy line 2014 rev a48d332
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
2115614.742: assert( object->magic == OBJECT_MAGIC ) failed; ../../../../../helpers/vc_pool/vc_pool.c::vc_pool_offsetof line 461 rev a48d332
vcdbg_ctx_get_dump_stack: dump_stack failed`

If I run 

`sudo vcgencmd get_config int`

it hangs. Ctrl-C does not work.

additional info:

`pi@raspberrypi:~ $ sudo vcgencmd get_config int
arm_freq=1500
audio_pwm_mode=514
config_hdmi_boost=5
core_freq=500
core_freq_min=200
disable_commandline_tags=2
disable_l2cache=1
display_hdmi_rotate=-1
display_lcd_rotate=-1
dvfs=3
enable_gic=1
force_eeprom_read=1
force_pwm_open=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq=500
gpu_freq_min=250
init_uart_clock=0x2dc6c00
lcd_framerate=60
mask_gpu_interrupt0=1024
mask_gpu_interrupt1=0x10000
max_framebuffers=2
over_voltage_avs=1250
pause_burst_frames=1
program_serial_random=1
total_mem=4096
hdmi_force_cec_address:0=65535
hdmi_force_cec_address:1=65535
hdmi_pixel_freq_limit:0=0x11e1a300
hdmi_pixel_freq_limit:1=0x11e1a300
pi@raspberrypi:~ $
`

p.s. I have a different hardware (Pi4 and HQ Camera) but with 5.10.25-v7l+ #1408 SMP and NO issues at all
An idea could be going back to the 5.10.25 and check if it still happens.. 

Please help me..
thanks
frank

6by9, 2021-04-28:
Duplicate of https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=310668

Raspistill command line used is apparently
```raspistill -n -md 2  -o myfile.jpg -e jpg -w 1024 -h 768 -q 70  -a 1024 -ae 26 -a '28/04/2021 17:27 | SR=06:18 | SS=20:24 | MR=22:18 | MS=-- | 97% | Full'```

sferlix, 2021-04-28:
Sorry for duplicating, I did not know it's the same.

I did additional steps. First of all, I've rolled the firmware back to the 5.10.25 version, as it works on a different hardware.

`
pi@AllSkyCamPi:~ $ uname -a
Linux AllSkyCamPi 5.10.25-v7l+ #1408 SMP Mon Mar 22 12:49:24 GMT 2021 armv7l GNU/Linux

`

Unfortunately, the issue stays the same. Anything else to check ?
thanks.
Frank

`pi@AllSkyCamPi:~ $ sudo vcdbg log msg
[...]
268125.217: Using RPI AWB
2448200.417: imx477_open: read id 1143 succes 0!
2448212.117: Using RPI AWB
2453633.707: vcos_abort: Halting`

 `

1913359.390: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
1913421.775: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
2093336.683: assert( temps->last_tone_mapping_params_valid ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_agc.c::isp_tuner_brcm_agc_switch_mode line 2745 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
2093385.633: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
2093447.848: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
2273368.909: assert( temps->last_tone_mapping_params_valid ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_agc.c::isp_tuner_brcm_agc_switch_mode line 2745 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
2273417.337: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
2273481.815: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
2453464.997: assert( temps->last_tone_mapping_params_valid ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_agc.c::isp_tuner_brcm_agc_switch_mode line 2745 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
2453514.938: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
2453577.272: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
2453633.656: assert( q->guardword ==0xa55a5aa5 ) failed; ../../../../../vcfw/rtos/common/rtos_common_malloc.c::rtos_pool_aligned_free line 194 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed

`

sferlix, 2021-04-28:
Please find below additional log info, collected after reboot and no raspistil in crontab.
After this, I will add the raspistill to crontab, then I will wait for the failure and I will post again.
Let me know if I can check for additional info.

login as: pi
pi@brallo.cardanet.it's password:
Linux AllSkyCamPi 5.10.25-v7l+ #1408 SMP Mon Mar 22 12:49:24 GMT 2021 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Apr 28 19:25:49 2021
pi@AllSkyCamPi:~ $
pi@AllSkyCamPi:~ $
pi@AllSkyCamPi:~ $
pi@AllSkyCamPi:~ $ sudo vcdbg log ex

[dozen of messages like this:]
debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

No exceptions found
pi@AllSkyCamPi:~ $ sudo vcdbg log msg
debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)
debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)
004918.301: arasan: arasan_emmc_open
005338.209: brfs: File read: /mfs/sd/config.txt
005339.030: brfs: File read: 1819 bytes
005388.993: HDMI0:EDID error reading EDID block 0 attempt 0
005395.016: HDMI0:EDID error reading EDID block 0 attempt 1
005401.041: HDMI0:EDID error reading EDID block 0 attempt 2
005407.063: HDMI0:EDID error reading EDID block 0 attempt 3
005413.087: HDMI0:EDID error reading EDID block 0 attempt 4
005419.110: HDMI0:EDID error reading EDID block 0 attempt 5
005425.135: HDMI0:EDID error reading EDID block 0 attempt 6
005431.157: HDMI0:EDID error reading EDID block 0 attempt 7
005437.182: HDMI0:EDID error reading EDID block 0 attempt 8
005443.205: HDMI0:EDID error reading EDID block 0 attempt 9
005444.221: HDMI0:EDID giving up on reading EDID block 0
005464.449: HDMI1:EDID error reading EDID block 0 attempt 0
005470.472: HDMI1:EDID error reading EDID block 0 attempt 1
005476.497: HDMI1:EDID error reading EDID block 0 attempt 2
005482.520: HDMI1:EDID error reading EDID block 0 attempt 3
005488.545: HDMI1:EDID error reading EDID block 0 attempt 4
005494.567: HDMI1:EDID error reading EDID block 0 attempt 5
005500.592: HDMI1:EDID error reading EDID block 0 attempt 6
005506.615: HDMI1:EDID error reading EDID block 0 attempt 7
005512.640: HDMI1:EDID error reading EDID block 0 attempt 8
005518.662: HDMI1:EDID error reading EDID block 0 attempt 9
005519.678: HDMI1:EDID giving up on reading EDID block 0
005521.363: brfs: File read: /mfs/sd/config.txt
006784.637: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
006787.254: *** Restart logging
006787.275: brfs: File read: 1819 bytes
006792.769: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
006798.797: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
006804.822: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
006810.848: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
006816.874: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
006822.901: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
006828.927: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
006834.955: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
006840.980: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
006847.008: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
006848.024: hdmi: HDMI0:EDID giving up on reading EDID block 0
006853.069: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
006859.096: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
006865.121: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
[... 41 lines cut ...]
007129.509: dtparam: audio=on
007139.425: brfs: File read: 1819 bytes
007152.581: brfs: File read: /mfs/sd/overlays/vc4-fkms-v3d.dtbo
007169.618: Loaded overlay 'vc4-fkms-v3d'
007209.419: brfs: File read: 1446 bytes
007211.696: brfs: File read: /mfs/sd/cmdline.txt
007211.735: Read command line from file 'cmdline.txt':
007211.749: 'console=serial0,115200 console=tty1 root=PARTUUID=efc8440b-02 rootf                                                                                        stype=ext4 elevator=deadline fsck.repair=yes rootwait'
008254.335: brfs: File read: 121 bytes
008824.571: brfs: File read: /mfs/sd/kernel7l.img
008824.592: Loading 'kernel7l.img' to 0x8000 size 0x66df48
008824.618: Device tree loaded to 0x2eff3b00 (size 0xc4f0)
008837.396: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
013757.797: vchiq_core: vchiq_init_state: slot_zero = 0xded80000, is_master = 1
016965.426: camsubs: Looking for camera 0: i2c_port = 0, led gpio = -1, power en                                                                                        able gpio = 133
017268.720: camsubs: Camera not found
017268.885: camsubs: Looking for camera 0: i2c_port = 0, led gpio = -1, power en                                                                                        able gpio = 133
017569.518: camsubs: Camera found OK
pi@AllSkyCamPi:~ $ sudo vcdbg log assert
debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)
debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)
004914.796: assert( timeout != 0 ) failed; ../../../../../vcfw/drivers/chip/vciv                                                                                        /2708/clock_2711.c::clock_wait_busy line 2014 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
pi@AllSkyCamPi:~ $

g3rmy, 2021-04-28:
Hey Frank,

I am having the exact same problem. It does seem to be related to the latest updates and upgrades as my raspistill code was stable for a year prior to last month. A couple things that I have followed that have helped (but not cured... it still hangs but not as often) are:

Increasing GPU memory: https://www.raspberrypi.org/documentation/configuration/config-txt/memory.md
This (maybe outdated) build solution: https://www.raspberrypi.org/forums/viewtopic.php?t=237974

Like I said, this has improved things but not completely resolved the issue.

sferlix, 2021-04-28:
> Hey Frank,
>
> I am having the exact same problem. It does seem to be related to the latest updates and upgrades as my raspistill code was stable for a year prior to last month. A couple things that I [...]

thanks for replying.

I already have gpu_mem = 256
Not sure about the responsibility of the firmware... I have an additional Pi4 and HQ Camera running with the same firmware and software. I have no issue with that :(

sferlix, 2021-04-29:
Here again. 
I have raised gpu_mem to 1024 and it worked all night long.. Then, at around 10:22am it crashed
Thinking about last days, it was working for weeks without any issue. Then, I replaced the SD Card with a new one, fresh installation of Raspbian Lite. Could these errors be related to the SD Card ?

please, any advice would be appreciated.
Frank

pi@AllSkyCamPi:~/sky $ sudo vcdbg log assert
debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)
... (many many lines like this. I've removed it)
debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)
1468931.146: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed

1648829.099: assert( temps->last_tone_mapping_params_valid ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_agc.c::isp_tuner_brcm_agc_switch_mode line 2745 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed

1648880.463: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed

1648941.105: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed

1828910.726: assert( temps->last_tone_mapping_params_valid ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_agc.c::isp_tuner_brcm_agc_switch_mode line 2745 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed

1828958.006: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed

1829022.623: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed

2007940.183: assert( temps->last_tone_mapping_params_valid ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_agc.c::isp_tuner_brcm_agc_switch_mode line 2745 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed

2007987.136: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed

2008049.588: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed

2008149.450: assert( *p==0xa55a5aa5 ) failed; ../../../../../vcfw/rtos/common/rtos_common_malloc.c::rtos_pool_aligned_free line 205 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed

2010143.383: assert( Timed out waiting to destroy ril.camera ) failed; ../../../../../interface/mmal/vc/ril/mmal_ril.c::fail_destroy line 1471 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed

pi@AllSkyCamPi:~/sky $
pi@AllSkyCamPi:~/sky $
pi@AllSkyCamPi:~/sky $ sudo vcdbg log msg
debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)
...
debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)
004918.301: arasan: arasan_emmc_open
005338.209: brfs: File read: /mfs/sd/config.txt
005339.030: brfs: File read: 1819 bytes
005388.993: HDMI0:EDID error reading EDID block 0 attempt 0
005395.016: HDMI0:EDID error reading EDID block 0 attempt 1
005401.041: HDMI0:EDID error reading EDID block 0 attempt 2
005407.063: HDMI0:EDID error reading EDID block 0 attempt 3
005413.087: HDMI0:EDID error reading EDID block 0 attempt 4
005419.110: HDMI0:EDID error reading EDID block 0 attempt 5
005425.135: HDMI0:EDID error reading EDID block 0 attempt 6
005431.157: HDMI0:EDID error reading EDID block 0 attempt 7
005437.182: HDMI0:EDID error reading EDID block 0 attempt 8
005443.205: HDMI0:EDID error reading EDID block 0 attempt 9
005444.221: HDMI0:EDID giving up on reading EDID block 0
005464.449: HDMI1:EDID error reading EDID block 0 attempt 0
005470.472: HDMI1:EDID error reading EDID block 0 attempt 1
005476.497: HDMI1:EDID error reading EDID block 0 attempt 2
[... 74 lines cut ...]
007211.749: 'console=serial0,115200 console=tty1 root=PARTUUID=efc8440b-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait'
008254.335: brfs: File read: 121 bytes
008824.571: brfs: File read: /mfs/sd/kernel7l.img
008824.592: Loading 'kernel7l.img' to 0x8000 size 0x66df48
008824.618: Device tree loaded to 0x2eff3b00 (size 0xc4f0)
008837.396: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
013757.797: vchiq_core: vchiq_init_state: slot_zero = 0xded80000, is_master = 1
016965.426: camsubs: Looking for camera 0: i2c_port = 0, led gpio = -1, power enable gpio = 133
017268.720: camsubs: Camera not found
017268.885: camsubs: Looking for camera 0: i2c_port = 0, led gpio = -1, power enable gpio = 133
017569.518: camsubs: Camera found OK
... 
1823655.337: imx477_open: read id 1143 succes 0!
1823664.801: Using RPI AWB
2002674.921: imx477_open: read id 1143 succes 0!
2002686.731: Using RPI AWB
**2008149.484: vcos_abort: Halting
2010143.311: mmal: fail_destroy: Timed out waiting to destroy ril.camera**

pi@AllSkyCamPi:~/sky $ sudo vcdbg log ex
debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)
debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)
debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)
...
No exceptions found

JamesH65, 2021-04-29:
gpu_mem = 1024 is much much too big,. Maximum you will ever need on a running (F)KMS Pi4 is 256, probably less. Not related to the issue I suspect, but a waste of memory, and large values of gpu_mem are known to cause other issues.

sferlix, 2021-04-29:
I agree...
I had to reboot the Pi and now it is no longer reachable... As it is in a
remote location, I need to figure out how to go there and restart it.

To me is still a mystery what vcos_abort means and how to fix it.
Hope someone here may help..

6by9, 2021-04-29:
> To me is still a mystery what vcos_abort means and how to fix it.

You've already had that answered on the forum thread (and this why it is frowned upon to start the same conversation in two places.
https://www.raspberrypi.org/forums/viewtopic.php?p=1857993#p1857993

> vcos_abort is the firmware totally giving up as otherwise it'll corrupt memory. Same as BUG_ON in the kernel, it is generally used sparingly, but means something has gone badly wrong.

Asserts from ```rtos_pool_aligned_free``` mean guard words on allocations have been trampled.
```assert( object->magic == OBJECT_MAGIC )``` from ```vc_pool_offsetof``` means that an image that has already been released has been accessed again.

I've tested the latest rpi-update release on an 8GB Pi and HQ camera with your command line. No issues observed so far. Now running on a tight loop....

sferlix, 2021-04-29:
6by9, thanks for the answer, but the info you provided are not helping me to overcome the issue. 
Or maybe they are just unclear to me, as I am not expert like you (that's why I ask for help here).
More specifically:

>  vcos_abort is the firmware totally giving up. [...] something has gone badly wrong.

     why ? A new Raspberry Pi4 with 4GB and Raspbian Lite is just supposed to collect an image with a raspistill every 3 minutes. The task itself should take a few seconds to get done. It worsk for some hours and, all of the sudden... vcos_abort. 
I have another 4Pi 4GB and HQ Camera, same software, no issue. 

> 2.  assert( object->magic == OBJECT_MAGIC ) from vc_pool_offsetof means that an image that has already been released has been accessed again.

By who ? each 3 minutes raspistill is supposed to generate an image with a different name (with timestamp). Raspistill hangs, but after 3 minutes a new instance of raspistill is launched, as it is in crontab job.
After this happens, of course, the numer of raspistill process instances in memory grows over time. sudo killall raspistill does not kill pending instance (why?). This is maybe the reason behind the error.

As I am not an expert, if you were me, what would you do ? Anything you would check, try or change ? Should I return the hardware (in this case, camera or Pi board) ?
Thanks
 Frank

6by9, 2021-04-29:
My test rig has now been running for an hour with no issues (no asserts logged other than the early ```clock_wait_busy``` one).
I'm fairly happy that this is therefore not going to be a generic issue.

You're using raspistill with the preview disabled. Is the camera connection actually reliable? Check the preview image is stable.

You are NOT running the latest firmware.
```
1468931.146: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed

1648829.099: assert( temps->last_tone_mapping_params_valid ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_agc.c::isp_tuner_brcm_agc_switch_mode line 2745 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed
```
Both those asserts were removed from the firmware source tree on 25th March, and have been in all rpi-update releases since 26th March (https://github.com/Hexxeh/rpi-firmware/commit/c0cf93a133dab106439c208f03d32155bc19e432)
What is the output of ```vcgencmd version```?
Take a clean SD card and retest on a fresh install.

sferlix, 2021-04-29:
Thank you 6by9.

Initially, I had the problem once per day. It was the release 5.17.10
So I decided to update to the last firmware: 5.10.32
The issue became so frequent (once per hour!) that I decided to roll back to the same firmware (5.10.25) version that I have in a similar hardware that is running the same software with zero issues. 

pi@AllSkyCamPi:~ $ vcgencmd version
Mar 19 2021 12:06:19
Copyright (c) 2012 Broadcom
version f66a788996046061ec93f246ffd5856f04eb8f38 (clean) (release) (start_db)

pi@AllSkyCamPi:~ $ raspistill -v

"raspistill" Camera App (commit 4a0a19b88b43 Tainted)

Camera Name imx477
Width 4056, Height 3040, filename (null)
Using camera 0, sensor mode 0

GPS output Disabled

Quality 85, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Time delay 5000, Timelapse 0
Link to latest frame enabled no
Full resolution preview No
Capture method : Single capture

Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Flicker Avoid Mode 'off'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
Camera component done
Encoder component done
Starting component connection stage
Connecting camera preview port to video render.
Connecting camera stills port to encoder input port
Closing down
Close down completed, all components disconnected, disabled and destroyed

sferlix, 2021-05-06:
Hi,

I did a new, fresh installation on a completely different Pi4 board with same 4GB RAM and Raspbian Lite and a different Pi HQ Camera.

I got the Pi4 frozen after 1 day. Looking at sudo vcdbg log assert, I have found the same message:

1468931.146: assert( pass == ISP_TUNER_PASS_NONE ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_focus.c::master_focus_post_isp_update line 1912 rev f66a788
vcdbg_ctx_get_dump_stack: dump_stack failed

Please, note that there was no vcos_abort, but just the above message.

So I decided to update the firmware to the last version.

Now, the firmware is the following:
```
pi@AllSkyCam2:~ $ sudo vcgencmd version
Apr 30 2021 15:54:43
Copyright (c) 2012 Broadcom
version 3e104c5ed69dc796755b7195973a8d00482d76dd (clean) (release) (start_db)

```

```
pi@AllSkyCam2:~ $ uname -a
Linux AllSkyCam2 5.10.33-v7l+ #1415 SMP Fri Apr 30 15:50:57 BST 2021 armv7l GNU/Linux
pi@AllSkyCam2:~ $

```
looking at the logs, I can see this messages:

```
pi@AllSkyCam2:~ $ sudo vcdbg log assert
005644.796: assert( timeout != 0 ) failed; ../../../../../vcfw/drivers/chip/vciv/2708/clock_2711.c::clock_wait_busy line 2014 rev 3e104c5
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
077297.828: assert( ok ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_lens_shading.c::ls_block_till_thread_done line 3094 rev 3e104c5
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
242544.393: assert( ok ) failed; ../../../../../middleware/ISP/tuner/isp_tuner_brcm_lens_shading.c::ls_block_till_thread_done line 3094 rev 3e104c5
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
```
Not sure what this means. Any insight ?

I will let the Pi4 working and let's see if it hangs again.
Thanks in advance.

clorenz150, 2021-06-01:
I had this problem also with intervals that became shorter and shorter. At last it stopped working within just a few minutes.
Updating to the latest firmware (5.10.39-v7l+) seems to have fixed it. It now ran for more than 20 hours, starting raspistill every 15 seconds.

sferlix, 2021-06-01:
I just run sudo apt update and sudo apd update. I got the version 

`Linux raspberrypi 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux`

According to [raspberry.org](https://www.raspberrypi.org/documentation/raspbian/applications/rpi-update.md), this is last stable version. Wondering if I should take the risk to install something that is not considered stable...

sferlix, 2021-06-01:
> I have raised gpu_mem to 1024 and it worked all night long.. Then, at around 10:22am it crashed

and then the Pi was no longer reachble.
I managed to reach the remote site and I discovered that the Pi was in Kernel Panic (because of the high level of GPU MEM).
Now I have a fresh install and I am still facing similar errors like this:

`May 31 02:58:55 raspberrypi kernel: (53921.669061) WARNING: CPU: 2 PID: 27628 at drivers/firmware/raspberrypi.c:64 rpi_firmware_transaction+0xec/0x128
May 31 02:58:55 raspberrypi kernel: (53921.669077) Firmware transaction timeout`

The discussion is [here](https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=312079&p=1872140#p1872140)
So I would close this issue and would invite contributors to join the previous link for supporting me.

closed 2021-06-01 by sferlix

thomasjungblut, 2021-11-25:
Was there a definitive solution to this problem?

I have slightly different hardware (https://electreeks.de/shop/raspberry-pi-kamera-nachtsicht-mit-infrarot-sperrfilter-175-grad-weitwinkel-objektiv/) and I'm using uv4l to stream motion jpg or the raw h264 stream.

After some time, I'm also suffering from the  `vcos_abort: Halting` problem.

I'm already on the latest firmware:
```
sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Your firmware is already up to date (delete /boot/.firmware_revision to force an update anyway)
pi@campi:~ $ cat /boot/.firmware_revision
6a5207946edcd45813d1dd1572ca8bd8101b68b6
```

```
$ vcdbg log msg

015425.849: camsubs: Looking for camera 0: i2c_port = 0, led gpio = -1, power enable gpio = 133
015426.572: camsubs: Camera not found
015426.720: camsubs: Looking for camera 0: i2c_port = 0, led gpio = -1, power enable gpio = 133
015428.356: camsubs: Camera found OK
3909827.988: mmal: mmal_vll_load: could not load VLL 'videnc.vll': 
4070848.121: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
770874.390: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
771044.659: vcos_abort: Halting
773032.575: mmal: fail_destroy: Timed out waiting to destroy ril.camera
```

```
$ vcdbg log assert

004992.850: assert( timeout != 0 ) failed; ../../../../../vcfw/drivers/chip/vciv/2708/clock_2711.c::clock_wait_busy line 2152 rev 99c5aad
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
771044.614: assert( *p==0xa55a5aa5 ) failed; ../../../../../vcfw/rtos/common/rtos_common_malloc.c::rtos_pool_aligned_free line 205 rev 99c5aad
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
773032.647: assert( Timed out waiting to destroy ril.camera ) failed; ../../../../../interface/mmal/vc/ril/mmal_ril.c::fail_destroy line 1471 rev 99c5aad
vcdbg_ctx_get_dump_stack: dump_stack failed
----------------
```

```
$ vcdbg log ex

771033.050: Exception #1: misaligned address

vcdbg_ctx_get_dump_stack: dump_stack failed
  Registers:
     r0:bf191398  r1:bf355928  r2:00000001  r3:40000006  r4:ffffffff  r5:4b356d46  r6:bf355928  r7:bf191398 
     r8:bf191398  r9:00000000 r10:3eef69c4 r11:bf355930 r12:3f336040 r13:3f336140 r14:3f336240 r15:bf3560b0 
    r16:3f336080 r17:3f3365a0 r18:77777777 r19:77777777 r20:77777777 r21:77777777 r22:77777777 r23:77777777 
     gp:3eef31e0  sp:3f335dcc  lr:3ecc613a usp:00000000 ssp:3f335dcc  tp:3ef9290c  sr:40000000  pc:3ec49568 

771033.063: Exception #2: misaligned address

vcdbg_ctx_get_dump_stack: dump_stack failed
  Registers:
     r0:4b356d46  r1:ffffffff  r2:ffffffff  r3:ffffffff  r4:ffffffff  r5:4b356d46  r6:bf355928  r7:bf191398 
     r8:bf191398  r9:00000000 r10:3eef69c4 r11:bf355930 r12:3f336040 r13:3f336140 r14:3f336240 r15:bf3560b0 
    r16:3f336080 r17:3f3365a0 r18:77777777 r19:77777777 r20:77777777 r21:77777777 r22:77777777 r23:77777777 
     gp:3eef31e0  sp:3f335dcc  lr:3ec4956e usp:00000000 ssp:3f335dcc  tp:3ef9290c  sr:40000000  pc:3ee03afe 

771033.070: Exception #3: misaligned address

vcdbg_ctx_get_dump_stack: dump_stack failed
[... 503 lines cut ...]
mask_gpu_interrupt0=1024
mask_gpu_interrupt1=0x10000
max_framebuffers=2
over_voltage_avs=-10000
pause_burst_frames=1
program_serial_random=1
total_mem=4096
hdmi_force_cec_address:0=65535
hdmi_force_cec_address:1=65535
hdmi_pixel_freq_limit:0=0x11e1a300
hdmi_pixel_freq_limit:1=0x11e1a300
device_tree=-
overlay_prefix=overlays/
hdmi_cvt:0=
hdmi_cvt:1=
hdmi_edid_filename:0=
hdmi_edid_filename:1=
hdmi_timings:0=
hdmi_timings:1=

cmdline.txt
-----------
coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=E4:5F:01:2A:86:53 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=50d7f66d-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait

```