raspberrypi/firmware #1575: Serial console unavailable when booting Raspberry Pi 4B over network

manuel-wagesreither, 2021-05-07:
# Problem description

When booting my Raspberry Pi 4B using a USB drive, I receive diagnostic information via UART while booting and can login through said UART afterwards. When netbooting the device and providing the exact same set of boot files (verified with `md5sum`) through TFTP, the device stops to send data through UART early during boot.

It seems to me, some part of the firmware forces the kernel parameter `8250.nr_uarts` to `0` when netbooting. Even when I set `8250.nr_uarts=1` in `cmdline.txt`, I get `8250.nr_uarts=0` in `/proc/cmdline` of the netbooted system. This is not the case when usb booting the device.

# Expected behaviour
While netbooting, the kernel sends diagnostic information over serial. After boot, it provides a console interface over serial interface.

# Actual behaviour
Kernel neither sends diagnostic information, nor does it provide a console interface over UART when netboot is used.

# How to reproduce

* Configure the device for (1) serial access and (2) netboot with fallback to usb boot.
* Connect your development host to the UART of the target and boot it via usb. Check reception of kernel boot messages and prompt. Check `/proc/cmdline` and notice that `8250.nr_uarts=1`.
* Copy the content of the USB drives boot partition to your development host and [make it available through TFTP](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/net_tutorial.md).
* Leave the USB drive in the Raspi. It still needs it for the rootfs.
* Netboot the device. Observe the reception of data through UART stops at some point. Check `/proc/cmdline` and notice that `8250.nr_uarts=0`.

# System information

* Model: Raspberry Pi 4B
* OS: Custom GNU/Linux distribution created with Yocto using [meta-raspberrypi](http://git.yoctoproject.org/cgit/cgit.cgi/meta-raspberrypi)
* Firmware version as per `vcgencmd version`: `Feb 25 2021 12:10:40 version 564e5f9b852b23a330b1764bcf0b2d022a20afd0 (clean) (release) (start)`
* Kernel version: `Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux`

# Configuration

`cmdline.txt`:
`dwc_otg.lpm_enable=0 console=tty1 console=serial0,115200 root=/dev/sda2 rootfstype=ext4 rootwait 8250.nr_uarts=1`

`config.txt`:
```
dtoverlay=vc4-fkms-v3d
arm_64bit=1
enable_uart=1
uart_2ndstage=1
```

# Logs

`/proc/cmdline` of netbooted device:
```
coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1  smsc95xx.macaddr=DC:A6:32:B8:04:5C vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 root=/dev/sda2 rootfstype=ext4 rootwait 8250.nr_uarts=0
```
`/proc/cmdline` of usb booted device:
```
coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1  smsc95xx.macaddr=DC:A6:32:B8:04:5C vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 root=/dev/sda2 rootfstype=ext4 rootwait 8250.nr_uarts=1
```

Serial mappings on both the netbooted and usb booted system
`/proc/device-tree/aliases/serial0`: `/soc/serial@7e215040pi`
`/proc/device-tree/aliases/serial1`: `/soc/serial@7e201000`

Received UART data when netbooting:
```
PM_RSTS: 0x00000020
RPi: BOOTLOADER release VERSION:d6d82cf9 DATE: Feb 16 2021 TIME: 13:23:40 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1613481816 0x75633c6c 0x00d03114 0x000a53fa
PM_RSTS: 0x00000020
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 3200
PCI reset
PCI reset
VLI: HUB2: 0xfff00000 0x24e6 MCU: 0xfff20000 0x15218
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
[... 258 lines cut ...]
[...]
[    0.191679] Serial: AMBA PL011 UART driver
[    0.204757] raspberrypi-firmware soc:firmware: Attached to firmware from 2020-08-24 18:50, variant start
[    0.208769] raspberrypi-firmware soc:firmware: Firmware hash is 8d19c1935e67d8ac2fe87c639d4e8b8ef250813b
[...]
[    1.197740] Serial: 8250/16550 driver, 1 ports, IRQ sharing enabled
[...]
[    1.378234] uart-pl011 fe201000.serial: cts_event_workaround enabled
[    1.379274] fe201000.serial: ttyAMA0 at MMIO 0xfe201000 (irq = 14, base_baud = 0) is a PL011 rev2
[    1.386169] printk: console [ttyS0] disabled
[    1.387233] fe215040.serial: ttyS0 at MMIO 0x0 (irq = 15, base_baud = 62500000) is a 16550
[    2.611908] printk: console [ttyS0] enabled
[...]
[  OK  ] Started Getty on tty1.
[  OK  ] Started Serial Getty on ttyS0.
[...]

Bora Horza Gobuchul (Poky-based) 1.0.0 raspberrypi4-64 ttyS0

raspberrypi4-64 login:
```

# Additional tags for search indexer
* PXE
* BOOTP

manuel-wagesreither, 2021-05-07:
The netboot firmware seems to take `enable_uart=1` into account only when the whole string resides below the `0x1000` boundary. It gets ignored if the final `1` character is at ≥ `0x1000` of `config.txt`. This does not apply when usb-booting the device.

Here's a [config.txt](https://github.com/raspberrypi/firmware/files/6444337/config.txt) which boots the device but will result in `8250.nr_uarts=0`. Serial console will not work.
```
00000fe0  61 64 20 74 68 65 20 45  44 49 44 20 64 61 74 61  |ad the EDID data|
00000ff0  20 66 72 0a 65 6e 61 62  6c 65 5f 75 61 72 74 3d  | fr.enable_uart=|
00001000  31 0a 23 23 20 20 20 20  20 64 65 76 69 63 65 0a  |1.##     device.|
00001010  23 23 0a 23 23 20 20 20  20 20 56 61 6c 75 65 20  |##.##     Value |
```

When you remove any character before the `enable_uart=1` at line 119, `8250.nr_uart` will be `1` and serial console will work as expected.

```
00000fe0  61 64 20 74 68 65 20 45  44 49 44 20 64 61 74 61  |ad the EDID data|
00000ff0  20 66 0a 65 6e 61 62 6c  65 5f 75 61 72 74 3d 31  | f.enable_uart=1|
00001000  0a 23 23 20 20 20 20 20  64 65 76 69 63 65 0a 23  |.##     device.#|
```

adudek16, 2021-06-16:
The is a difference in the two lines. The top one (broken) is  
 `fr.enable_uart=1`  
The bottom on is   
`f.enable_uart=1'  

Looks like there is a typo, but it probably should be able to handle if the line is >_ 0x1000

Malvineous, 2021-07-04:
@adudek16 The `.` here is used because the 0x0A character (a newline) cannot be printed in the hex view.  Here the `.` is where the line starts, so they both say `enable_uart=1`, there is no typo.

caipiblack, 2022-06-27:
I had the exact same problem with RaspberryPI3 booting over PXE and your issue saved my time.

I confirm that if we remove all of the comments in config.txt file, it works fine. 

Did someone know where is the limit of 0x1000 exactly ? Raspberry PI bootcode ?

JamesH65, 2022-06-27:
IIRC its somewhere deep down in the file handling code in the bootloader, which we haven't touched in ten years, so people are reluctant to make changes to it. Could be wrong, sure someone will correct me if so!

albireotech, 2023-10-19:
Thank you for posting this issue - it's a couple years later, but the same/similar issue still appears to be present with the serial console during a PXE boot, and shortening the config.txt resolved it for me as well, so this report was a real lifesaver.  Your attention and diligent work tracking this down saved me a bunch of time, so again, thank you very much!