raspberrypi/firmware #1578: SSH file in boot and keys

lucasfolino, 2021-05-12:
I am not sure if this is a Linux issue or a Raspberry Pi issue.  My first question is what does the SSH in boot actually do?  I know it indicates to turn SSH on for the Raspberry PI during install.  What is the mechanism that it does that?  Is it possible to make it if the file is not empty.  Say with a private key inside that during install it would use that private key for the SSH server and not generate a new one.  My use case is I am writing documentation on Linux/Pi how tos and updating my local keys is a pain.  Is there another way to accomplish this without generating a new key, log in and replace that key?  FOr me to do that I have to clear my local keys, log into the pi.  Replace it with the new key, then reconnect.  The other option is just replacing the old keys with new ones for passwordless log in, but it is about he same amount of work.

6by9, 2021-06-08:
Use Pi Imager.
Press Ctrl-Shift-X to get into the advanced options. 
Enable SSH, and provide it with the authorized_keys file for the user "pi". (Configure wifi too if you wish).

lucasfolino, 2021-06-08:
Authorized keys is not the private key that is generated when the system is created.  Although this is useful to add authorized keys I don't think it fits the use case I am looking for.  Being able to drop a file that is stored elsewhere when I reinstall would be very convenient for restore or reimage.  This would allow me to backup the key in an ssh file and if I restore the pi I could just drop this file in with not only enabling ssh but restoring the key as needed.

6by9, 2021-06-08:
Apologies - misread keys as being key for remote access, not private key.

PiBakery https://www.raspberrypi.org/blog/pibakery/ is another potential option. Or ask for it to be added to Pi Imager  (https://github.com/raspberrypi/rpi-imager/issues)

None of this is a function of the firmware, it's part of the Raspberry Pi OS distro.
Handling of the ssh file was added in https://github.com/RPi-Distro/raspberrypi-sys-mods/commit/76931b0bcba3a5adc6f8f245fa8485884bc2400e