raspberrypi/firmware #1585: Camera stopped working. assert( *p==0xa55a5aa5 ) failed

arkanoid87, 2021-06-03:
**Describe the bug**
camera stopped working

**To reproduce**
can't reproduce

**Expected behaviour**
camera keeps working

**Actual behaviour**
camera stopped working without apparent reason. "raspinfo" and "vcgencmd get_camera" hang forever. "vcdgb log assert" show error

**System**
```
System Information
------------------

Raspberry Pi 3 Model A Plus Rev 1.0
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2021-01-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 21090519d85bdaa1615d5d5057d37b09368ea5d2, stage2                                                                                                    

Linux 66cda39a 5.10.11-v7+ #1399 SMP Thu Jan 28 12:06:05 GMT 2021 armv7l GNU/Linux
Revision        : 9020e0
Serial          : 0000000066cda39a
Model           : Raspberry Pi 3 Model A Plus Rev 1.0
```
then hangs forever. Raspinfo doesn't quit.

**vcdbg log msg**
```
001335.782: brfs: File read: /mfs/sd/config.txt
001336.312: brfs: File read: 224 bytes
001389.474: HDMI0:EDID error reading EDID block 0 attempt 0
001390.742: HDMI0:EDID error reading EDID block 0 attempt 1
001392.006: HDMI0:EDID error reading EDID block 0 attempt 2
001393.273: HDMI0:EDID error reading EDID block 0 attempt 3
001394.539: HDMI0:EDID error reading EDID block 0 attempt 4
001395.805: HDMI0:EDID error reading EDID block 0 attempt 5
001397.069: HDMI0:EDID error reading EDID block 0 attempt 6
001398.335: HDMI0:EDID error reading EDID block 0 attempt 7
001399.600: HDMI0:EDID error reading EDID block 0 attempt 8
001400.867: HDMI0:EDID error reading EDID block 0 attempt 9
001401.889: HDMI0:EDID giving up on reading EDID block 0
001403.362: brfs: File read: /mfs/sd/config.txt
002250.947: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
002253.138: *** Restart logging
002253.165: brfs: File read: 224 bytes
002254.043: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
002255.313: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
002256.583: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
002257.853: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
002259.121: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
002260.391: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
002261.660: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
002262.930: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
002264.198: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
002265.469: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
002266.495: hdmi: HDMI0:EDID giving up on reading EDID block 0
002266.830: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
002268.100: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
002269.369: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
002270.639: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
002271.907: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
002273.178: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
002274.447: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
[... 106 lines cut ...]
[443108.036099]  r10:ffffe000 r9:00000000 r8:81bcc3b8 r7:81fb2910 r6:000a001f r5:81e1b088
[443108.036112]  r4:82c39cc0
[443108.036135] [<803310b0>] (__fput) from [<80331324>] (____fput+0x18/0x1c)
[443108.036157]  r9:00000000 r8:80f48d1c r7:8ae4be00 r6:8ae4c354 r5:8389c840 r4:82c39cc0
[443108.036180] [<8033130c>] (____fput) from [<8014053c>] (task_work_run+0xb4/0xe4)
[443108.036203] [<80140488>] (task_work_run) from [<80123744>] (do_exit+0x468/0xb58)
[443108.036225]  r9:00000000 r8:8265f448 r7:8ac4de94 r6:ffffe000 r5:8265f400 r4:8ae4be00
[443108.036247] [<801232dc>] (do_exit) from [<80123ed0>] (do_group_exit+0x50/0xe0)
[443108.036261]  r7:862ef8c0
[443108.036284] [<80123e80>] (do_group_exit) from [<80131500>] (get_signal+0x1c0/0xad0)
[443108.036302]  r7:862ef8c0 r6:00000008 r5:00000000 r4:00418004
[443108.036324] [<80131340>] (get_signal) from [<8010b620>] (do_work_pending+0x14c/0x5a8)
[443108.036346]  r10:80e05008 r9:fffffe30 r8:5ac3c35a r7:8ac4df44 r6:76e78518 r5:8ac4dfb0
[443108.036359]  r4:ffffe000
[443108.036379] [<8010b4d4>] (do_work_pending) from [<80100074>] (slow_work_pending+0xc/0x20)
[443108.036393] Exception stack(0x8ac4dfb0 to 0x8ac4dff8)
[443108.036411] dfa0:                                     fffffffc c01cc402 7e800464 00010001
[443108.036431] dfc0: 76f79404 76f79308 7e8004a8 00000036 0001169c 00022634 ffffffff 00000003
[443108.036450] dfe0: 76f79240 7e800454 76f65928 76e7851c 60000010 00000003
[443108.036471]  r10:00000036 r9:8ac4c000 r8:80100204 r7:00000036 r6:7e8004a8 r5:76f79308
[443108.036485]  r4:76f79404
```

**Additional context**
camera has been running for 5 days without issues before stopping. The raspberry has not been physically touched in the past week.

arkanoid87, 2021-06-03:
after reboot camera works again. Here you find the complete [raspinfo](https://github.com/raspberrypi/firmware/files/6588593/raspinfo.txt) output.

6by9, 2021-06-04:
```vcos_abort: Halting``` is the firmware having encountered an unrecoverable situation.

```assert( *p==0xa55a5aa5 ) failed; ../../../../../vcfw/rtos/common/rtos_common_malloc.c::rtos_pool_aligned_free line 205``` is that a guard word on an allocation wasn't there when the object was freed, most likely meaning something overran the allocation.

Which version of camera module was this? OV5647 v1, IMX219 v2, or IMX477 HQ?
What application were you using to access the camera module? raspistill, raspivid, V4L2, picamera, or something else?

arkanoid87, 2021-06-07:
It's IMX219.
I'm accessing it with rpicamsrc https://github.com/thaytan/gst-rpicamsrc that's literally a "GStreamer wrapper around the raspivid/raspistill". It is an isolated case, I've more than one pi running with same software & hardware and only one (so far) explosed this. Actually not 100% sure if os libs are same version, but they came from raspberry os.