raspberrypi/firmware #1590: Feature request: Put individual cores to low power on BCM2711 based Pi's

p-paule, 2021-06-17:
Dear Community,

**enabling the option to put individual cores on BCM2711 based Pi's into a low power mode would enable battery or USB powered applications for the CM4 and Pi 4B.**

What I know so far:
- With some modifications to the bootloader it seems to be possible to get ALL cores into low power on HALT (POWER_OFF_ON_HALT=1)
(See https://raspberrypi.stackexchange.com/questions/114422/what-is-the-minimum-power-required-for-an-rpi-4-in-halt-or-shutdown)
- Also eg /boot/cmdline.txt maxcpus=1 will disable all but one core but this seems not to do the trick as the disabled cores still consume power

**The requested feature would achieve the same as HALT but with individual cores.** Eg. switch all cores into low power but one.

pelwell, 2021-06-17:
You are saying that you would like to be able to make individual cores idle in way that saves a significant amount of power. How do you propose we do this? Is there a feature of the BCM2711 that makes this possible?

p-paule, 2021-06-17:
Hi @pelwell, thanks for the quick reply.
I'm trying to figure out if this is possible within the SOC and PMIC. There could be a hardware limitation that only the whole SOC can be powered down, but not individual cores. So any insights from someone experienced like you would be very welcome.

pelwell, 2021-06-18:
It's not possible.

closed 2021-06-18 by pelwell

pelwell, 2021-06-18:
https://www.raspberrypi.org/forums/viewtopic.php?p=1878876#p1878876

Was @timg236's answer not authoritative enough for you? Cross-posting wastes time and reduces your chances of getting help in the future.

p-paule, 2021-06-18:
Understood @pelwell, sorry about that.