raspberrypi/firmware #1591: kms locks up with 4096x2160-60 unless overclocked

alanbork, 2021-07-03:
modetest -s 89:4096x2160-60

causes kms to lock up unless you overlock the pi4 in config.txt.  I thought that you had a system set up to dynamically overclock the CPU when the video mode required it? perhaps 4k60p didn't get added to that list yet.

error message is:

[ 592.431498] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] ERROR [CRTC:76:crtc-3] flip_done timed out

only way to recover is to ssh in remotely and kill modetest.

this is a long-standing issue and still observed as of f29ab05611eef385d17675c77190df0b87a0d456,

alanbork, 2021-07-03:
this used to be an issue for the "other" 4k as well but has since been fixed:https://github.com/raspberrypi/firmware/issues/1568

popcornmix, 2021-07-05:
Can you confirm if you are reporting that `3840x2160-60` is working and `4096x2160-60` still locks up, or if both are working now?

alanbork, 2021-07-07:
3840 still works, 4096 still locks up; never even gets a single flip in.

Am I imagining that a feature was added for dynamic overclocking, for
exactly this reason?

***@***.***:~# kmsmodeline 0
/dev/dri/card0 does not have DRM resources, using card1
3840x2160-60.00hz(60): 594.000000 3840 4016 4104 4400 2160 2168 2178 2250
+hsync

 +vsync
Initialized EGL version: 1.4
GL Viewport size: 3840x2160
16.67 ms        60.00 hz
16.67 ms        60.00 hz
16.67 ms        60.01 hz
16.67 ms        59.99 hz
16.67 ms        60.00 hz
16.67 ms        60.00 hz
16.67 ms        60.00 hz
16.66 ms        60.01 hz
16.67 ms        60.01 hz
16.67 ms        59.99 hz
16.67 ms        60.00 hz
16.67 ms        60.00 hz
16.68 ms        59.95 hz
16.65 ms        60.05 hz
16.67 ms        60.01 hz
16.67 ms        60.00 hz
16.67 ms        59.99 hz
16.66 ms        60.01 hz
16.67 ms        60.00 hz
16.66 ms        60.01 hz
16.67 ms        60.00 hz
16.67 ms        59.99 hz
16.67 ms        60.00 hz
16.67 ms        60.01 hz
16.67 ms        60.00 hz
16.67 ms        60.00 hz
16.66 ms        60.01 hz
16.67 ms        59.98 hz
16.67 ms        60.00 hz
16.67 ms        60.00 hz
16.67 ms        60.00 hz
16.67 ms        60.00 hz
16.66 ms        60.01 hz
16.67 ms        59.99 hz
16.67 ms        60.00 hz
16.67 ms        60.00 hz
16.67 ms        60.00 hz
16.67 ms        60.01 hz
16.67 ms        60.01 hz
16.67 ms        59.99 hz
16.67 ms        60.00 hz
16.66 ms        60.01 hz
16.67 ms        60.00 hz
16.67 ms        60.01 hz
16.67 ms        60.01 hz
16.67 ms        59.99 hz
16.67 ms        60.00 hz
16.67 ms        60.00 hz
16.67 ms        60.00 hz
16.67 ms        60.01 hz
16.66 ms        60.01 hz
16.67 ms        59.98 hz
16.66 ms        60.01 hz
16.67 ms        59.99 hz
16.67 ms        60.00 hz
16.67 ms        60.01 hz
16.67 ms        60.01 hz
16.67 ms        59.99 hz
------------------------
16.6665 ms      60.0004 hz (mean)
16.6667 ms      60.0000 hz (ideal)
***@***.***:~# kmsmodeline 1
/dev/dri/card0 does not have DRM resources, using card1
4096x2160-60.00hz(60): 594.000000 4096 4184 4272 4400 2160 2168 2178 2250
+hsync

 +vsync
Initialized EGL version: 1.4
GL Viewport size: 4096x2160
[hang]