raspberrypi/firmware #1592: Bootloader bug - DHCP debug messages - read beyond end of string

Malvineous, 2021-07-03:
When doing a network boot on a Pi 4 with the serial console enabled, messages such as the following appear:
```
00000000  52 61 73 70 62 65 72 72  79 20 50 69 20 42 6f 6f |Raspberry Pi Boo|
00000010  74 ff 3c 09                                      |t.<.            |
[43:9]: 'Raspberry Pi Boot?<    PXEClient?'

00000000  52 61 73 70 62 65 72 72  79 20 50 69 20 42 6f 6f |Raspberry Pi Boo|
00000010  74 ff 0f 10                                      |t...            |
[43:9]: 'Raspberry Pi Boot?example.com?'
```
It looks like the debug message is printing the DHCP response without ensuring it is null-terminated first, so whatever is following it in memory is also being printed unintentionally.

I imagine the security implications of this are not high, but I suppose it's conceivable that a malicious DHCP server could return a crafted response that causes the bootloader to behave unexpectedly.

**System**
* Which model of Raspberry Pi? 4B
* Which firmware version? RPi: BOOTLOADER release VERSION:c2f8c388 DATE: Apr 29 2021 TIME: 17:11:29 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1619712685 0x40bd2946 0x00c03111 0x000a46

timg236, 2021-07-05:
Thanks for the bug report. DHCP menu options don't have to be null-terminated and the hexdump is just printing the contents of that field to assist debug (of non-null terminated options from the DHCP server).
The "[43:0]" message is harmless but confusing so we'll just get rid of it.

Malvineous, 2021-07-25:
Just as another note to this, I noticed that the Pi4 graphical bootloader also suffers from this issue.  While it's displaying boot diagnostics it has a tendency to print garbage on the screen as it prints data in memory beyond the end of the string.  I guess the bootloader also needs to be fixed to either null-terminate the DHCP option before printing it, or to only print the given number of characters instead of printing until it encounters a `\0`.

peterharperuk, 2021-07-26:
I think this bit of on screen boot diagnostics have been fixed at the same time - by removing them completely.

timg236, 2021-07-26:
Yep, this can be closed now

closed 2021-07-26 by pelwell