raspberrypi/firmware #1593: gpu_mem setting chaos

mkreisl, 2021-07-07:
My config.txt

```
initramfs initramfs.gz followkernel
[all]
gpu_mem_1024=224
[pi3]
gpu_mem_1024=226
[pi4]
gpu_mem_1024=256
[all]
```

Booting a Pi3

```
root@kmxbilr2 ~ # vcgencmd get_mem gpu
gpu=256M
root@kmxbilr2 ~ # vcgencmd version
Jun  8 2021 13:09:11 
Copyright (c) 2012 Broadcom
version 03df0e450c356822f03454c27edea347fd8d211b (clean) (release) (start_x)
root@kmxbilr2 ~ # 

```

So my question: why is the gpu_mem value for Pi4 used instead of that of Pi3?

popcornmix, 2021-07-07:
What does `vcgencmd get_mem arm` report?

mkreisl, 2021-07-07:
```
root@kmxbilr2 /lib/modules # vcgencmd get_mem arm
arm=768M
root@kmxbilr2 /lib/modules # vcgencmd get_mem gpu
gpu=256M
root@kmxbilr2 /lib/modules # 
```

mkreisl, 2021-07-07:
It occurs to me that I already had the problem with the gpu_mem many months ago, but didn't pursue it any further.

But since the gpu_mem value is no longer relevant with the Pi4 and Kodi v19, I promptly fell on my face again.

popcornmix, 2021-07-07:
gpu_mem is handled by bootloader rather then firmware (bootcode.bin on pi3). 
@timg236 sections like `[pi3]`should be handled by bootloader?

timg236, 2021-07-07:
Don't know about bootcode.bin - presumably so

mkreisl, 2021-07-07:
Oh, that does indeed seem to be the case

I forgot to tell you that I usually boot my Raspberry Pi's from the network, and for some reason the bootcode.bin is loaded from the TFTP root directory instead of the actual directory for that Pi. 
I had noticed this before, but after 2 years or more, it's easy to forget.
After I updated bootcode.bin, it works as desired.
However, I now have another problem (onboard wifi no longer works with the Pi3) but I have to find out more about that first

closed 2021-07-07 by mkreisl

timg236, 2021-07-07:
@mkreisl Thanks for verifying!