raspberrypi/firmware #1596: bssid / freq_list not respected in wpa_supplicant.conf (assuming this may be an issue with the kernel wifi driver?)

anfractuosity, 2021-07-12:
**Describe the bug**

I set the ssid, bssid, freq_list of my 5GHz AP by doing:

ssid="blah"
bssid=....
freq_list=5220

In my wpa_supplicant.conf

However, the Pi 4B, will at some point fallback from the 5GHz connection to my 2.4GHz AP, even
though it has a different bssid and frequency.

**Expected behaviour**

I would expect after specifying the bssid that the Pi would not connect to alternative bssids.

**Actual behaviour**

Alternative bssids are connected to, than the one specified in wpa_supplicant.conf

**System**

Raspberry Pi 4 Model B Rev 1.2
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

kwinz, 2022-05-20:
I have the same issue. `freq_list` option in `wpa_supplicant.conf` seems to be ignored.

bpsib, 2023-03-30:
My guess is that this is something to do with firmware roaming support. It can be turned off by creating `/etc/modprobe.d/brcmfmac.conf` and add the following:
`options brcmfmac roamoff=1`

I haven't had chance to try this myself so let me know if this fixes the issue.

lobstaj, 2023-05-23:
> My guess is that this is something to do with firmware roaming support. It can be turned off by creating `/etc/modprobe.d/brcmfmac.conf` and add the following: `options brcmfmac roamoff=1`
> 
> I haven't had chance to try this myself so let me know if this fixes the issue.

This works like a charm, thank you very much!

anfractuosity, 2023-07-01:
I've just been testing setting bssid again recently in wpa_supplicant, using the latest kernel/raspberry pi OS.

Even without roamoff=1, it seems to be currently sticking to the 5GHz channel for me.

Will keep my eye on it, to see if it ever changes.

milux, 2023-07-01:
> My guess is that this is something to do with firmware roaming support. It can be turned off by creating `/etc/modprobe.d/brcmfmac.conf` and add the following: `options brcmfmac roamoff=1`
> 
> I haven't had chance to try this myself so let me know if this fixes the issue.

Negative for me. :disappointed:

bpsib, 2023-07-01:
> Negative for me. 😞

@milux Did you reboot? It didn't work for me until I rebooted.

milux, 2023-07-03:
> > Negative for me. 😞
> 
> @milux Did you reboot? It didn't work for me until I rebooted.

Of course I did. I'm aware of the fact that changes to kernel module parameters generally require a reboot to take effect. :wink: 
I tried to force the WiFi module to stick with 5 GHz in client configuration, and failed multiple times.

Meanwhile, I simply accepted that the driver(s) are too buggy to reliably setup WiFi client and WiFi AP via the same network interface, and ordered a separate USB WiFi dongle. :man_shrugging:

jacopotediosi, 2024-02-27:
I had the same problem: my RaspBerry Pi 4B+ kept switching between the two APs with the same ssid, continuously interrupting the connection.

I was about to start debugging the problem. Doing `ps aux | grep wpa_supplicant` I discovered there were two wpa_supplicant processes.

A quick Google search (https://unix.stackexchange.com/questions/568052/is-it-normal-to-have-two-running-wpa-supplicants and https://forums.raspberrypi.com/viewtopic.php?t=292401#p1768589) made me realize that one of the two was useless, and I disabled it with the `sudo systemctl disable wpa_supplicant.service` command.

After a reboot, the problem seems to be solved and now the Raspberry correctly honors the `bssid`, `scan_list `and `freq_list` instructions within the `networks` section in the `/etc/wpa_supplicant/wpa_supplicant.conf` file. Even by launching a scan with the `sudo iw wlan0 scan` command, now only the access point with the 2.4GHz frequency appears, as I had set.

I suspect the service I disabled caused some conflict.

Below is the complete contents of my `/etc/wpa_supplicant/wpa_supplicant.conf` file:
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=IT

network={
        ssid="MY_WIFI_SSID"
        psk=HASH_REDACTED
        scan_freq=2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472
        freq_list=2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472
}
```

VitanovG, 2024-12-29:
> My guess is that this is something to do with firmware roaming support. It can be turned off by creating `/etc/modprobe.d/brcmfmac.conf` and add the following: `options brcmfmac roamoff=1`
> 
> I haven't had chance to try this myself so let me know if this fixes the issue.

I have the exact same issue as above. This does not work. I think the problem is with managed connections.

Ignored settings:
-bssid
-freq_list
-country code

 Router wants to switch to 5GHz from 2.4GHz (in my case) and the PI (3A) allows the switch despite freq_list is specified and bans 5GHz channels. Also despite specifying 2.4GHz bssid only to connect to. I even set the country code to Bangladesh, which should disable all 5GHz channels that my router supports but it still does not prevent the switch to 5GHz on wlan0. Currently there seems to be no solution to force 2.4GHz for a connection. (or 5GHz for that matter). This is a problem, since if you connect to wifi with wlan0 and host an AP on uap0, then the AP will have the same frequency as the wlan0. Since there are a bunch of IOT devices that cannot connect to 5GHz. If 2.4GHz connection for wlan0 is not enforcable then wlan0 and uap0 might switch to 5GHz and all IOT devices are deconnected from the AP.

iw wlan0 set channel 3
returns Operation not supported (-95)