raspberrypi/firmware #1597: Feature - Pi Zero HDMI mode 1036x810

rickyrockrat, 2021-07-15:
This is a request for the following mode for HDMI:
1036x810, interlaced, 875 lines.
Is this possible for the Pi Zero?

mateoconlechuga, 2021-07-23:
That's a weird resolution.

Malvineous, 2021-07-25:
Are you asking for this video mode to be included in the Pi for everyone to use?  That probably won't happen unless it's for a device that is cheap and very popular.

If you just want the mode for yourself, have you read the documentation?  Specifically the part about [custom video modes](https://www.raspberrypi.org/documentation/configuration/config-txt/video.md#custom-mode)?  If you can't get that to work you'd be better off asking for support over on the Raspberry Pi forum first before logging a feature request.

rickyrockrat, 2021-07-29:
Yes. I have since realized I'm not a complete idiot - some parts are missing. :) I apparently have a particular weakness regarding documentation. hdmi_timing seems to work just like the docs. Who knew?

closed 2021-07-29 by rickyrockrat