raspberrypi/firmware #1598: HDMI screen turns off when sustained high CPU usage on all cores

vanfanel, 2021-07-15:
**Describe the bug**
When CPU cores are under heavy usage (4 cores at ~100%, as with building software using `make -j4`) for a certain time, HDMI is turned off.

**To reproduce**
-Update to latest firmware with rpi-update
-Try to build a big C/C++ project via `make -j4`

**Expected behaviour**
Simply build the thing.

**Actual behaviour**
HDMI is turned off.

**System**
Raspberry Pi 4b+ v1.2, 2GB RAM,  

OS version:
```
pi@raspberrypi:~/src/Commander-Genius/b4 $ cat /etc/rpi-issue 
Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 30e2dd32ba47cc3bec15ab1413c16a17e5797775, stage4
```

Firmware version:
```
pi@raspberrypi:~ $ vcgencmd version               
Jul 14 2021 14:20:55 
Copyright (c) 2012 Broadcom
version 1ecd7d49359f3b48737f1a9e33c2f1513f90743d (clean) (release) (start)
```
Kernel version:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.49-v8+ #1436 SMP PREEMPT Wed Jul 14 14:20:10 BST 2021 aarch64 GNU/Linux

```

**Additional context**
It started to happen after updating from 5.10.27 or so. Didn't happen before.

popcornmix, 2021-07-16:
Exact rpi-update version when this started would be useful.
Are you using fkms or kms driver?
Are you saying hdmi output returns when make completes?

vanfanel, 2021-07-16:
@popcornmix Sadly I can't say when this started because I had been months without updating. Any revision you suspect and I can "force"? (Along with how to force it, which I have never done)

I am using KMS.

HDMI does not come back until I reboot.

popcornmix, 2021-07-16:
It's not an issue I've seen myself or seen reported, so can't really guess.
Does dmesg have any errors after this?

popcornmix, 2021-07-16:
What resolution/refresh rate is hdmi monitor? Does a lower one still have the issue?
Does force_turbo=1 help?

vanfanel, 2021-07-18:
@popcornmix 
I cleared dmesg, then ran compilation on all four cores, then just after the monitor goes off, I get these on dmesg:
```

pi@raspberrypi:~ $ dmesg
[   93.150825] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[  103.390806] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
[  113.630833] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:32:HDMI-A-1] flip_done timed out
```

About resolution and refresh rate, I am using this mode in config.txt to force the mode of my choice:
```
hdmi_group=2
hdmi_mode=39
```

`force_turbo=1` has no effect.

`config_hdmi_boost=4` has no effect.

Setting a "secure" video mode like this:
```
hdmi_group=2
hdmi_mode=4
```
...has no effect.

popcornmix, 2021-07-19:
> About resolution and refresh rate, I am using this mode in config.txt to force the mode of my choice:

config.txt settings only affect the initial simple framebuffer mode before kms takes over.
You should be using standard linux methods for configuring the hdmi mode
(e.g. arandr if using X or a "video=" cmdline.txt setting.

What is the resolution/refresh rate you are actually using before you get the "flip_done timeout"?
What does "vcgencmd get_throttled" report after the "flip_done timeout"?

vanfanel, 2021-07-19:
@popcornmix

For these experiments, I have added this to `commandline.txt`

`video=HDMI-A-1:640x480@60`

And I am also setting this in config.txt as I said:
```
hdmi_group=2
hdmi_mode=4

```
So I am, in fact, using the basic 640x480 at 60Hz video mode.
That doesn't change anything (except the console resolution, of course!)

As for the command you asked, this is what it says:
```
pi@raspberrypi:~/src/sm64ex-alo $ vcgencmd get_throttled
throttled=0xe0000
```

popcornmix, 2021-07-19:
> So I am, in fact, using the basic 640x480 at 60Hz video mode.
That doesn't change anything (except the console resolution, of course!)

I'm still not clear if you are saying the "flip done timeout" is occurring at 640x480@60Hz or a higher resolution.
The hdmi mode can be set in many places:
 initially by firmware (affected by hdmi_group/hdmi mode in config.txt)
 by kernel when creating the console for kms/fkms (affected by video=<> in cmdline.txt)
 by user code like X (e.g. setting the last mode configured with arandr).
 by other applications that can do modesetting (e.g. kodi).

what is the resolution you are running at when you get "flip done timeout" and what were you running? (e.g. X etc)

vanfanel, 2021-07-19:
@popcornmix I usually run my Pi at 1360x768, but when I am running what you ask me to run, etc... then I move to basic 640x480@60Hz.
There's nothing wrong with the video mode in use, and it has no impact on the issue, because I have tried a lot of different video modes, from basic 640x480@60Hz to 1080p, and the issue is the same.

So, simply put, I am using 640x480@60 when I report anything on this thread.

I don't have an X server. TTY console, of course, runs on legacy fbdev, just like in every GNU/Linux system as far as I know.
When I do compilation on all four cores, there's NOTHING running except the fbdev TTY console.

So, to be clear: I have no Xorg server. I set the video mode in config.txt or in cmdline.txt. Since you told me to use the "video=..." directive in cmline.txt, that's what I use. That's all.

popcornmix, 2021-07-19:
Does the "flip done" message always correspond to `vcgencmd get_throttled` returning a non-zero value (i.e. throttling occurring)?

vanfanel, 2021-07-19:
@popcornmix What happens is this.
On idle system, or when running, let's say, SDLPop, Scummvm, etc... I normally get this:
```
pi@raspberrypi:~/src/Raze/b4 $ vcgencmd get_throttled
throttled=0x0
```

But after a couple of seconds of building anything with -j3, -j4, etc.. screen goes off, and I get this:
```
pi@raspberrypi:~ $ vcgencmd get_throttled
throttled=0x0
pi@raspberrypi:~ $ vcgencmd get_throttled
throttled=0x80008
pi@raspberrypi:~ $ vcgencmd get_throttled
throttled=0xe0008
pi@raspberrypi:~ $ vcgencmd get_throttled
throttled=0xe0006
pi@raspberrypi:~ $ vcgencmd get_throttled
throttled=0xe0008
pi@raspberrypi:~ $ vcgencmd get_throttled
throttled=0xe0008
pi@raspberrypi:~ $ vcgencmd get_throttled
throttled=0xe0006
pi@raspberrypi:~ $ vcgencmd get_throttled
throttled=0xe0008
pi@raspberrypi:~ $ vcgencmd get_throttled
throttled=0xe0008
pi@raspberrypi:~ $ vcgencmd get_throttled
throttled=0xe0006
pi@raspberrypi:~ $ vcgencmd get_throttled
```

As you can see, initially it returns zero, and then non-zero values as monitor goes off.

It's always the same.

vanfanel, 2021-07-22:
@popcornmix I went back from kernel 5.10.50 to to stable 5.10.17  (remember I am on a full aarch64 Raspberry Pi OS) via:
`sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel`

...and after rebooting, I can do whatever I want on all the CPU cores: no more HDMI display turn-off.

vanfanel, 2021-07-24:
@popcornmix Another thing to note is that I use the vc4-hdmi device (ARM-side ALSA driver).
I have added that information to the first post.

vanfanel, 2021-08-12:
For devs trying to reproduce this issue locally on Pi OS 64bits: simply large build a project in C++ (not plain C) using `make -j4` 
In case you still can't see it happening, do it as root or remove your user's rlimits so you can really cause a 100% CPU usage.

Now 5.10.52 is the "stable" kernel, so it happens again on my system after doing a simple `sudo apt-get update && upgrade`.

popcornmix, 2021-08-13:
@vanfanel can you give this [test firmware](https://drive.google.com/uc?id=1VV_FhKXwA3HGskixkDaI7rMlNbMVQ2vt&export=download) a try?
I think the issue is when arm throttles (due to high temperature) it was incorrectly reducing core frequency below that required for the hdmi mode.

vanfanel, 2021-08-14:
@popcornmix Tested, but I am sorry to say that it's still happening.

popcornmix, 2021-08-16:
Can you report output of `vcgencmd version` when using the test firmware?
Can you confirm that display is fine when you have `throttled=0x0` or `throttled=0x80008`, but occurs when any addional bits are set?

vanfanel, 2021-08-16:
@popcornmix:
```

pi@raspberrypi:~ $ vcgencmd version
Aug 13 2021 13:03:32 
Copyright (c) 2012 Broadcom
version 5ffbdf498f77137ac0fbb2f63214eeb3346a3969 (tainted) (release) (start)
```

Also, when display is fine, I always get:
```
pi@raspberrypi:~ $ vcgencmd get_throttled
throttled=0x0
```

Then after 1 minute building a C++ project with `ninja -j4` or `make -j4` I start seeing:

```
pi@raspberrypi:~ $ vcgencmd get_throttled
throttled=0x80000
```

Then later I see:

```
pi@raspberrypi:~ $ vcgencmd get_throttled
throttled=0x80008
```

**At this point, the display is turned off**
And then, just when display is turned off I see this:

```
pi@raspberrypi:~ $ vcgencmd get_throttled
throttled=0xe0006
```

And then, while the display is still off, I see:

```
pi@raspberrypi:~ $ vcgencmd get_throttled
throttled=0xe0008

```

...from this point (remember: display is off and won't come back even if the compilation finishes OK), these two `0xe0006` and `0xe0008` alternate. Display is off until I reboot.

popcornmix, 2021-08-17:
This is very surprising, as I can reproduce your description easily (you can even set `temp_limit=65` to make it happen more quickly). 

With default firmware if I'm in a 4kp60 mode and I hit the temperature limit (signalled by `THROTTLED_HIGH_TEMP=2` and `THROTTLED_LIMIT_TURBO=4`) then core freq gets lowered to 200MHz which isn't enough to sustain 4kp60 and we lose (permanently) display output.

However with the test firmware we no longer limit the core frequency and this issue doesn't occur.

I've just reproduced this on a different Pi4 and it still fixes it.

Can you post your config.txt and cmdline.txt in case they are having an effect.
Also report `vcgencmd measure_clock core` before and after the hdmi output is lost.

vanfanel, 2021-08-17:
1 - Before display goes off:
```

pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=500000992
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=500000992
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=500000992
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=500000992
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=500000992
```
(Value is stable until display goes off)

2 - After display goes off:

```
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=333333984
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=199995120
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=333333984
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=333333984
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=333333984
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=333333984
pi@raspberrypi:~ $ 
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=333333984
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=333333984
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=499987808
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=500000992
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=500000992
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=500000992
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=500000992
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=500000992
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=333333984
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=200008304
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=333333984
pi@raspberrypi:~ $ vcgencmd measure_clock core
frequency(1)=333333984

```

Now, this is my `config.txt` 
file:///media/manuel/BOOT/config.txt

and this is my `cmdline.txt`
file:///media/manuel/BOOT/cmdline.txt

You will see I am using an slight overclock, but it goes with the corresponding overvoltage. Using official HDMI cable and power supply.

popcornmix, 2021-08-18:
Okay, it's 
```
hdmi_group=2
hdmi_mode=39 #  1360x768  60Hz
```
that stops it working. If you remove that I suspect the issue will be resolved.
Note that `hdmi_mode`/`hdmi_group` (and pretty much all `hdmi_` settings) don't work with the kms driver (which is driven by settings on the arm side).
But I'll try to find out why they don't play well with kms and the temperature limit.

vanfanel, 2021-08-18:
@popcornmix I have removed every `hdmi_*` setting and I am still seeing the issue.
It must be annoying, sorry.

popcornmix, 2021-08-18:
It does seem there are two similar issues here.
The first involves core_freq being set too low when throttling. 
That is easy to reproduce. Use a high clock rate hdmi mode (e.g. 4kp60) and throttle.
That is fixed with test firmware (and now rpi-update).

The second seems to be the M2MC clock (aka hsm clock in kernel driver) being set to 0.
It noticed when clocks change after throttling, but the problem seems to occur earlier, 
and it seems to occur with certain hdmi resolutions (possibly your edid gives the same hdmi mode without the `hdmi_` settings).

vanfanel, 2021-08-18:
@popcornmix You are right, it's only happening with the 1360x758 video mode.
I have tried other resolutions via the "video=..." kernel parameter, and the issue doesn't show it's ugly face.

JamesH65, 2021-08-18:
This thread is worth a read for the 1366x768 mode. https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=284866

popcornmix, 2021-08-18:
@JamesH65 it's 1360x768, and the display is fine up until the first throttle, so I don't think that thread is relevant.

popcornmix, 2021-08-18:
@vanfanel can you try this [test firmware](https://drive.google.com/uc?id=10zOGA1phaAtMPV2xAezyTISAggKNONgk&export=download)

vanfanel, 2021-08-19:
@popcornmix Tried! The issue at 1360x768 (forced in cmdline.txt via the "video=..." parameter) is not appearing anymore.
So, if you have no objections, this could be closed.

Thanks a lot for looking into this, for the time you invested into fixing the issue (and discovering it, to begin with... happening with a video mode only was unexpected).

popcornmix, 2021-08-19:
Fix should be in latest rpi-update firmware.

vanfanel, 2021-11-05:
@popcornmix Same HDMI poweroff on throttling is **happening again** in kernel  5.10.76-v8 with `video="HDMI-A-1:1280x720@60"`

popcornmix, 2021-11-05:
Can you confirm `vcgencmd version`.

vanfanel, 2021-11-05:
> Can you confirm `vcgencmd version`.

Oct 28 2021 14:14:14 
Copyright (c) 2012 Broadcom
version bcdfe55d56b75b4be52c10d73bdb4a60e2b49f75 (clean) (release) (start)

cagnulein, 2022-04-11:
any news @vanfanel @popcornmix ?

popcornmix, 2022-04-11:
The issue I could reproduce here has been resolved. @cagnulein what is your issue?

cagnulein, 2022-04-11:
@popcornmix i have the same issue with these kernel and firmware

[image]

the issue starts when the CPU is over 80C
I have a FULLHD 1920x1080 external monitor to the HDMI and the screen starts to turn on and off when this is happening

let me know if you need more information. I have the pi4 on a climatic chamber if you need to run several test.
Thanks

cagnulein, 2022-04-11:
@popcornmix as @vanfanel saw in the past, i have

when it's working

# vcgencmd measure_clock core
frequency(1)=500000992

when it's not working

# vcgencmd measure_clock core
frequency(1)=333333984

popcornmix, 2022-04-11:
We'll need a lot more information. Output of 
```
vcgencmd version
uname -a
```
Any non standard options in config.txt or cmdline.txt?
and what the hdmi mode you are in (single hdmi?).
Also what are you running (e.g. desktop, or something that uses drm to add planes?)
Does enabling performance governor (or using `force_turbo=1`) avoid the issue?

cagnulein, 2022-04-11:
@popcornmix 

> vcgencmd version
> uname -a

you can find them here https://github.com/raspberrypi/firmware/issues/1598#issuecomment-1095003217

[cmdline.txt](https://github.com/raspberrypi/firmware/files/8464616/cmdline.txt)
[config.txt](https://github.com/raspberrypi/firmware/files/8464617/config.txt)

I'm running Raspberry Pi OS with the default desktop (no application are running). I'm stressing it with stress-ng --cpu 2 and putting the cliamthic chamber to 45C.

Let's try the turbo_mode=1, I didn't try it yet. Thanks

cagnulein, 2022-04-11:
same problem with turbo_mode=1 @popcornmix

cagnulein, 2022-04-11:
i'm not sure. I'm repeating the test with force_turbo=1 now. I will let you know. Thanks and sorry for the confusion

cagnulein, 2022-04-11:
@popcornmix your firmware/kernel are newer: do you think it could be the culript?

popcornmix, 2022-04-11:
You could try rpi-update, but I can't think of anything that might have changed since your version (just over a month ago).

cagnulein, 2022-04-12:
@popcornmix good news! After updating the firmware the issue seems to gone away! I will do a long test this night, but it seems pretty stable right now with also the force_turbo=1. I just start the test for the night without the force_turbo=1 setting.

This is the log from rpi-update

```
root@raspberrypi:/home/pi# rpi-eeprom-update
*** UPDATE AVAILABLE ***
BOOTLOADER: update available
   CURRENT: Mon 22 Nov 2021 11:23:32 AM UTC (1637580212)
    LATEST: Tue 25 Jan 2022 02:30:41 PM UTC (1643121041)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1

```

It seems strange to me that the current version was the 22 Nov 2021, when the `vcgencmd version` gave me 
[image]

is there an explanation for this?

popcornmix, 2022-04-12:
The kernel is contained in `/boot/kernel*.img` and reports it's version from `uname -a`.
The firmware is contained in `/boot/start*.elf` and reports it's version from `vcgencmd version`.
The bootloader is contained in the SPI EEPROM and reports its version from `rpi-eeprom-update`.

> After updating the firmware the issue seems to gone away!

Can you tell me what command you used to update? Your firmware looks to be unchanged (Mar 1 2022).

cagnulein, 2022-04-12:
> Can you tell me what command you used to update? Your firmware looks to be unchanged (Mar 1 2022).

the screenshot above is just to show the previous version. I ran 

```
root@raspberrypi:/home/pi# rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
#############################################################
WARNING: This update bumps to rpi-5.15.y linux tree
See: https://forums.raspberrypi.com/viewtopic.php?t=322879

'rpi-update' should only be used if there is a specific
reason to do so - for example, a request by a Raspberry Pi
engineer or if you want to help the testing effort
and are comfortable with restoring if there are regressions.

DO NOT use 'rpi-update' as part of a regular update process.
##############################################################
Would you like to proceed? (y/N)
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   173  100   173    0     0   1802      0 --:--:-- --:--:-- --:--:--  1821
100  124M  100  124M    0     0   960k      0  0:02:12  0:02:12 --:--:--  649k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 5.15.33-v7l+
 *** depmod 5.15.33-v7+
 *** depmod 5.15.33+
 *** depmod 5.15.33-v8+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
ldconfig: /usr/lib/libddcutil.so.0 is not a symbolic link

 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to eaac3e5367944f2caf6eda88f81282633a3d7128
 *** A reboot is needed to activate the new firmware

```

popcornmix, 2022-04-12:
Okay, so rpi-update fixed it. Which suggests it was the firmware update that fixed it. (`vcgencmd version` should now return a newer date).

cagnulein, 2022-04-12:
@popcornmix confirmed

```
root@raspberrypi:/home/pi# vcgencmd version
Mar 24 2022 13:19:26
```

same as yours now

cagnulein, 2022-04-12:
@popcornmix just a clarification: i'm running debian 10.4 here

popcornmix, 2022-04-12:
Might have been worth mentioning at start. We have no control over firmware shipped with debian.
If the current stable debian firmware has this issue and the stable RPiOS firmware does not, then you
may want to report this to debian maintainers who can bump the firmware.

cagnulein, 2022-04-12:
I mean that's the output of /etc/debian_version 
The OS is RaspianOS but it's not up to date. Thanks for all the time and the clarifications! I hope the tonight's test will be fine!

popcornmix, 2022-04-12:
Okay, there is a [debian distribution](https://raspi.debian.net/) which we don't maintain.
But it sounds like you are using RPiOS buster (the legacy version).
For that using rpi-update is the best workaround, but you would be recommended to switch to RPiOS bullseye.

Buster will only get security fixes and critical bug fixes. Bullseye will get much more frequent updates and improvements.