raspberrypi/firmware #1599: No analog sound form Pi Zero W, even with external circuitry

Gordon999, 2021-07-21:
**Is this the right place for my bug report?**
Apologies if the wrong place but I am not sure where to report it, I have tried on the Pi Forum but no response

**Describe the bug**
 Following this method https://www.raspberrypi.org/blog/tinker ... ero-audio/ and building the required circuitry and adding
dtoverlay=pwm-2chan,pin=19,func=2,pin2=13,func2=4 to /boot/config.txt I can't get any analog sound. The choices for audio ouput do not offer an analog/ Jack  option. Used to work fine with older Raspbian OS.

**To reproduce**
Install latest Raspi OS, build circuitry, modify config .txt and try choosing analog audio as sound output.

**Expected behaviour**
Used to work with old OS so expected to work.

**Actual behaviour**
No audio choice for analog / jack sound.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)? Lastest 32 bit FULL
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

pelwell, 2021-07-21:
1. Your referenced URL is broken.
2. Although the built-in analog audio uses the PWM peripheral, PWM and audio are not the same thing. In fact, `dtoverlay=pwm-2chan` clashes with audio use, as the overlay documentation states:
```
pi@raspberrypi:~$ dtoverlay -h pwm-2chan
Name:   pwm-2chan

Info:   Configures both PWM channels
        Legal pin,function combinations for each channel:
          PWM0: 12,4(Alt0) 18,2(Alt5) 40,4(Alt0)            52,5(Alt1)
          PWM1: 13,4(Alt0) 19,2(Alt5) 41,4(Alt0) 45,4(Alt0) 53,5(Alt1)
        N.B.:
          1) Pin 18 is the only one available on all platforms, and
             it is the one used by the I2S audio interface.
             Pins 12 and 13 might be better choices on an A+, B+ or Pi2.
          2) The onboard analogue audio output uses both PWM channels.
          3) So be careful mixing audio and PWM.
          4) Currently the clock must have been enabled and configured
             by other means.

Usage:  dtoverlay=pwm-2chan,<param>=<val>

Params: pin                     Output pin (default 18) - see table
        pin2                    Output pin for other channel (default 19)
        func                    Pin function (default 2 = Alt5) - see above
        func2                   Function for pin2 (default 2 = Alt5)
        clock                   PWM clock frequency (informational)

```
3. Use `dtoverlay=audremap` instead:
```
pi@raspberrypi:~$ dtoverlay -h audremap
Name:   audremap

Info:   Switches PWM sound output to GPIOs on the 40-pin header

Usage:  dtoverlay=audremap,<param>=<val>

Params: swap_lr                 Reverse the channel allocation, which will also
                                swap the audio jack outputs (default off)
        enable_jack             Don't switch off the audio jack output
                                (default off)
        pins_12_13              Select GPIOs 12 & 13 (default)
        pins_18_19              Select GPIOs 18 & 19
```

Gordon999, 2021-07-21:
Thanks. The link https://www.raspberrypi.org/blog/tinkernut-diy-pi-zero-audio/ (its a Pi blog)

Is this 'clash' only revelant to the latest OS as it works fine on old raspbian OS ?

pelwell, 2021-07-21:
Hmmm - the tutorial says 13 & 18, an interesting combination but one that should work. However, the `audremap` overlay doesn't support that combination.

You have selected 13 & 19 which is wrong - it maps PWM1 onto both pins and leaves PWM0 unconnected.

PWM usage will clash with audio usage if the Linux pwm driver is used, but you might get away with it if you don't.

I recommend using `raspi-gpio get` to see the current state of the pins and confirm that the desired functions are enabled in the right places.

Gordon999, 2021-07-21:
I don't believe 13 & 19 are wrong, worked fine before as far as I know. 19 is an alternative to 18.

Thanks for the help, I got it working with

dtparam=audio=on
audio_pwm_mode=2
dtoverlay=audremap,pins_13,19

still can't choose analog as an option but plays OK.

maybe they can update the blog etc

popcornmix, 2021-07-21:
> audio_pwm_mode=2

This is a default option, so can be removed without change.

Gordon999, 2021-07-21:
Thanks for the help and advice, working OK now.

Gordon999, 2021-07-21:
Solved

closed 2021-07-21 by Gordon999