raspberrypi/firmware #1601: Failure to boot after kernel and/or bootloader update

Arson31, 2021-07-30:
**Is this the right place for my bug report?**
This repository contains the GPU firmware used on the Raspberry Pi. This software is the closed source part of the Raspberry Pi system, it includes booting (including network booting and USB booting), low-level power and clock control, FKMS and legacy HDMI control (not full KMS that is in the linux kernel), hardware legacy codecs (MPEG2, H264 and VC1), encode hardware including the ISP (image sensor pipeline) and camera control, audio output (analogue and HDMI audio).

If you believe that the issue you are seeing is within this area, this is the right place. If not, we have other repositories for the linux kernel at [github.com/raspberrypi/linux](https://github.com/raspberrypi/linux) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**

After installing the packages below the system runs into a major slow down and other service errors after a reboot (nginx failures)
Get:1 http://archive.raspberrypi.org/debian buster/main armhf libraspberrypi-doc armhf 1:1.20210727-1 [31.4 MB]
Get:2 http://archive.raspberrypi.org/debian buster/main armhf libraspberrypi-dev armhf 1:1.20210727-1 [400 kB]
Get:3 http://archive.raspberrypi.org/debian buster/main armhf raspberrypi-kernel armhf 1:1.20210727-1 [79.6 MB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf libraspberrypi-bin armhf 1:1.20210727-1 [341 kB]
Get:5 http://archive.raspberrypi.org/debian buster/main armhf libraspberrypi0 armhf 1:1.20210727-1 [847 kB]
Get:6 http://archive.raspberrypi.org/debian buster/main armhf raspberrypi-bootloader armhf 1:1.20210727-1 [4518 kB]
Get:7 http://archive.raspberrypi.org/debian buster/main armhf linux-libc-dev armhf 1:1.20210727-1 [1011 kB]

The second reboot got hung and never rebooted so I had to power it off. I turn the device back on again but the device would only show the boot screen for 1 second, loses signal and doesn't boot.

SSH does not work as I keep pinging the static IP address I assigned to it and doesn't find the host for ever 3 minutes.

**To reproduce**
Install the above packages using apt-get update and attempt reboot 2 times.

**Expected behaviour**
Raspberry Pi should boot normally with no consistent slowdowns

**Actual behaviour**
The Raspberry Pi has high CPU usage according to htop and sometimes hangs after rebooting. 

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

Raspberry Pi 4 Model B Rev 1.4 8GB
Raspbian GNU/Linux 10 (buster) armv7l
Kernel before install: 5.10.17-v7l+

**Logs**
Cannot get to them since it will not boot.

**Additional context**
Open Media Vault 5.6.13-1 (Usul) is installed but I did not post the problem on the forums since these packages come from the Raspberry Pi repo list. The Raspberry Pi is powered by a Canakit USB C power kit and attached is a small case fan from Canakit, a external HDD attached using USB 3, and an ethernet cable connected directly to my router. I do have a similar setup with Open Media Vault 5.6.13-1 (Usul) installed and after installing the above packages I did not have the same issue. I restored my OS system on a different micro SD card from a previous backup and the problem doesn't occur. But when I check for updates on the OMV control panel the same updates show up (the same packages above). I installed the same updates again and the problem repeats. I restored my system with the same backup image but I cannot update with the packages listed above.

JamesH65, 2021-07-30:
What happens if you remove all peripherals, ie the USB drive?

popcornmix, 2021-07-30:
Can you post your config.txt and cmdline.txt

Arson31, 2021-07-31:
> What happens if you remove all peripherals, ie the USB drive?

Same issue, I did not try this without removing the ethernet cable or the mini HDMI.

> Can you post your config.txt and cmdline.txt
I've attached is the cmdline.txt and config.txt but this is a restore from a previous image without the updates (don't know if that would make a difference). I forgot to mention that I did change the memory split to 256 and did some overclocking: over_voltage=2, arm_freq=1750, and gpu_freq=750.

[cmdline.txt](https://github.com/raspberrypi/firmware/files/6911934/cmdline.txt)
[config.txt](https://github.com/raspberrypi/firmware/files/6911935/config.txt)

popcornmix, 2021-08-01:
Also remove overclock settings if you have any instability.
If that doesn't help, does removing the gpu_mem setting help?

Anti-ctrl, 2021-08-01:
IDK if its relevant, but i have a similar Problem as OP. Yesterday i updated my Pi3B+, rebooted and everything was fine. Today at ~11 o'clock GMT+2, the Pi hung up. I thought "No big deal, just cut Power and reboot, as always".

Surprise surprise, the Pi didnt came back up, even tho it seemed like it, so i investigated further and chucked an HDMI cable in the Port. The Pi booted fine and got a connection. Well.. I thought everything is fine, so back in its place and..No connection. USB SSD blinked as it was booting, but without HDMI Connection i dont get anything out of it, at all, which is really weird.

Arson31, 2021-08-02:
I removed all the overclocking elements and peripherals; leaving just the ethernet cable plugged straight to my router before updating. I rebooted a few times and see no issues (I did notice shutdowns took a bit longer than usual). I decided to add the overclocking elements back in. Everything booted just fine and I checked my RPi-Monitor Docker container web page and displays an error message: Can not get information (dynamic.json) from RPi-Monitor server. 

I proceeded to shutdown but the device never shutdown. I plugged in the mini HDMI to a separate monitor and there was no signal. I powered off the device and turn it back on again. Same thing as before the screen flashed and the device doesn't boot. I removed the overclocking elements out of the config.txt and powered the machine on again. The machine booted normally and RPi-Monitor shows all system information correctly. 

I proceeded to shutdown the machine again plugged my peripherals again (HDD) and powered the machine again. The device booted up normally as it should and around 10 minutes in with all my docker containers starting up I noticed a ton of load on the CPU and my nginx service failing for the OMV login admin. 

There is noticeable lag on web interfaces and some few seconds of time connecting over SSH. I see constant CPU spikes too. I'm not sure if it's because of the notable difference with the overclocking settings. Node exporter has been showing a constant >95% CPU usage.

I guess it's clear that the overclocking elements were the cause of the booting problems but now I'm getting performance issues.

I'm having overall weird incidents which makes me want to switch to my other SD card that does not contain these updates and having no significant impact on performance.

popcornmix, 2021-08-02:
Posting your config.txt and cmdline.txt would be useful.

Arson31, 2021-08-02:
> Posting your config.txt and cmdline.txt would be useful.

[config.txt](https://github.com/raspberrypi/firmware/files/6918463/config.txt)
[cmdline.txt](https://github.com/raspberrypi/firmware/files/6918465/cmdline.txt)

popcornmix, 2021-08-02:
Can you try removing the `gpu_mem` setting?
If that doesn't help then replace:
```
dtoverlay=vc4-fkms-v3d
```
with
```
dtoverlay=vc4-fkms-v3d,cma-256
```
and report if that helps.

Arson31, 2021-08-03:
Removing the gpu_mem setting made things worse. A single SSH connection took me over 3 minutes to connect.

I replaced `dtoverlay=vc4-fkms-v3d` with `dtoverlay=vc4-fkms-v3d,cma-256` and everything seems to work as normal.

I'm attaching the new files for review:

[cmdline.txt](https://github.com/raspberrypi/firmware/files/6920236/cmdline.txt)
[config.txt](https://github.com/raspberrypi/firmware/files/6920237/config.txt)

As for the overclocking settings, I assume it would be fixed in the next release or would I have to replace them with different values?

I haven't noticed any significant performance difference with overclocking turned on so would overclocking make much of a difference?

popcornmix, 2021-08-03:
Overclock settings and not guaranteed. Since the last update there has been some juggling with PLLs used for the different sources and they might just not so support so high an overclock on your specific silicon.

To be honest, gpu_freq=750 over_voltage=2 seems very ambitious to me. That's a 50% overclock with a small over-voltage. I'd have thought reducing gpu_freq or increasing over_voltage would likely be needed if you have instability.

Your use case having issues with the higher cma setting (with gpu_mem settings removed) is unusual. Can you describe what was running when ssh was slow? Were you serving files to other devices? Were torrents (or other files) downloading? Anything else that might significantly use network, disk, cpu or ram?

Arson31, 2021-08-03:
> Can you describe what was running when ssh was slow? Were you serving files to other devices? Were torrents (or other files) 
downloading? Anything else that might significantly use network, disk, cpu or ram?

I have fail2ban, SMB/CIFS, SSH, clam antivirus (runs on a schedule for specific folders but only on one day at times when the server is not as active), portainer, nginx (probably for the OMV control panel), and docker with some stacks and individual containers running. I stopped a stack which contains multiple containers (including a torrent client) for troubleshooting purposes and I still had performance issues. The only stacks is see could impact performance would be Plex, Watchtower, and some containers for system monitoring purposes (node-exporter, grafana, and prometheus).

popcornmix, 2021-08-03:
The latest rpi-update reduces the default cma to 320M by default and reduces it further (to 256M) when gpu_mem is high (>256) or running on a 1GB  pi4.
Can you check whether that helps issue with `cma-256` removed?

If you happen to observe which software is unhappy when cma=512, I'd be very interested.

Arson31, 2021-08-04:
I changed these values in the config.txt:

`[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
#dtoverlay=vc4-fkms-v3d,cma-256
dtoverlay=vc4-fkms-v3d,cma-512
max_framebuffers=2`

I did not touch the overclock settings or the gpu_mem.

I'm assuming that's what you mean, but my htop thread spiked over 70% sometimes 80%, some docker containers but I think that's because they're starting up. OpenMediaVault-webgui sometimes show up but I just notice htop is always the worst. Using the arrow keys when in htop takes several seconds to select threads. Plex container has been using > 100% CPU for about here and there. It's kind of hard to tell since htop is having a hard time refreshing because of the CPU usage.

popcornmix, 2021-08-04:
Is it easy to disable some of the docker containers for testing?
If you could identify the specific one that results in the slowdown it would narrow it down.

Note: it may be one specific docker container causes the issue. It may be any docker container causes the issue. Or it may be too many docker container causes the issue (assuming it's some form of out-of-memory).

Victrid, 2021-08-04:
I’ve experienced the similar situation on my raspberry pi 3b+ as [this comment](https://github.com/raspberrypi/firmware/issues/1601#issuecomment-890505737) describes, after flashing my tf card with a latest arch Linux arm rpi.

Nothing has been installed and configured and the system is as blank as out of the box. The only things I plugged in were power line, Ethernet cable, and HDMI, with the last one can I boot the pi after a reboot.

Both raspbian and arch uses the same infrastructure to boot and they have not much in common when entered Linux environment, so it could probably be in the stage when initiating the environment.

Hope this information will help you to locate the problem.

Arson31, 2021-08-04:
I could turn stop Docker completely. It's hard to navigate on Portainer, even on SSH due to the high CPU usage my connection gets timed out. It's just strange everything is using more CPU usage than it needs but switching from `cma-512` to `cma-256` uses maybe 16% - 30% total CPU on average.

popcornmix, 2021-08-04:
> It's hard to navigate on Portainer, even on SSH due to the high CPU usage my connection gets timed out

I'd suggest doing the configuring with `cma-256` so system is responsive, and only then switch to `cma-512` and reboot.

Linux can get ultra slow when it is low on memory, either through swap, or by continuously evicting read-only pages (that then need to be reloaded when code runs). I suspect it's something like this, although unused cma should be free for use by most purposes (any movable pages).

I'm guessing there is some kernel  allocation that can't use cma and causes things to grind when memory is low. Finding a simple example that provokes it would help with investigation.

For now, we've reduced the default cma to 320M, and have made the firmware lower this further to 256M when there is a large (>256M) gpu_mem setting or when running on a 1GB Pi. That is in rpi-update and should appear in apt soon, and should avoid some of these issues. But we'd like to be in a position to configure more cma as it can be beneficial.

Arson31, 2021-08-07:
I've upgraded the same packages with the new 20210805-1 release. I shutdown the device, removed all peripherals, and changed to `cma-512` again and booted the device. I see no HDMI signal but the device does boot and all the services start with all the docker containers running looking like there is no significant performance impact like there were before. 

I stopped all docker containers except for RPi-monitor and I'm seeing 4% max of CPU usage from htop and maybe max of 2% from RPi-monitor so 6% spike overall at the best but on average it's about 3% overall because of htop. I'm seeing a 2.6% usage of memory out of 7.71GB.

I shutdown down the device and revert all settings I had before (overclocking and gpu memory) and the device will not boot. I shutdown down the device commented out the overclocking settings, `gpu memory=256`, and `dtoverlay=vc4-fkms-v3d` then added `dtoverlay=vc4-fkms-v3d,cma-512`.

I plugged in my USB3 HDD and the mini HDMI and turned on the device. The screen flashed the boot screen and HDMI outputted text as normal. I checked htop and RPi-monitor and CPU and memory usage is very low. I started all the docker stacks and containers one at a time starting the ones that use the lowest resources. Runc init has been spiking to 19% max CPU at times and htop to around 7%. I started Plex and Plex has been using minimal CPU usage. 

I started my seedbox stack containing multiple containers I noticed slowdowns and high CPU usage. Plex has been shown using over 50 - 70% CPU usage on average and spikes over 90%. Htop is shown using a quite a bit as well. I stopped the seedbox stack (which took a while to stop and seeing decent overall CPU usage) and started it again. Again the CPU usage is high os I'm assuming it's my Glueton container thats' causing the problem. I cannot show my stack's contents but it's similar to the stack below but with flaresolver:

https://gist.githubusercontent.com/dnburgess/9b95c72292ec8b2badb92a667702c70f/raw/f11822e5b853e42d7b475cbd954f391f5a77a1b7/gistfile1.txt

I changed back to `cma-256` again started the seedbox stack again and all is well.

So it must be something to do with the stack with `cmd-512`. Overclocking still doesn't boot but with the settings I have now will do. I do not see any benefit of having the original configuration compared to the settings I have now.

closed 2021-08-10 by Arson31