raspberrypi/firmware #1602: Modifying the default .dtb file

seamusdemora, 2021-07-30:
The listing at [`firmware/extra/dt-blob.dts`](https://github.com/raspberrypi/firmware/blob/master/extra/dt-blob.dts) presumably contains the device tree source for the RPi 4B (and all other RPi models). 

I am just trying to learn more about how the device tree works, and [AIUI](https://www.raspberrypi.org/documentation/configuration/device-tree.md), this `.dts` file is compiled to create a `.dtb` file. The `.dtb` file is loaded by `start.elf` during the boot process. `start.elf` may make some runtime modifications to the `.dtb` file, and it merges user-supplied parameters and overlays (presumably from `/boot/config.txt`) to create a "merged" `.dtb` file. This merged "blob" is then passed to the kernel via pointer. 

To aid the learning process, I would like to try compiling my own `.dtb` file, make small changes, and observe the result. Of course verification requires knowledge of the correct outcome, and this brings me to my question:

The [`dt-blob.dts` file](https://github.com/raspberrypi/firmware/blob/master/extra/dt-blob.dts) contains the following line for the RPi 4B: 

```
pin@p130 { function = "output"; termination = "no_pulling"; polarity = "active_low"; startup_state = "active"; }; // PWR_LED (RED)
```  
If I change `polarity = "active_low";` to `polarity = "active_high";` **OR** add the parameter `drive_strength_mA = < 2>;`, would that have the effect of extinguishing the RED PWR LED --or-- reducing its intensity? If not, can you explain?

pelwell, 2021-07-31:
https://www.google.com/search?q=device+tree+Raspberry+Pi

pelwell, 2021-07-31:
Hint: dt-blob is not Device Tree, it's just abusing the DT tools to store basic pin configuration data

seamusdemora, 2021-07-31:
@pelwell - Thanks! Semantics often seems to be half the problem :)

closed 2022-11-28 by timg236