raspberrypi/firmware #1604: Raspberry PI doesn't support square resolutions iva HDMI.

briandilley, 2021-08-02:
**Describe the bug**
1:1 aspect rations aren't supported via HDMI on the raspberry pi.

**To reproduce**
Use a screen with a 1:1 aspect ration

**Expected behaviour**
I expect the HDMI display to work with a 1:1 aspect ration (ie: 1080x1080)

**Actual behaviour**
It doesn't work

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
All

* Which OS and version (`cat /etc/rpi-issue`)?
All

* Which firmware version (`vcgencmd version`)?
All

* Which kernel version (`uname -a`)?
All

**Additional context**
I've got a 5" round LCD screen connected to an HDMI board. The round LCD screen supports square resolutions (ie: 1:1 aspect ratios like 1080x1080 or 720x720).

JamesH65, 2021-08-02:
The System entries above are unhelpful. What devices do you know it does not work on? Which graphics drivers are you using (legacy, FKMS, KMS?), which firmware version are you using, what kernel are you using?

Can you dump the EDID from the device?

briandilley, 2021-09-04:
haha - sorry about the sparse ticket details - will update now.

briandilley, 2021-09-04:
So I got it working after all :)  I used these options in config.txt:

```
hdmi_cvt=1080 1080 60 1 0 0 0
hdmi_group=2
hdmi_mode=87
hdmi_drive=2
```

Of course that's configuring it as 4:3 aspect ratio even though it's 1:1 - but it seems to work (I haven't put anything on the display to verify yet though.

closed 2022-11-29 by timg236