raspberrypi/firmware #1605: Network booting no longer works

reraikes, 2021-08-02:
Using RPF's "Network boot your Raspberry Pi" tutorial (https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/net_tutorial.md), a Raspberry Pi 3B+ client network boots perfectly from a Raspberry Pi 4B server using firmware versions up through and including 5.10.27 (acf3b95). Monitoring /var/log/daemon.log displays:

Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 available DHCP subnet: 192.168.1.255/255.255.255.0
Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 vendor class: PXEClient:Arch:00000:UNDI:002001
Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 PXE(eth0) b8:27:eb:6c:4c:16 proxy
Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 tags: eth0
Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 broadcast response
Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 sent size: 1 option: 53 message-type 2
Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 sent size: 4 option: 54 server-identifier 192.168.1.117
Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 sent size: 9 option: 60 vendor-class 50:58:45:43:6c:69:65:6e:74
Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 sent size: 17 option: 97 client-machine-id 00:16:4c:6c:b6:16:4c:6c:b6:16:4c:6c:b6:16...
Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 sent size: 32 option: 43 vendor-encap 06:01:03:0a:04:00:50:58:45:09:14:00:00:11...
Aug 2 12:21:02 raspberrypi dnsmasq-tftp[548]: file /pxe-boot/bootsig.bin not found
Aug 2 12:21:02 raspberrypi dnsmasq-tftp[548]: sent /pxe-boot/bootcode.bin to 192.168.1.116
Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 available DHCP subnet: 192.168.1.255/255.255.255.0
Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 vendor class: PXEClient:Arch:00000:UNDI:002001
Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 PXE(eth0) b8:27:eb:6c:4c:16 proxy
Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 tags: eth0
Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 broadcast response
Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 sent size: 1 option: 53 message-type 2
Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 sent size: 4 option: 54 server-identifier 192.168.1.117
Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 sent size: 9 option: 60 vendor-class 50:58:45:43:6c:69:65:6e:74
Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 sent size: 17 option: 97 client-machine-id 00:16:4c:6c:b6:16:4c:6c:b6:16:4c:6c:b6:16...
Aug 2 12:21:02 raspberrypi dnsmasq-dhcp[548]: 653460281 sent size: 32 option: 43 vendor-encap 06:01:03:0a:04:00:50:58:45:09:14:00:00:11...
Aug 2 12:21:02 raspberrypi dnsmasq-tftp[548]: error 0 Early terminate received from 192.168.1.116
Aug 2 12:21:02 raspberrypi dnsmasq-tftp[548]: failed sending /pxe-boot/b66c4c16/start.elf to 192.168.1.116
Aug 2 12:21:02 raspberrypi dnsmasq-tftp[548]: file /pxe-boot/b66c4c16/autoboot.txt not found
Aug 2 12:21:02 raspberrypi dnsmasq-tftp[548]: error 0 Early terminate received from 192.168.1.116
Aug 2 12:21:02 raspberrypi dnsmasq-tftp[548]: failed sending /pxe-boot/b66c4c16/start.elf to 192.168.1.116
Aug 2 12:21:02 raspberrypi dnsmasq-tftp[548]: sent /pxe-boot/b66c4c16/config.txt to 192.168.1.116
Aug 2 12:21:02 raspberrypi dnsmasq-tftp[548]: file /pxe-boot/b66c4c16/recovery.elf not found
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: sent /pxe-boot/b66c4c16/start.elf to 192.168.1.116
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: sent /pxe-boot/b66c4c16/fixup.dat to 192.168.1.116
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: file /pxe-boot/b66c4c16/recovery.elf not found
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: error 0 Early terminate received from 192.168.1.116
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: failed sending /pxe-boot/b66c4c16/config.txt to 192.168.1.116
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: sent /pxe-boot/b66c4c16/config.txt to 192.168.1.116
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: file /pxe-boot/b66c4c16/dt-blob.bin not found
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: file /pxe-boot/b66c4c16/recovery.elf not found
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: error 0 Early terminate received from 192.168.1.116
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: failed sending /pxe-boot/b66c4c16/config.txt to 192.168.1.116
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: sent /pxe-boot/b66c4c16/config.txt to 192.168.1.116
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: file /pxe-boot/b66c4c16/bootcfg.txt not found
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: error 0 Early terminate received from 192.168.1.116
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: failed sending /pxe-boot/b66c4c16/bcm2710-rpi-3-b-plus.dtb to 192.168.1.116
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: sent /pxe-boot/b66c4c16/bcm2710-rpi-3-b-plus.dtb to 192.168.1.116
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: error 0 Early terminate received from 192.168.1.116
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: failed sending /pxe-boot/b66c4c16/overlays/overlay_map.dtb to 192.168.1.116
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: sent /pxe-boot/b66c4c16/overlays/overlay_map.dtb to 192.168.1.116
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: error 0 Early terminate received from 192.168.1.116
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: failed sending /pxe-boot/b66c4c16/config.txt to 192.168.1.116
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: sent /pxe-boot/b66c4c16/config.txt to 192.168.1.116
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: error 0 Early terminate received from 192.168.1.116
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: failed sending /pxe-boot/b66c4c16/cmdline.txt to 192.168.1.116
Aug 2 12:21:04 raspberrypi dnsmasq-tftp[548]: sent /pxe-boot/b66c4c16/cmdline.txt to 192.168.1.116
Aug 2 12:21:06 raspberrypi dnsmasq-tftp[548]: file /pxe-boot/b66c4c16/recovery8.img not found
Aug 2 12:21:06 raspberrypi dnsmasq-tftp[548]: file /pxe-boot/b66c4c16/recovery8-32.img not found
Aug 2 12:21:06 raspberrypi dnsmasq-tftp[548]: file /pxe-boot/b66c4c16/recovery7.img not found
Aug 2 12:21:06 raspberrypi dnsmasq-tftp[548]: file /pxe-boot/b66c4c16/recovery.img not found
Aug 2 12:21:06 raspberrypi dnsmasq-tftp[548]: error 0 Early terminate received from 192.168.1.116
Aug 2 12:21:06 raspberrypi dnsmasq-tftp[548]: failed sending /pxe-boot/b66c4c16/kernel8.img to 192.168.1.116
Aug 2 12:21:06 raspberrypi dnsmasq-tftp[548]: file /pxe-boot/b66c4c16/kernel8-32.img not found
Aug 2 12:21:06 raspberrypi dnsmasq-tftp[548]: error 0 Early terminate received from 192.168.1.116
Aug 2 12:21:06 raspberrypi dnsmasq-tftp[548]: failed sending /pxe-boot/b66c4c16/kernel8.img to 192.168.1.116
Aug 2 12:21:06 raspberrypi dnsmasq-tftp[548]: error 0 Early terminate received from 192.168.1.116
Aug 2 12:21:06 raspberrypi dnsmasq-tftp[548]: failed sending /pxe-boot/b66c4c16/kernel7.img to 192.168.1.116
Aug 2 12:21:06 raspberrypi dnsmasq-tftp[548]: file /pxe-boot/b66c4c16/armstub8-32.bin not found
Aug 2 12:21:06 raspberrypi dnsmasq-tftp[548]: error 0 Early terminate received from 192.168.1.116
Aug 2 12:21:06 raspberrypi dnsmasq-tftp[548]: failed sending /pxe-boot/b66c4c16/kernel7.img to 192.168.1.116
Aug 2 12:21:06 raspberrypi dnsmasq-tftp[548]: error 0 Early terminate received from 192.168.1.116
[... 92 lines cut ...]
Aug 2 12:33:02 raspberrypi dnsmasq-tftp[548]: failed sending /pxe-boot/b66c4c16/kernel8.img to 192.168.1.116
Aug 2 12:33:02 raspberrypi dnsmasq-tftp[548]: error 0 Early terminate received from 192.168.1.116
Aug 2 12:33:02 raspberrypi dnsmasq-tftp[548]: failed sending /pxe-boot/b66c4c16/kernel7.img to 192.168.1.116
Aug 2 12:33:02 raspberrypi dnsmasq-tftp[548]: file /pxe-boot/b66c4c16/armstub8-32.bin not found
Aug 2 12:33:02 raspberrypi dnsmasq-tftp[548]: error 0 Early terminate received from 192.168.1.116
Aug 2 12:33:02 raspberrypi dnsmasq-tftp[548]: failed sending /pxe-boot/b66c4c16/kernel7.img to 192.168.1.116
Aug 2 12:33:02 raspberrypi dnsmasq-tftp[548]: error 0 Early terminate received from 192.168.1.116
Aug 2 12:33:02 raspberrypi dnsmasq-tftp[548]: failed sending /pxe-boot/b66c4c16/kernel7.img to 192.168.1.116
Aug 2 12:33:05 raspberrypi dnsmasq-tftp[548]: sent /pxe-boot/b66c4c16/kernel7.img to 192.168.1.116
Aug 2 12:33:14 raspberrypi dnsmasq-dhcp[548]: 2344883336 available DHCP subnet: 192.168.1.255/255.255.255.0
Aug 2 12:33:14 raspberrypi dnsmasq-dhcp[548]: 2344883336 available DHCP subnet: 192.168.1.255/255.255.255.0

Expected logs are NOT produced in /var/log:

root@raspberrypi:/pxe-root/b66c4c16/var/log# ls -l
total 40
-rw-r--r-- 1 root root 0 May 7 10:00 alternatives.log
drwxr-xr-x 2 root root 4096 Aug 2 00:42 apt
-rw-r--r-- 1 root root 0 May 7 10:00 bootstrap.log
-rw-rw---- 1 root utmp 0 May 7 10:00 btmp
-rw-r--r-- 1 root root 26956 Aug 2 00:42 dpkg.log
-rw-r--r-- 1 root root 0 May 7 10:00 faillog
-rw-rw-r-- 1 root utmp 0 May 7 10:00 lastlog
drwx------ 2 root root 4096 Aug 2 12:33 private
-rw-rw-r-- 1 root utmp 384 Aug 2 12:33 wtmp

timg236, 2021-08-03:
These log entries suggest that the Kernel and DTB were loaded correctly. The firmware would then start Linux. 

"Aug 2 12:33:01 raspberrypi dnsmasq-tftp[548]: sent /pxe-boot/b66c4c16/bcm2710-rpi-3-b-plus.dtb to 192.168.1.116"
"Aug 2 12:33:05 raspberrypi dnsmasq-tftp[548]: sent /pxe-boot/b66c4c16/kernel7.img to 192.168.1.116"

What's the failure mode e.g. kernel panic? 

If you have a USB serial cable then adding "uart_2ndstage=1" to config.txt will show the TFTP logs from VideoCore

reraikes, 2021-08-03:
The Raspberry Pi 4B does not exhibit the problem.
The problem is limited to the Raspberry Pi 3B+.
The problem is related to the Ethernet port.

With 5.10.27 or below, the Ethernet port is enabled and the kernel is successfully downloaded and started.
The Ethernet port is then disabled briefly, re-enabled, and operation is normal and as expected.
The green LED blinks rapidly at one-second intervals.

With anything above 5.10.27, the Ethernet port is enabled and the kernel is successfully downloaded and started.
The Ethernet port is then disabled and is never re-enabled.
The green LED blinks rapidly at one-second intervals.

nicponj, 2021-08-04:
> The Raspberry Pi 4B does not exhibit the problem.
> The problem is limited to the Raspberry Pi 3B+.
> The problem is related to the Ethernet port.
>
> With 5.10.27 or below, the Ethernet port is enabled [...]

I can confirm what @reraikes is stating, as I am experiencing the same issue and symptoms as on my RPi 3B+. My 4Bs are not exhibiting these problems. The problem is not related to the current current kernel, as it loads fine via network boot. The issue presents after installing raspberrypi-bootloader (Version: 1:1.20210727-1) and it's related dependencies. If I keep the older bootloader (Version: 1:1.20210527-1) and its dependencies installed the issue does not present itself.

jnicpon, 2021-08-04:
> The Raspberry Pi 4B does not exhibit the problem.
> The problem is limited to the Raspberry Pi 3B+.
> The problem is related to the Ethernet port.
>
> With 5.10.27 or below, the Ethernet port is enabled [...]

I can confirm what @reraikes is stating. The issue only occurs on my 3B+ units and not on 4B units. In performing further tests, I was able to install and netboot the latest kernel on the 3B+ (5.10.52); however, the issue presents itself when installing the most recent update to raspberrypi-bootloader and its dependencies (1:1.20210527-1). If I remain at the old version of raspberrypi-bootloader (1:1.20210527-1), I do not experience these issues. I am using this as a current workaround.

In addition, the 3B+ is able to use the latest bootcode.bin file from the 1:1.20210527-1 update without issue, but the problem seems to be related to the NIC drivers and dependencies loaded in the prior to the linux kernel. I hope this helps.

-John

reraikes, 2021-08-04:
I just tried network booting a Raspberry Pi 2B (v1.1) using an SD card containing only bootcode.bin (the one from 5.10.52 or previous ones all work).

The results are the same as with a Raspberry Pi 3B+.

5.10.52 fails after successfully downloading the kernel and starting.

After simply downgrading the firmware to 5.10.27 (rpi-update acf3b95a82bdecd28d95c7b294cf5d5a9c3ced31), all is well.

peterharperuk, 2021-08-05:
I can reproduce this as reported acf3b95a is ok. The kernel seems to get stuck on boot when using 922deae6d

timg236, 2021-08-06:
Please can you post the cmdline/config.txt files and UART logs

reraikes, 2021-08-06:
[cmdline.txt](https://github.com/raspberrypi/firmware/files/6944053/cmdline.txt)
[config.txt](https://github.com/raspberrypi/firmware/files/6944054/config.txt)

No UART logs available.

reraikes, 2021-08-06:
UART logs:

[UART-5.10.27.txt](https://github.com/raspberrypi/firmware/files/6947963/UART-5.10.27.txt)
[UART-5.10.52.txt](https://github.com/raspberrypi/firmware/files/6947964/UART-5.10.52.txt)

timg236, 2021-08-09:
Thanks. It looks as though NFS boot was working enough to load systemd but something is happening late in boot which fails

- Would you be able to try an SD card with Raspberry Pi OS + rpi-udpate on this board?
- Does setting arm_freq=600 in config.txt help?

reraikes, 2021-08-09:
The Raspberry Pi 3B+ boots/runs fine from an SD card (Raspberry Pi OS Lite + apt update + apt full-upgrade + rpi-update).

Setting 'arm_freq=600' in config.txt changes nothing with NFS boot.

ivoruetsche, 2021-08-09:
I have the same problem and I revert my NFS server to the last state before the "apt dist-upgrade", but without success, still not booting with the reverted state...

reraikes, 2021-08-09:
> I have the same problem and I revert my NFS server to the last state before the "apt dist-upgrade", but without success, still not booting with the reverted state...

You need to revert the firmware of the OS that is served up to the client, NOT the firmware of the OS running the server itself:

rpi-update acf3b95a82bdecd28d95c7b294cf5d5a9c3ced31

It's the client's downloaded OS that has the problem, NOT the server's OS (running the latest on the server is not a problem).

Depending on how you have things implemented, this may require doing a chroot to the client OS before running rpi-update (or simply replacing the client's OS on the server with a firmware version no higher than 5.10.27).

ivoruetsche, 2021-08-09:
> > I have the same problem and I revert my NFS server to the last state before the "apt dist-upgrade", but without success, still not booting with the reverted state...
> 
> You need to revert the firmware of the OS that is served up to the client, NOT the firmware of the OS running the server itself:

Yes, sure, that's clear - I reverted my /nfs (for / on Raspi) and my /var/lib/tftpboot (for /boot on Raspi) directory on the (ubuntu) server, where all the bootcode etc. resists from the Raspi's.

ivoruetsche, 2021-08-09:
hmmm, I tried the downgrade "rpi-update acf3b95a82bdecd28d95c7b294cf5d5a9c3ced31", but it did an upgrade from 5.10.17 to 5.10.27 and it still not working.
That's strange, because everything is restored to an old state and the firmware was also an older version...

reraikes, 2021-08-09:
> I tried the downgrade "rpi-update acf3b95a82bdecd28d95c7b294cf5d5a9c3ced31", but it did an upgrade from 5.10.17 to 5.10.27 and it still not working.

"rpi-update acf3b95a82bdecd28d95c7b294cf5d5a9c3ced31" will install firmware 5.10.27 regardless of the version you are currently on.

Anything up to and including 5.10.27 runs perfectly as the client OS for me and the other posters,  Anything higher fails.

It's the OS firmware files that are downloaded to the client over Ethernet that matter (the server OS can remain at 5.10.52).

ivoruetsche, 2021-08-09:
...I have no idea, but it looks like, that my tftp service do caching... After I reboot the TFTP/NFS server it took the old files and my Raspi will start now

ivoruetsche, 2021-08-10:
I know what's happen...
- /var/lib/tftpboot with the new kernel
- I renamed it to /var/lib/tftpboot.old
- Restore the old state to /var/lib/tftpboot
I don't restart tht tftp daemon, but it followed the renamed directory and took all the time the files from /var/lib/tftpboot.old instead of /var/lib/tftpboot

peterharperuk, 2021-08-10:
I think we understand the problem now. Something in the firmware is trying to access a file after the file system has been closed down.

If you want to use the latest firmware for testing there's a workaround until the problem is fixed. Put "start_x=1" in config.txt and start_x.elf will be loaded instead - it doesn't suffer from this problem. I just tried this and it seemed to allow the device to boot properly. Can anyone else double check if this works for them?

reraikes, 2021-08-10:
Putting "start_x=1" in config.txt allows 5.10.52 to boot and run here.

peterharperuk, 2021-08-10:
Thanks for confirming that @reraikes

reraikes, 2021-08-10:
@peterharperuk,

Putting "start_x=1" in config.txt does NOT allow 5.10.52 to boot and run when network booting 64-bit Raspberry Pi OS on a client.  Any version up to and including 5.10.27 boots and runs 64-bit Raspberry Pi OS on a client.

reraikes, 2021-08-11:
> There is a rpi-update available which may fix this issue. Please update and report.

The latest commit appears to fix Raspberry Pi OS 32-bit.

The latest commit does NOT fix Raspberry Pi OS 64-bit.

Raspberry Pi OS 64-bit boots/run fine with 5.10.27 or lower.

reraikes, 2021-08-11:
@popcornmix

The situation with network booting Raspberry Pi OS 64-bit is a little different...

With "start_x=1" in config.txt:

5.10.46 (23fd671886ba071e04e9787835a05dfc84137679) : OK

5.10.46 (48aef33c6139b7a38013c741f18bf4a1851a1b6d) : OK

5.10.46 (f29ab05611eef385d17675c77190df0b87a0d456) : BROKEN

ivoruetsche, 2021-08-11:
Salü @popcornmix

So looks fine...
I just updated to 3f30de94e5128211a4b3c310cf5a37503af5df96 and kernel to 
Linux localhost 5.10.52-v7+ #1443 SMP Wed Aug 11 13:57:08 BST 2021 armv7l GNU/Linux

My Raspi is booting with this combination and the "start_x=1" in config.txt.

gruss ivo

reraikes, 2021-08-11:
> My Raspi is booting with this combination and the "start_x=1" in config.txt.

"start_x=1" in config.txt can (should) be removed with this fix.

ivoruetsche, 2021-08-11:
> > My Raspi is booting with this combination and the "start_x=1" in config.txt.
> 
> "start_x=1" in config.txt can (should) be removed with this fix.

Hmmm, no, it's not working

reraikes, 2021-08-11:
> > > My Raspi is booting with this combination and the "start_x=1" in config.txt.
> > 
> > 
> > "start_x=1" in config.txt can (should) be removed with this fix.
> 
> Hmmm, no, it's not working

Then you must not have updated (rpi-update) the firmware that's being downloaded over the network into the client Raspberry Pi.

I just tested again (Raspberry Pi OS 32-bit without "start_x=1" in client's config.txt):

5.10.52 (2073a27299fc17cf93152e9c8d9f6998fe484c97) : FAILURE

5.10.52 (3f30de94e5128211a4b3c310cf5a37503af5df96) : SUCCESS

peterharperuk, 2021-08-12:
>> The latest commit does NOT fix Raspberry Pi OS 64-bit.

@reraikes I've updated to 3f30de9 and I put arm_64bit=1 in config. ini. It seems to boot ok and according to "uname -m" is running "aarch64". What problems are you seeing?

reraikes, 2021-08-12:
> > > The latest commit does NOT fix Raspberry Pi OS 64-bit.
> 
> @reraikes I've updated to 3f30de9 and I put arm_64bit=1 in config. ini. It seems to boot ok and according to "uname -m" is running "aarch64". What problems are you seeing?

Are you starting with?:

https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2021-05-28/2021-05-07-raspios-buster-arm64-lite.zip

Simply adding "arm_64bit=1" to a 32-bit image is not the same.

reraikes, 2021-08-12:
@peterharperuk

I just rechecked and the failure is 100% reproducible here.  Here are the steps for the client Raspberry Pi OS:

1. Start with a fresh image:

https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2021-05-28/2021-05-07-raspios-buster-arm64-lite.zip

2. Add an 'ssh' file to the BOOT partition

3. Upgrade it:

apt update && apt full-upgrade

4. Add "start_x=1" to config.txt

5. Use firmware 5.10.46 (48aef33c6139b7a38013c741f18bf4a1851a1b6d):

rpi-update 5.10.46 48aef33c6139b7a38013c741f18bf4a1851a1b6d

6. Note that network booting this to the client is successful and runs as expected

7. Use firmware 5.10.46 (f29ab05611eef385d17675c77190df0b87a0d456) or later:

rpi-update f29ab05611eef385d17675c77190df0b87a0d456

8. Note that network booting this to the client fails

peterharperuk, 2021-08-13:
It seems to work for me, as does the tip at ad5406d7597e017298474a96323ced593da01457

reraikes, 2021-08-13:
> It seems to work for me, as does the tip at ad5406d7597e017298474a96323ced593da01457

What is "the tip at ad5406d7597e017298474a96323ced593da01457"?

Can we get some input from the Raspberry Pi OS developers (@popcornmix), please?

timg236, 2021-08-13:
> > It seems to work for me, as does the tip at ad5406d7597e017298474a96323ced593da01457
> 
> What is "the tip at ad5406d7597e017298474a96323ced593da01457"?
> 
> Can we get some input from the Raspberry Pi OS developers (@popcornmix), please?

peterharperuk is a Raspberry Pi OS developer and he fixed the firmware issue. Given the nature of the fix the 64 bit issue sounds like a different problem.  Logs from your 64 bit setup showing the exactly where during boot would likely speed up the debugging process.

reraikes, 2021-08-13:
> Logs from your 64 bit setup showing the exactly where during boot would likely speed up the debugging process.

[cmdline.txt](https://github.com/raspberrypi/firmware/files/6984674/cmdline.txt)
[config.txt](https://github.com/raspberrypi/firmware/files/6984675/config.txt)
[daemon-5.10.46 (48aef33c6139b7a38013c741f18bf4a1851a1b6d).txt](https://github.com/raspberrypi/firmware/files/6984676/daemon-5.10.46.48aef33c6139b7a38013c741f18bf4a1851a1b6d.txt)
[daemon-5.10.46 (f29ab05611eef385d17675c77190df0b87a0d456).txt](https://github.com/raspberrypi/firmware/files/6984677/daemon-5.10.46.f29ab05611eef385d17675c77190df0b87a0d456.txt)
[UART-5.10.46 (48aef33c6139b7a38013c741f18bf4a1851a1b6d).txt](https://github.com/raspberrypi/firmware/files/6984678/UART-5.10.46.48aef33c6139b7a38013c741f18bf4a1851a1b6d.txt)
[UART-5.10.46 (f29ab05611eef385d17675c77190df0b87a0d456).txt](https://github.com/raspberrypi/firmware/files/6984679/UART-5.10.46.f29ab05611eef385d17675c77190df0b87a0d456.txt)

reraikes, 2021-08-13:
Please ignore the close/reopen.  My mouse pointer jumped as I clicked on "Comment".

popcornmix, 2021-08-13:
@reraikes are you running headless? Can you try with kms driver commented out?

reraikes, 2021-08-13:
> are you running headless?

Yes

> Can you try with kms driver commented out?

That's the source of the problem.

With the kms driver commented out, all is well without "start_x=1" in config.txt up through the latest commit:

5.10.52 (ad5406d7597e017298474a96323ced593da01457)

popcornmix, 2021-08-18:
Duplicate of https://github.com/raspberrypi/linux/issues/4457

peterharperuk, 2021-08-18:
I'm going to close this as I believe the original network boot issue is fixed. We have tweaked our testing to catch regressions like this in future. Please reopen if you disagree.

reraikes, 2021-08-18:
Should I open a new issue for the 64-bit problem?

Or is the solution going to be that the KMS driver must be removed in headless network boot installations?

peterharperuk, 2021-08-18:
I assumed that issue would be addressed by https://github.com/raspberrypi/linux/issues/4457 ?

reraikes, 2021-08-18:
I'm confused.

I was not aware of https://github.com/raspberrypi/linux/issues/4457 (it was not mentioned here before 10 minutes ago).

The problem is still present in the currently available Raspberry Pi OS 64-bit software.  Is there a forthcoming update that will incorporate a solution to this problem?  It's still broken in the latest updates (apt + rpi-update).

peterharperuk, 2021-08-18:
I'll clarify and get back to you. Reopening so I don't forget.

gorky, 2022-03-14:
Seeing this issue in 5.10.103-v8+ when there is a data HDD attached to the USB port.